pbatard/rufus #709: Error creating partition

danieldsc, 2016-03-08:
Hi

I got the "error while formatting" and after reading this forum I downloaded last version of Rufus, 2.7, and tried the Alt+z to zero my device.
When I tried to create Windows 7 bootable device again I got the message "error creating partition" right after I clicked start. Then I formatted using windows (FAT32) and then I got the same error while formatting message. Now I am trying Rufus bad blocks checking, but any ideas about what it might be? I don't know where to get the log file. Please inform.

Thanks

pbatard, 2016-03-08:
To get the log, just click on the "Log" button (bottom left). A new Window will open with a "Save Log" button you can click (or you can just copy/paste the content).

It does look to me like your USB could indeed be defective, but without the log of the error you get when formatting, or the result of the bad blocks check, that's hard to say for sure.

danieldsc, 2016-03-08:
Sorry, it was in portuguese and although I am brazilian I got confused with the word because of translation. Here it is (bad blocks checking still going):

Rufus version: 2.7.855
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0416
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCULXFRER_BR_DVD'
  Size: 4292116480 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: W7-COMPLETO-x86-x64-AUTO-ATIVAÇÃO-downloadsfulltorrent.blogspot.com.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x0000001F] Um dispositivo conectado ao sistema não está funcionando.
Could not reset partitions
Re-mounted volume as 'D:' after error

Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x0000001F] Um dispositivo conectado ao sistema não está funcionando.
Could not reset partitions
Re-mounted volume as 'D:' after error

Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Bad Blocks: Checking from block 0 to 15769599
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing

danieldsc, 2016-03-08:
Now the complete log after bad blocks checking (still got the error while formatting):

Rufus version: 2.7.855
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0416
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCULXFRER_BR_DVD'
  Size: 4292116480 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: W7-COMPLETO-x86-x64-AUTO-ATIVAÇÃO-downloadsfulltorrent.blogspot.com.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x0000001F] Um dispositivo conectado ao sistema não está funcionando.
Could not reset partitions
Re-mounted volume as 'D:' after error

Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 7.5 GB (8074035200 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
[... 361 lines cut ...]

Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02B59E48
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.5 GB (8072986624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02B59E48
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.5 GB (8072986624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

danieldsc, 2016-03-08:
After the bad blocks checking I used windows drive repairing tool, then I tried Rufus again and got this log after receiving same error while formaating message:

Rufus version: 2.7.855
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0416
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02B59E48
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8072986624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCULXFRER_BR_DVD'
  Size: 4292116480 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: W7-COMPLETO-x86-x64-AUTO-ATIVAÇÃO-downloadsfulltorrent.blogspot.com.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'D:' after error

Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02BE7AB7
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x07)
  Size: 7.5 GB (8072986624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 981, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02BE7AB7
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x07)
  Size: 7.5 GB (8072986624 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

danieldsc, 2016-03-09:
I tried to use a different flash drive and now it is copying ISO file. I will report outcome.

pbatard, 2016-03-09:
```
Bad Blocks: 13357056
Bad Blocks: 13357057
Bad Blocks: 13357058
Bad Blocks: 13357059
Bad Blocks: 13357060
Bad Blocks: 13357061
(...)
```
means your USB is defective. I cannot support you if you try to write to that drive as __it contains bad blocks__.

danieldsc, 2016-03-09:
It worked out on a different flash drive. Thanks Pete! You may close this.

pbatard, 2016-03-09:
Okay.

closed 2016-03-09 by pbatard