pbatard/rufus #711: NoDriveTypeAutoRun gets deleted

englishbond, 2016-03-13:
..
 Hi, thanks for your very useful work, but ..
 I'm running XP Pro SP3. I always keep  
 "NoDriveTypeAutoRun"=dword:000000DF
 -- After running either version of Rufus I just got,
 viz., v. 1.4.10 and v. 2.7.855.0 ...
 then when I put in a USB memory stick,
 that annoying window popped up and I had to
 fix the key
 [HKLM\SOFTWARE\Microsoft\Windows\CurrentVersion\policies\Explorer]
 back how I want it.
 ~~~~
 Can you leave this value the way you found it please ..
 Ta very much
 (Can I be the first one to notice this disturbance ?)
 PS. I had to create a GitHub account to say this. Oh dear.
..

pbatard, 2016-03-13:
Rufus uses the Microsoft recommended way of modifying the `NoDriveTypeAutoRun` policy (which is trough Local Group Policies) and does restore it on exit.

Are you saying that you don't see your `NoDriveTypeAutoRun` restored to `000000DF` after Rufus exits (which shouldn't happen)? Or are you simply annoyed that, while Rufus runs, the value is being changed.

englishbond, 2016-03-14:
(Just signed-in, I wrote this before I saw your reply .. )
 The info from
 http://support.microsoft.com/kb/967715
 under "How to selectively disable specific Autorun features"
 was used to guide our setting the registry, so ...
 Ok, I get it. You're not setting the  "NoDriveTypeAutoRun" registry value
 which controls the behaviour. You're using the "Group Policy" API to set
 a "Policy" which alters the registry value. But .. Out of the box, as we have it,
 the machines never had the folder
 %systemroot%\system32\GroupPolicy
 from the first beginning, so no  registry.pol  file neither, but do have
 [HKLM\SOFTWARE\Microsoft\Windows\CurrentVersion\policies\Explorer]
 "HonorAutoRunSetting"=dword:00000001
 "NoDriveTypeAutoRun"=dword:000000DF
 which are inserted by a script on the unattended-setup disk.
 So when  Rufus  runs, it creates a new file
 %systemroot%\system32\GroupPolicy\Machine\Registry.pol
 which does not reflect the current state of the registry entry,
 then when  Rufus  quits, this wrong file causes the deletion
 of the  "NoDriveTypeAutoRun" value.
 People who shouldn't be using Windows, honestly, have never heard of
 gpedit.msc, don't know where to find it, can't hardly read and write, etc.,
 still need to use Windows.
 _Unsatisfactory Workaround_
 I added in the appropriate places to the template file
 %systemroot%\inf\system.adm
 the line  (under  POLICY !!Autorun , 2 instances )
 	NAME !!Autorun_CDonly	VALUE NUMERIC 223
 and  ( in the strings )
  Autorun_CDonly="All but CD-ROM"
 ... then ran  gpedit.msc  to set the "Policy" how we want it.
 Then sure enough,  Rufus  does what you said, sets the value
 back to how it was before. Good-O.
 But this is an enormous faff. And I don't know how to set it up
 with a script, and it creates the new "GroupPolicy" folder
 wasting a bit of disk space.
 Compared to directly controlling the "NoDriveTypeAutoRun" value
 and possibly deleting the
 [HKCU\Software\Microsoft\Windows\CurrentVersion\Explorer\MountPoints2\?]
 "_AutorunStatus"=
 .., it looks like more work.
 Thanks for helping me find-out these details, and for all that research and coding.
..

pbatard, 2016-03-14:
Yeah, I played with manipulating the registry directly on Windows 10, and confirmed the same behaviour. When the Autorun policy is set through Group Policy things are restored as they should, but if you play with the registry key directly, that's not the case.

I tried to see if I could add a workaround for that in Rufus, but the problem, is even if after I added code that looks up the `HKLM\SOFTWARE\Microsoft\Windows\CurrentVersion\policies\Explorer\NoDriveTypeAutorun` in the registry before the policy is applied, store its value if it is set, and restore that registry key __after__ the Policy has been reverted (i.e. after issuing [`IGroupPolicyObject::Save()`](https://msdn.microsoft.com/en-us/library/windows/desktop/aa374281%28v=vs.85%29.aspx) and `Release()`), there still appears to be a delay before the policy is applied and the relevant registry keys are modified, which means that `NoDriveTypeAutorun` still gets deleted (NB: I made sure that my registry restoration code was correct, by confirming that the key was being created if I altered one character for instance, so I am 100% positive that `NoDriveTypeAutorun` gets altered by Windows __after__ it reports that the Group Policy has been reverted).

Therefore, I'm afraid that I don't see a workable solution for that issue as it doesn't look like Windows offers an option to find out when a Group Policy has been applied/reverted, to know when the registry accessed by said policy is safe to modify and the only way I see would be to add a delay on exit, before trying restore the registry key, which is less than ideal and more than I'm willing to go for something that is unlikely to affect many people. Besides, I don't think the root of the problem is with Rufus but more with Microsoft not having implemented a mechanism that saves/restores affected registry keys when changing Group Policies...

The only thing I can do here is second Microsoft's advice that these registry keys should __not__ be modified directly, but only indirectly through Group Policies, and that you may observe unwanted behaviour if you do modify them directly...

As I don't think there's anything else I can do, I will close this issue.

closed 2016-03-14 by pbatard