pbatard/rufus #713: Cannot create Architect Linux  USB drive with ISO mode

llde, 2016-03-15:
Rufus (tested with versions 2.6 2.7 and 2.8 beta) cannot correctly create an bootable usb drive for Architect Linux (Arch Linux graphical installer) hibryd ISO. When I start the boot procedure, it blocks trying to mount the usb drive himself, but it cannot succed so the boot procedure stops (Unable to mount  drive + usb drive uuid and label), and a tty terminal is shown.

Using the DD mode, it succed, and his properly bootable

pbatard, 2016-03-15:
That's because, unlike the maintainers of Arch Linux, the maintainers of Architect Linux failed to consider that not everybody is going to use ISOHybrid, and that as a result, they need to cater for FAT32 filesystems, and __especially__ be mindful that the label they need to use for the partition to mount for should not be more than 11 characters max, and only use uppercase.

As a result, the following from `architect-2016.03.01-64bit.iso`, in `/arch/boot/syslinux/archiso_sys64.cfg` will NEVER work if the file system to mount is FAT32:
```
APPEND archisobasedir=arch archisolabel=architect-201603-64bit
```
because a label such as `architect-201603-64bit` is simply not possible for FAT32.

Instead, they should have taken a page of what Arch is doing and used something like this (from `archlinux-2016.01.01-dual.iso`'s /arch/boot/syslinux/archiso_sys64.cfg`):
```
APPEND archisobasedir=arch archisolabel=ARCH_201601
```
as well as setting their ISO label accordingly.

I think you should report the issue to the Architect Linux maintainers, and ask them to follow more closely on what Arch does, or at least, be aware that if the labels they use for their ISO are not compatible with FAT32, they are going to leave some users stranded.

Oh, and for the record, while ISOHybrid generally works, one thing Linux distro maintainers should also be aware of is that [not every Windows user is happy about using a method that will make it look like their drive has suddenly been greatly reduced in size](https://www.reddit.com/r/techsupport/comments/499b5c/usb_stick_capacity_shrunk_to_2mb/), or even vanished (which is why Rufus recommends to only write in DD/ISOHybrid mode last). So they should really follow suit with what Arch does and ensure that their process can work with FAT32.

This being said, if many Arch derivatives start running into this issue (IIRC this is the second time I'm seeing something like for an Arch based distro), I __may__ add a workaround in Rufus that patches the labels in `/arch/boot/syslinux/archiso_sys64.cfg` and `/arch/boot/syslinux/archiso_sys32.cfg`, as I'm already doing if I spot such a label in the global ISOLinux/GRUB config files. However, because these config files are very Arch specific (and could be changed tomorrow if the Arch people feel like it), I'd rather avoid having to do that in Rufus, unless it becomes apparent that this is becoming a widespread issue.

llde, 2016-03-15:
Thanks for your time. I reported to Carl Duff creator and mantainer of architect linux, this issue.

pbatard, 2016-03-21:
I will close this issue now, but feel free to comment if you have updates from the Architect Linux maintainers.

closed 2016-03-21 by pbatard