pbatard/rufus #715: Windows 10 Enterprise ISO not UEFI bootable

Bro-Account, 2016-03-16:
Hi,

I have searched quite a lot about this issue already, but I could not find any related reason.
I want to make a Windows 10 Enterprise 1511 x64 ISO bootable with Rufus, I'm using the default settings to do this.

However, on different DELL computers (Latitude, Optiplex - most recent BIOS updates), no option to UEFI boot the stick will appear. The same happens with different USB sticks. But when I choose the Windows To Go installation it will work like a charm - so the ISO as well as the BIOS and program settings should be fine and the architecture compatible.

It is booting with YUMI too, however Fat32 is used which will not suffice for customized WIMs later.

Any idea why it isn't working and how to fix it?

Thanks a lot!

[rufus.zip](https://github.com/pbatard/rufus/files/176112/rufus.zip)

pbatard, 2016-03-16:
You are partitioning as MBR (first option), and therefore your USB will only be bootable on a BIOS system or in BIOS compatibility mode (UEFI-CSM) which is probably disabled by your UEFI firmware.
When using a Windows ISO, Rufus only allows you to select either BIOS boot or UEFI boot but not both. The reason for this is explained [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode).

So please make sure you select "GPT partition scheme for UEFI" under "Partition scheme and target system type" __after__ having selected your Windows 10 ISO, and try again. You should find that it works much better.

Oh, and you don't have to use FAT32 for UEFI boot (at least when using Rufus). Rufus will happily let you create an NTFS USB that can be booted through UEFI (which allows you to have a >4GB customized WIM).

Bro-Account, 2016-03-16:
Hi, yeah I got that, it's actually why I want to use Rufus for this! :smiley: 

I should have found this, I was searching through the FAQ a while... m(
It's working that way indeed! Thanks for you super fast response!

One, now seeming strange thing I found out later and wanted to add to my initial post is that the MBR stick I created is actually UEFI bootable on the Latitude E7450. I confirmed that by checking 

reg query HKLM\System\CurrentControlSet\Control /v PEFirmwareType

But better something that shouldn't work is actually working than the other way round! :smile:

pbatard, 2016-03-16:
> One, now seeming strange thing I found out later and wanted to add to my initial post is that the MBR stick I created is actually UEFI bootable on the Latitude E7450

That can be explained if the UEFI firmware of the E7450 contains an NTFS driver. If that's the case then, because Rufus does not actually remove the UEFI boot files, and the only thing usually preventing UEFI boot is the use of NTFS, then you may find that the USB will indeed boot in UEFI mode. As more and more UEFI firmwares seem to start embedding an NTFS UEFI driver, I may actually add code in Rufus to actually remove the UEFI boot files when creating a BIOS/UEFI-CSM Windows drive, but I'm kind of waiting to see if this is going to become an actual issue for some users.

> I confirmed that by checking `reg query HKLM\System\CurrentControlSet\Control /v PEFirmwareType`

I tend to just launch `msinfo32` and then check if it reports `UEFI` under `BIOS mode`. Oh, and it will also tell you if Secure Boot is being used, if you are interested.

If it's okay with you, I will close this issue now.

closed 2016-03-16 by pbatard

Bro-Account, 2016-03-16:
> I tend to just launch msinfo32 and then check if it reports UEFI under BIOS mode. Oh, and it will also tell you if Secure Boot is being used, if you are interested.
Good to know for the actual installed Windows. For WinPE I think I have to use my way though.

So, I learned a lot! Thanks for the great support again! :+1:

Hattener, 2017-07-31:
http://www.vanskeys.com, they gave us professional solutions on activating our computers. Very patience, I like this guy.﻿