pbatard/rufus #716: BIOS recognize bootable Windows XP/7 but not Linux

francesco1119, 2016-03-18:
I need to reinstall OS on several POS.
Their Bios is a normal "Phoenix - AwardBIOS v6.00PG"

I have tried with several USB keys and with several BIOS options (USB-HDD, USB-ZIP, USB-FDD, USB-CDROM).

When Windows XP/7 ISO are burned on the USB the computer recognize the USB key at startup (USB-HDD mode) and I can install Windows on it.
But, with the same BIOS settings, the computer won't recognize ANY Linux live distro (tried Ubuntu 12, Ubuntu 14, Xubuntu, Slax, Caine, Deft)

(all ISO are 32bit)

pbatard, 2016-03-18:
I'd like to help you, but I'm pretty sure this is not a Rufus issue. What happens if you try with Unetbootin or YUMI or some other tool? Also have you tried writing these ISOs (some of which are ISOHybrid) in `DD` mode when Rufus proposed the option?

If you still have the same issue, then it's clearly not a Rufus issue and you will have to contact your computer manufacturer, as I hope you can understand that I have no idea about the hundreds of different BIOS quirks and configurations that different computers have, to enable boot, and it is unreasonable to ask the writer of an application for help with a hardware/environmental issue that has nothing to do with their app.

I will therefore close this issue. However, if you can demonstrate that only bootable drives created by Rufus (even in `DD` mode) are affected, and that drives created by other applications work just fine, feel free to reopen (but please provide a complete log of how you created one of the non working Linux bootable USB drives).

closed 2016-03-18 by pbatard

francesco1119, 2016-03-20:
Thank you,

You were right, Rufus worked by using the |DD| mode instead of |ISO|