pbatard/rufus #717: Device eliminated because it appears to contain no media.

Skoshea, 2016-03-18:
Heya. I'd just like to preface that I'm -really- unfamiliar with USB flashes. 
So much so, that this is actually my first drive. On that, I'm not trying to hold Rufus accountable for "breaking" my flash; just looking for some help outside of the FAQ.

I made a Windows 10 bootable for my poor laptop, which had to be booted from startup because it had no OS, blah blah. Simple enough stuff. The first time around it went fine and worked as advertised. My next step was to remove the files/.iso from the drive and put back my music and some other files, which I had moved from said poor laptop, to my even poorer desktop. Once I did that, something, not entirely sure what, went wrong with my poor laptop. So, I went to reset my laptop again and put back the Windows 10 .iso back onto the flash drive.. Somewhere along the line, as I was writing the files to the drive again, I was presented with this log. (Which I believe is full.)

Rufus version: 2.7.855
Windows version: Windows Vista SP2 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6544)
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB SD Reader USB Device' (058F:9360)
Device eliminated because it appears to contain no media
Found USB device 'Generic USB SM Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
1 device found
Could not open GUID volume '\\?\Volume{63fec300-e928-11e5-96d7-00904b840e39}': [0x0000001F] A device attached to the system is not functioning.
No volume information for drive 0x85
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3771, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0004BF48
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 28.9 GB (31023300608 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CCSA_X64FRE_EN-US_DV5'
  Size: 4414115840 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: Win10_1511_1_English_x64.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE5 for write access
No drive letter was assigned...
Will use 'K:' as volume mountpoint
Could not open GUID volume '\\?\Volume{63fec300-e928-11e5-96d7-00904b840e39}': [0x0000001F] A device attached to the system is not functioning.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Could not read volume name
Format error: Undetermined error while formatting.
Could not open GUID volume '\\?\Volume{63fec300-e928-11e5-96d7-00904b840e39}': [0x0000001F] A device attached to the system is not functioning.

Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6544)
Found USB device 'Generic USB CF Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB device 'Generic USB MS Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB SD Reader USB Device' (058F:9360)
Device eliminated because it appears to contain no media
Found USB device 'Generic USB SM Reader USB Device' (????:????)
Device eliminated because it appears to contain no media
1 device found
Could not open GUID volume '\\?\Volume{63fec300-e928-11e5-96d7-00904b840e39}': [0x0000001F] A device attached to the system is not functioning.
No volume information for drive 0x85
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3771, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001581BD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 28.9 GB (31023300608 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

The drive is about a week or so old. While I'm not doubting that I could've simply gotten' a faulty drive this time around, I figured I'd troubleshoot a bit more than not at all, before I buy another one.

Please and thank you for any assistance. /o/

I apologize in advance for any lack of clarity on this issue or ignorance of the subject.

pbatard, 2016-03-18:
`[0x0000001F] A device attached to the system is not functioning.`

This is usually the sign of a hardware issue with the flash drive. Unfortunately, a drive being new does not mean it's not going to fail.

The following also tends to point to a hardware failure:
```
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (NTFS)...
Could not read volume name
Format error: Undetermined error while formatting.
```

I would encourage you to run a bad blocks check in Rufus (check the first checkbox) and see what happens. All in all however, it does look to me like you'll have to replace your drive.

Skoshea, 2016-03-18:
> Unfortunately, a drive being new does not mean it's not going to fail.

Aye, of course. -sigh
It's a good thing these are pretty cheap these days.

Hm, will do. I'll update in a moment.

Skoshea, 2016-03-18:
Excuse the double post, but I'm not sure how notifications work on this forum. 

Interestingly enough, (or maybe not at all) running a bad blocks check allowed me to download everything again. 

I won't post the entire log, as it's very long and simply listing destinations of files, but here's what I'd imagine is important. 

Anything strange in there? Also, should I still get another drive?

Finalizing, please wait...
F:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...
Volume label is J_CCSA_X64FRE_EN-US_DV5.

CHKDSK is verifying files (stage 1 of 3)...
  1152 file records processed.
File verification completed.
  0 large file records processed.
  0 bad file records processed.
  0 EA records processed.
  0 reparse records processed.
CHKDSK is verifying indexes (stage 2 of 3)...
  1366 index entries processed.
Index verification completed.
  0 unindexed files processed.
CHKDSK is verifying security descriptors (stage 3 of 3)...
  1152 security descriptors processed.
Security descriptor verification completed.
  107 data files processed.
Windows has checked the file system and found no problems.

  30296191 KB total disk space.
   4342244 KB in 969 files.
       464 KB in 109 indexes.
     68031 KB in use by the system.
     65536 KB occupied by the log file.
  25885452 KB available on disk.
      4096 bytes in each allocation unit.
   7574047 total allocation units on disk.
   6471363 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6544)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3771, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01565C49
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.9 GB (31023300608 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-03-18:
Well, it could also be an issue with your USB connectors, though this is an USB 2.0 device and not USB 3.0 (where connections are prone to interferences when using extension cables for instance). Or it could be something in your OS environment that temporarily interfered with writing to the USB. Some overzealous security solutions are also known to create problems. Or it could be flaky flash memory behaviour.

If the bad blocks check worked, and you don't seem to get any errors, then your device is probably okay, as the bad blocks check does write/read the whole capacity of the drive at low level. So I would keep trying to use the device for a while and see what happens

However, I have to point out that `[0x0000001F] A device attached to the system is not functioning` errors are external to Rufus (basically they happen when Rufus asks the OS to perform an operation, and the OS replies that it wasn't able to complete the request because of what it perceives as a hardware issue), and tend to indicate some kind of hardware issue, so you probably want to keep an eye out for this drive.

Skoshea, 2016-03-18:
Hm, that sounds quite likely. This computer is pretty battle-tested. 

Gotcha'. I'll do just that.

Thank you very much sir, you're a saint. I hope you find love, riches and all that good stuff.

Have a good evening.

closed 2016-03-18 by Skoshea

pbatard, 2016-03-18:
>  I hope you find love, riches and all that good stuff.

Thanks for the kind words. I feel like I'm closer to all that thanks to talking to you, so I will wish you the same. :smile: