pbatard/rufus #718: Rufus stalls when confronted with unusually large allocation units

deadair3210, 2016-03-21:
I have a 64 GB SD card that I have formatted to FAT32 with guiFormat, and when it is selected in Rufus, the program stalls when attempting to detect the sector count of the card, and finally gives up with a Semaphore timeout.
Version - 2.8.883 Beta
Log - https://gist.github.com/deadair3210/7cdb963598f2efb47d2c
[image]

pbatard, 2016-03-21:
Okay, a few things:
1. You're using a non USB SD card, which is __EXPERIMENTAL__ in Rufus 2.8 (see the [Changelog](https://github.com/pbatard/rufus/blob/master/ChangeLog.txt))
2. I don't see the link between large allocation units and this issue.
3. The current version of RidgeCrop's `fat32format` and `guiFormat` actually have a bug with regards to how they compute the FAT size (see the TL;DR of [this SysLinux mailing list post](http://www.syslinux.org/archives/2016-February/024850.html)). I don't believe this is likely to create the issue you are seeing (especially as I don't believe that how the card was formatted has anything to do with this), but you might want to be mindful that `guiFormat` may have some bugs.
4. By the looks of your error, the problem has to do with __something__ on your OS preventing read access to the MBR of your SD card. My guess is that you have a security application or something that is preventing read-access to the lower sectors of your card, especially as, as indicated in the next point, I am not seeing the issue when I try to replicate it. What happens for instance, if you try to read the MBR using [HxD](https://mh-nexus.de/en/hxd/) (Download and install the app, run it as admin, go to _Extras_ &rarr; _Open disk..._ and select the _Removable Disk #_ that corresponds to your SD card &mdash; I suspect you will get an error here too).
5. I don't see the issue at all with the following 64GB microSD card, which I formatted with `guiFormat`, and with the same version of Windows 10 x64 as the one you are using:
```
C:\>chkdsk G:
The type of the file system is FAT32.
Volume Serial Number is 1216-083F
Windows is verifying files and folders...
File and folder verification is complete.

Windows has scanned the file system and found no problems.
No further action is required.
   62,536,768 KB total disk space.
           32 KB in 1 hidden files.
           32 KB in 1 files.
   62,536,672 KB are available.

       32,768 bytes in each allocation unit.
    1,954,274 total allocation units on disk.
    1,954,271 allocation units available on disk.
```
and
```
Rufus version: 2.8.883 (Beta)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 3.0 device 'Lexar microSD RDR USB Device' (05DC:B051)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7787, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x020EB874
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.7 GB (64053313536 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

deadair3210, 2016-03-21:
Unfortunately, I dont have a USB card reader, and I can no longer get the semaphore timeout to happen. Must have been a weird fluke. Since guiFormat has issues, do you have any recommendations for tools to format in Large FAT32

pbatard, 2016-03-21:
> Must have been a weird fluke

Looks like it. Unfortunately, there are a lot of security and other apps on Windows that can prevent legitimate low level access to flash or disk media. Looking for error code `0x00000079` , I also seems to me like [a lot of people are reporting this issue with SD cards](https://social.technet.microsoft.com/Forums/windows/en-US/c3fc9f5d-c073-4a9f-bb3d-b7bb8f893f78/error-0x80070079-the-semaphore-timeout-period-has-expired?forum=itprovistanetworking), which I could see related to flaky hardware, or flaky contacts when the card is inserted, and which seems unrelated to Rufus, since people are getting this kind of error even when simply copying files.

> do you have any recommendations for tools to format in Large FAT32

Well, Rufus 2.8 corrects the issue that `guiFormat` has... ;) But of course, this requires being able to access your card first.

Since I don't think there's much I can do about this issue, I will close it.

closed 2016-03-21 by pbatard

deadair3210, 2016-03-21:
Ok, thanks. I would test if the card can be accessed now that the semaphore issue has disappeared, but the card now has things on it. The issue appeared after I formatted the card with guiFormat, but disappeared after I formatted it with a tool on my 3DS, which the card is now being used for.