pbatard/rufus #720: Rufus hangs on "writing master boot record"

Scarfsickle, 2016-03-22:
This post is really for /u/_Akeo_. There have been no crashes, but it still is not working out for me.

00000001	0.00000000	[2380] Couldn't get Steam Controller input : ACCESS_DENIED - reverting to default M/KB Configuration	
00000002	4.78901291	[2380] Couldn't get Steam Controller input : ACCESS_DENIED - reverting to default M/KB Configuration	
00000003	6.49129152	[1320] *** Rufus init *** 	
00000004	6.49154520	[1320] Will use settings from registry 	
00000005	6.49165535	[1320] found locale 'en-US' 	
00000006	6.49197149	[1320] loc file not found in current directory - embedded one will be used 	
00000007	6.49315310	[1320] localization: extracted data to 'C:\Users\Zypher\AppData\Local\Temp\Ruf88B1.tmp' 	
00000008	6.49553823	[1320] localization: found locale 'en-US' 	
00000009	6.49613428	[1320] localization: found locale 'ar-SA' 	
00000010	6.49617863	[1320] embedded.loc(329): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000011	6.49617863	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000012	6.49617863	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000013	6.49688101	[1320] localization: found locale 'az-AZ' 	
00000014	6.49772835	[1320] localization: found locale 'bg-BG' 	
00000015	6.49853325	[1320] localization: found locale 'zh-CN' 	
00000016	6.49916840	[1320] localization: found locale 'zh-TW' 	
00000017	6.49975681	[1320] localization: found locale 'hr-HR' 	
00000018	6.49979782	[1320] embedded.loc(2276): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000019	6.49979782	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000020	6.49979782	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000021	6.50040722	[1320] localization: found locale 'cs-CZ' 	
00000022	6.50105095	[1320] localization: found locale 'da-DK' 	
00000023	6.50170946	[1320] localization: found locale 'nl-NL' 	
00000024	6.50175238	[1320] embedded.loc(3438): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000025	6.50175238	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000026	6.50175238	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000027	6.50241947	[1320] localization: found locale 'fi-FI' 	
00000028	6.50328875	[1320] localization: found locale 'fr-FR' 	
00000029	6.50412750	[1320] localization: found locale 'de-DE' 	
00000030	6.50499153	[1320] localization: found locale 'el-GR' 	
00000031	6.50510931	[1320] embedded.loc(5046): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000032	6.50510931	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000033	6.50510931	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000034	6.50609779	[1320] localization: found locale 'hu-HU' 	
00000035	6.50687647	[1320] localization: found locale 'id-ID' 	
00000036	6.50692415	[1320] embedded.loc(5862): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000037	6.50692415	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000038	6.50692415	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000039	6.50788164	[1320] localization: found locale 'it-IT' 	
00000040	6.50871420	[1320] localization: found locale 'ja-JP' 	
00000041	6.50875998	[1320] embedded.loc(6671): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000042	6.50875998	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.17 and v1.0.19.	
00000043	6.50875998	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000044	6.50957632	[1320] localization: found locale 'ko-KR' 	
00000045	6.50962782	[1320] embedded.loc(7090): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000046	6.50962782	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000047	6.50962782	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000048	6.51036596	[1320] localization: found locale 'lv-LV' 	
00000049	6.51119518	[1320] localization: found locale 'lt-LT' 	
00000050	6.51191139	[1320] localization: found locale 'ms-MY' 	
00000051	6.51196051	[1320] embedded.loc(8279): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000052	6.51196051	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.16 and v1.0.19.	
00000053	6.51196051	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000054	6.51265097	[1320] localization: found locale 'nb-NO' 	
00000055	6.51345205	[1320] localization: found locale 'fa-IR' 	
00000056	6.51442003	[1320] localization: found locale 'pl-PL' 	
00000057	6.51530790	[1320] localization: found locale 'pt-BR' 	
00000058	6.51615238	[1320] localization: found locale 'pt-PT' 	
00000059	6.51694012	[1320] localization: found locale 'ro-RO' 	
00000060	6.51778412	[1320] localization: found locale 'ru-RU' 	
00000061	6.51863909	[1320] localization: found locale 'sr-SP' 	
00000062	6.51946259	[1320] localization: found locale 'sk-SK' 	
00000063	6.51947832	[1320] embedded.loc(11895): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000064	6.51947832	[1320] If you are the translator, please update your translation with the changes that intervened between v1.0.18 and v1.0.19.	
00000065	6.51947832	[1320] See https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt 	
00000066	6.52026939	[1320] localization: found locale 'sl-SI' 	
00000067	6.52099037	[1320] localization: found locale 'es-ES' 	
00000068	6.52177858	[1320] localization: found locale 'sv-SE' 	
[... 417 lines cut ...]
00000164	157.97813416	[3616] OnDeviceChange	
00000165	158.05630493	[6884] Analyzing existing boot records... 	
00000166	158.06242371	[6884] Drive has a Zeroed Master Boot Record 	
00000167	158.06484985	[6884] Volume has an unknown Partition Boot Record 	
00000168	158.06527710	[6884] Deleting partitions... 	
00000169	158.07170105	[6884] Clearing MBR/PBR/GPT structures... 	
00000170	158.07228088	[6884] Erasing 128 sectors 	
00000171	158.53768921	[6884] Partitioning (MBR)... 	
00000172	158.54855347	[6884] Closing existing volume... 	
00000173	158.55001831	[6884] Waiting for logical drive to reappear... 	
00000174	158.87942505	[6884] Formatting (NTFS)... 	
00000175	158.94244385	[6884] Using cluster size: 4096 bytes 	
00000176	158.94288635	[6884] Quick format was selected 	
00000177	159.37205505	[6884] Creating file system... 	
00000178	182.34429932	[6884] Format completed. 	
00000179	182.34585571	[6884] Writing master boot record... 	
00000180	761.96545410	[6884] read_sectors: Read error [0x0000045D] The request could not be performed because of an I/O device error. 	
00000181	761.96612549	[6884]   Read: 0, Expected: 512 	
00000182	761.96643066	[6884]   StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200 	
00000183	761.96673584	[6884] Drive does not have an x86 Master Boot Record 	
00000184	2486.94360352	[6884] read_sectors: Read error [0x00000079] The semaphore timeout period has expired. 	
00000185	2486.94409180	[6884]   Read: 0, Expected: 512 	
00000186	2486.94458008	[6884]   StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200 	
00000187	2486.94482422	[6884] Could not read MBR 	
00000188	6652.48583984	[6884] Cancelling

pbatard, 2016-03-22:
1. Why are you using Rufus 2.6 even after you downloaded the 2.7 update?
2. Why did you interrupt the bad blocks check (which is a good way of finding out if your device has an issue)?
3. The following is indicative of a hardware or environmental problem: `Found USB 1.0 device 'SanDisk Cruzer Glide USB Device'`. If your device is detected as using USB 1.0 speed, then there's something very wrong with your computer. At the very least, your device should be
4. The following usually indicates a hardware problem: `The request could not be performed because of an I/O device error.`

All in all, it looks to me like whatever you are observing had nothing to do with Rufus, but everything to do with your hardware or environment. Have you tried a different USB flash drive? Have you tried a full bad blocks check? Can you confirm that you have full administrative access with the account you use?

pbatard, 2016-03-24:
Any update on this?

pbatard, 2016-03-25:
No update &rarr; Closed.

closed 2016-03-25 by pbatard

Scarfsickle, 2016-03-26:
Oh god, so sorry! I thought you'd reply on reddit!!!! I'm so dumb.

1.) I wasn't aware I was using the old version! I'll be more careful when I
try it.
2.) It hung on that as well.
3.) and 4.) Well, I have no clue how to fix that, ah.

 I will try a friend's flash drive. I'll check for "bad blocks" if the
operation doesn't work without the check. And yes, I am the admin of my
computer.

Thank you so much for getting back to me, I apologize for all my delayed
replies.