pbatard/rufus #721: Rufus continues without warning even if downloading newer Syslinux files fails

ljani, 2016-03-23:
I was using Rufus to burn the Ubuntu 15.10 netinst ISO image to usb stick. I was prompted by Rufus for [Internet access](http://i.imgur.com/SJ4Gzm9.png), which I allowed. But I had not allowed Rufus to access through my firewall and Rufus didn't warn me about this, just silently continued the installation with the embedded Syslinux.

I'd argue it's better to warn the user if the download fails before continuing with the embedded version.

Here's the log:

```
Using image: mini.iso
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20150813/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - will try to use embedded 6.03 version instead
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/20150813/ldlinux.bss
Unable to send request: The attempt to connect to the server failed.
Could not download the file - will try to use embedded 6.03 version instead

Format operation started
```

Thanks for the wonderful software!

pbatard, 2016-03-23:
Currently, the fallback to embedded is likely to work (since 6.03 is stable, and most distros have moved to using the release), so I don't think a warning is warranted. I think the message from the log is enough. The main issue was that, at some stage, 6.03 versions could be incompatible with one another, which is actually why, even if Rufus embeds a 6.03 version, we may try to find a match on the server, but since the 6.03 release, things are much better.

You should get an error when the versions don't match (e.g. your ISO uses 6.04 but Rufus can only provide 6.03).

All in all then, I don't think I will change the current behaviour.

closed 2016-03-23 by pbatard

ljani, 2016-03-23:
Ah, okay. Based on the request, I thought it was important to have the exact version. Thanks.