pbatard/rufus #723: ATTEMPTED_WRITE_TO_READONLY_MEMORY (WimFsf.sys)

tamtakoe, 2016-03-25:
I have an error on blue screen in installation begining: ATTEMPTED_WRITE_TO_READONLY_MEMORY (WimFsf.sys)

MBR BIOS and UEFI mode
Windows 10 pro last official ISO, created by MediaCreationTool
Intel nuc6i5SYK, bios v. 0039

It does not work in BIOS mode and works in MBR UEFI mode with error
But it works in GPT UEFI mode

pbatard, 2016-03-25:
1. As always, full log of how you create the USB would be great.
2. According to google, wimfsf.sys errors could indicate an issue with your ISO. Please download the relevant retail Windows 10 from [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench) and try with that one instead of the one produced by the Media Creation tool. Also, once you have selected your ISO in Rufus (the new one from Tech Bench), please click on the `#` button at the bottom of the Rufus interface and post your SHA-1, so that we can double check that this ISO is not corrupted.
3. Finally, what happens if you repartition (MBR) and format your drive manually and just copy the files from the ISO. This should boot through UEFI and indicate whether this really has anything to do with Rufus.

pbatard, 2016-04-03:
No update from OP &rarr; Closing.

closed 2016-04-03 by pbatard