pbatard/rufus #724: Can't make a bootable drive from ISO using exFAT

guitarjorge24, 2016-03-26:
When I select the exFAT format, I can't choose an ISO to use to create a bootable drive. Is it not possible to boot from an exFAT formatted drive? Or is it just that this feature is not supported in Rufus?

pbatard, 2016-03-26:
It is possible if you really want to but the issue is that there's no ISO out there that is set to handle exFAT filesystems, so it's pointless.

DOS/FreeDOS definitely can't boot from exFAT, Linux distros won't either (unless the distro maintainers really went out of their way), UEFI most certainly won't natively, and I doubt the Microsoft Windows installer can work with exFAT (but even if they do, it's pointless preferring exFAT over NTFS). All in all, even if, in the absolute, someone __could__ technically create an ISO that is compatible with exFAT boot, it's just not worth it to try to add support for it in Rufus... and then have to answer questions such as "Why does my ISO doesn't boot from exFAT?", for the 99.99% of ISOs out there  that will never boot from exFAT.

Can you please clarify __why__ you think you need exFAT?

guitarjorge24, 2016-03-27:
That's strange, about an hour ago I actually managed to make a bootable USB flash drive from a Windows 10 ISO while formatting the USB in exFAT. I used [this program](http://www.isotousb.com/)
I managed to boot from it on a Toshiba laptop (I'll try it later on a desktop PC)

The reason I prefer to use exFAT instead of NTFS is that exFAT is read and write compatible with both Windows and Mac OS (you can't write to NTFS on a Mac). Besides using this particular USB flash drive as a Windows 10 installation disk, I also use it to transfer files from home PCs into workplace Macs (which I can't do with an NTFS drive). I wish Macs didn't exist but they're part of our society, like it or not.

pbatard, 2016-03-27:
Well, as the UltraISO site says:
> This software currently only support Windows bootable disk, can work with both BOOTMGR and NTLDR boot mode

So that's probably why it might be easier to make it work with exFAT. Also, I'd be very surprised if that exFAT drive boots in UEFI mode.  On the other hand, Rufus also supports Linux, DOS, UEFI and other boot modes, so adding exFAT just for Windows images, and just to satisfy the __very few__ people who will want to use their bootable drive between a Mac and a PC (and __MUST__ have a >4GB file to justify not being able to use FAT32) seems like a waste of development time, to be brutally honest. [I just don't see enough people who would benefit from that feature compared to the amount of time it would take me to add and __more importantly__ support it in Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

I also find it a bit strange that you must use a bootable drive created by Rufus to transfer files between home and work. Surely, you can reformat the drive to exFAT if you need to transfer files after you're done with creating bootable media for work, and I would also expect you to have a dedicated USB drive at work for OS installation if you are being tasked with performing those (I doubt a workplace would be so cheap as to refuse the purchase of a flash drive, especially if it's related to performing OS installs, and I would __strongly__ encourage you to use a different flash drive if you must copy files to or from work). Besides, considering that a decent USB 3.0 drive should only set you back 3 minutes on Windows 10 boot drive creation in Rufus, reformatting a flash drive for boot or file transfer isn't that big a deal (and may also prevent you from inadvertently lift confidential work data which you don't need, and which your company may not be too happy to see being copied by an employee).

All in all, I don't see much of a case to justify adding exFAT support specifically for Windows installation media in Rufus. Instead I would suggest that you use the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode to enable dual BIOS+UEFI (which should allow you to use FAT32 for Windows installation, and therefore allow R/W on Mac), or use UltraISO if you really __MUST__ deal with Windows bootable media and use exFAT.

closed 2016-03-27 by pbatard

guitarjorge24, 2016-03-27:
I don't _need_ to use the USB as both bootable drive and file transferer; I just didn't want to have to carry around multiple USB drives. I think what you're saying is reasonable and agree that it's probably not worth your time to add this feature. Thanks for replying

pbatard, 2016-03-27:
Thanks for understanding. If I had all the time in the world, I wouldn't mind adding exFAT support for ISOs that may support it, as it sounds like a nice challenge. But unfortunately, I've also got to be realistic with the features I add. :cry:

Of course, if exFAT support becomes more widespread in the future (e.g. non Windows ISOs start to support it), or if I see a lot of demand, I may reconsider it.

alsayari, 2016-05-22:
It 'd be good to have this future added to Rufus which will set it apart from competitors. Useing exFAT is growing, and many users now operate one both platform so it would be nice to start including it as future especially that nobody has done it. Rufus is fast and very easy to use, and I would like to thank you for creating such an excellent tool that I have been using and prove to be useful.

deadair3210, 2016-05-22:
Its not that the feature isnt wanted in Rufus, its that it makes no sense right now, as almost no ISO's support booting from exFAT

pbatard, 2016-05-22:
> as almost no ISO's support booting from exFAT

Precisely. The boot process is a multi stage affair, which is A LOT more complex than people imagine. It's not because you might be able to get a first stage to boot through exFAT, which is the only part where Rufus can intervene, that you will get anywhere.

For instance, Linux distros must have special boot code in their later boot stage to be able to boot from FAT32. Without this, you may boot the kernel, but the whole process will probably fail along the way, if it is designed only to look for FAT32 content. So there's a good chance most Linux distros won't work if you're using exFAT.

Same goes for UEFI. There is no UEFI firmware that I am aware of, that allows booting from exFAT. So, instead of what is the case for FAT32 (where you can just copy the `bootx64.efi` or `bootia32.efi` files for the whole thing to boot), you'd have to use workarounds, which no guarantee whatsoever that the later boot stages will be able to handle exFAT content.

Also, on a separate matter, exFAT is [encumbered by software patents](https://en.wikipedia.org/wiki/ExFAT#Restrictive_licensing_and_software_patents) which makes it unfriendly for use with Open Source software (which Rufus is), so, on political grounds, I would tend to discourage its use. As a matter of fact, I have failed to see the adoption of exFAT growing, with most parties, for which FAT32 limitations are becoming an issue, choosing to switch to NTFS (which has its own issues, but is better supported) instead. For instance, boot loaders like Syslinux have introduced NTFS support, but haven't done much when it comes to exFAT support.

All in all, I'm afraid there simply just isn't enough support and demand for exFAT boot, for me to spend time on it, so I don't think you will be seeing such a feature in Rufus. Instead, I would encourage you to use NTFS, which is better supported, and for which Rufus does have provisions in the form of [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs).

nateridderman, 2016-11-29:
It would be great if this was mentioned in the FAQ. I think it's common for large flash drives to be formatted with exFAT by default, in which case it's not clear why rufus won't work with them.

kvoz123, 2016-12-04:
The problem is: there isn't any good choice to handle files/system in combined Mac-Wxy-Linux mix. NTFS isn't the best system for Macs, Linux (you can't fscheck it). I agree with your reasons to not support exFAT, but what do you think about UDF? It supported natively onto all these systems, it is bootable, and there isn't file size limitations?

pbatard, 2016-12-04:
I went with NTFS because it's the __de facto__ fallback when FAT32 is not applicable or too limited. For instance, bootloaders like Syslinux didn't try to push for UDF or exFAT use as a FAT32 alternative, but spent time making sure the latest versions worked with NTFS. Also, a lot of UEFI firmwares, such as intel NUCs, are starting to provide a native NTFS driver (but no UDF or exFAT).

So, as far as I'm concerned, NTFS has already won the title of FAT32 replacement, because that's what everybody seems to be aiming to use. This means that I'm not gonna bother trying to support file systems that are unlikely to see that much usage for boot, and that don't bring much of anything compared to NTFS.

Finally, remember that Rufus is fully open source (including the part that allows booting UEFI from NTFS even if you don't have a native driver), so if you __really__ can't stand NTFS and want something else, you can most certainly create your own version with the file system of your choice.

kvoz123, 2016-12-04:
Thank you for explanation! Unfortunately, NTFS hasn't support in Linux for chkdisk functionality, and it doesn't support user/group info.  %-(

knpk13, 2016-12-17:
@pbatard Yikes the filesystem situation seems to be messy. So I have a SanDisk Ultra 128GB Type-C USB drive. Should I format it as NTFS or FAT32 for usage as a bootable USB? Also, for normal usage, exFAT should be the best choice right? Weird that exFAT is becoming the FAT32 replacment and that linux doesn't support it. Also is it okay to frequently reformat drives to different filesystems?

pbatard, 2016-12-17:
> Should I format it as NTFS or FAT32 for usage as a bootable USB?

You should use whatever file system Rufus will select by default. This will change according to the ISO you pick. And remember that Rufus will __always__ reformat your drive when you create a bootable USB (because there are way too many circumstances where we need to repartition the drive and/or start with conditions that we can control so that we can ensure __reliable__ boot).

> Also is it okay to frequently reformat drives to different filesystems?

Many people mistakenly assume that reformatting a drive is a very low level and super critical operation, that is hazardous operation. This couldn't be further from the truth. In quick format mode (which is the default), only a handful of blocks are written, and, as far as the flash drive is concerned, those blocks are __no more different than regular data blocks__. You can completely screw up a format or partition operation, and your drive will be no less usable. Also, almost all flash drives these days have circuitry that aims at reducing wear and tear from write operation, so that even if you tell your drive to write block #0 repeatedly (as would typically happen if you repartition a drive over and over), internally a different memory block will be written, so again, it's no more hazardous than writing to a file.

knpk13, 2016-12-17:
@pbatard I tried Large FAT32 (default). It didnt show up in my XPS 15's boot menu. I tried FAT32 on an older drive, that booted fine. That's why I asked. How can I help?

pbatard, 2016-12-17:
> It didnt show up in my XPS 15's boot menu.

You'll have to figure your boot options then. I hope you can understand that there's no way I can guide Rufus users with every single hardware manufacturer out there, and the various quirks their BIOS/UEFI firmware might have. This, I'm afraid, is something you need to figure out for yourself.

knpk13, 2016-12-17:
@pbatard Ok, that's fair. The other options (NTFS, UDF, and exFAT) don't work. I thought of asking you because all other drives work with my UEFI. When I use "Large FAT32" on a large flash drive instead of "FAT32" (which isnt available for large drives), it doesn't work.

pbatard, 2016-12-17:
I doubt that FAT32 vs Large FAT32 is an issue. But if it is, then whoever designed the BIOS/UEFI firmware screwed up. And, no, you cannot select regular FAT32 for large drives. That's why Large FAT32 exists in the first place, because regular FAT32 can't work with large drives.

knpk13, 2016-12-17:
Sorry to bother you, but can you help me one last time @pbatard ? I can't find documentation on "Large FAT32". I haven't heard of the term before. Can you point me towards resources for large FAT32?

pbatard, 2016-12-17:
http://www.ridgecrop.demon.co.uk/index.htm?fat32format.htm

filamento, 2016-12-19:
Sir,

May I ask why is supporting exFAT such an additional burden?

exFAT is the ideal format for Windows 8 and later (Microsoft's bootsect.exe supports writing exFAT boot sectors since Windows 8)

And bootmgr.exe supports booting from exFAT since Windows 8 also.

And installing Windows Server 2012 and later in MBR mode from a USB stick is MUCH MUCH easier with exFAT (because install.wim is larger than 4GB and doesn't work with FAT32)

Yes, you can format a USB stick in NTFS mode, but it is non-optimal. NTFS is not made for simple USB flash drives. The journalling is not optimal, the write strategy is not optimal, the performance is not optimal.

Flash media like SD cards use exFAT instead of NTFS for very good reasons!

pbatard, 2016-12-19:
I think I already explained, many times over, why I don't see it as a good investment of my time.
If that's still not enough, please see [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

filamento, 2016-12-19:
Ok, fair enough.

Just keep it in mind, because as time passes, exFAT will be more and more important (flash media is getting bigger, and it will come pre-formatted from the factory in exFAT instead of FAT32)

Remember that NTFS will NEVER be an appropiate format for simple flash devices (SSD are another story, completely different). So your current decision to keep using NTFS, while understandable because of compatibility concerns, is nonetheless wrong in the long term.

NTFS is not for flash media, was never meant to, and never will.

Kind regards.

pbatard, 2016-12-19:
> exFAT will be more and more important

Yet, strangely, the UEFI committee, hardware manufacturers and the Open Source community still don't seem to have been convinced with exFAT's oh so incredible advantages compared to NTFS. What we see instead is manufacturers adding native support for NTFS, __NOT__ exFAT into their UEFI firmwares, and the Open Source community (such as Syslinux) concentrate on NTFS compatibility rather than exFAT (thanks in no small part to Microsoft having made their would-be FAT replacement proposal encumbered with patents).

Once again, and whether you like it or not, there is such thing as __de facto__. NTFS is what most people have chosen as __de facto__ replacement for FAT, and as long as this is the case, I don't see much point in adding support for another file system, when it __will not bring any single advantage over NTFS__, except calming down purist who think that flash drive technology is still in its infancy and that using NTFS means that you should expect a life expectancy of 1, 2 weeks tops!

But, since the message doesn't seem to go through, let me educate you a bit further:
- Do you not realize that Rufus is mostly designed for one-of OS installation? This means that, once a user is done installing their OS, and they want to use their drive for data, they can (and usually should) reformat it to whichever file system they like, and remove the OS installation files. I really fail to see what using exFAT over NTFS for the vast majority of Rufus usage scenario which is OS installation, as this is pretty much a __read-only__ operation.
- If you want to have to spend your time explaining to people while their OS can't install because Rufus allowed them to select exFAT, and they happen to use a Windows 7 ISO, please be my guest. Or I guess you expect that I'm going to have to add code that's able to detect whether an ISO is Windows 7 or not, to prevent that. Yeah, I'm sure this will take no time at all, especially with AIO ISOs...
- I'd really like to see your sources on flash drive media degradation when NTFS is being used, especially for a usage scenario that is 95% read-only ops. Oh, and I've also been promised, by many people over the years, that exFAT "will be more and more important". Yet, that still hasn't happen. I hope you can understand that, if I'm going to try to anticipate a trend, I'd rather spend my development time on supporting trends that tend to have some factual credibility to support them. For instance, I will dispute the fact that manufacturers are switching to using exFAT for large flash drives as the last large flash drive I bought was NTFS formatted.
- I don't develop for what people would like to see happen. I develop for what is actually happening. As long as the UEFI committee has made no stride to include exFAT support natively (even though they fairly easily could, as, if the GRUB FAT vs exFAT driver different is anything to go by, there isn't much to it), I'm going to follow their lead and not bother with exFAT either, especially when I have comprehensive support for NTFS and it all but looks to me like the __only__ benefit of adding exFAT support would be to silence a bunch of vocal people who mistakenly think that NTFS will reduce their flash drive's life.

So that's about all I have say about exFAT support.

If you still want to press me to add exFAT support in Rufus, then you'd better come prepared with __verifiable proofs__ that exFAT would bring significant advantages for users.

filamento, 2016-12-19:
I don't know what is the most common usage-case for Rufus, to be honest. I can only give you anecdotal evidence.

In my case, and several of my friends, flash drives are formatted in exFAT to use them seamlessly with Mac OS and Windows.

Sometimes I use the flash drive to install Windows 8 (or windows 10) on a friend's computer, and I find it a real hassle to backup everything in the drive, reformat it, install Windows, and then reformat it again and restore my files again!

I understand that too few people are like me and my friends, so I guess we are a small minority of the Rufus user base.

But the reasons I explained are simple and clear: exFAT is much MUCH better for us.

NTFS also has problems when you try to optimize its performance on USB drives, it becomes less reliable:

https://blogs.msdn.microsoft.com/oldnewthing/20130101-00/?p=5673

pbatard, 2016-12-19:
> https://blogs.msdn.microsoft.com/oldnewthing/20130101-00/?p=5673

Please try again. From that same article:

> It seems that people missed the first sentence of this article. Write-behind caching is disabled by default on removable drives. You get into this mess only if you override the default.

As to what you indicate about exFAT format being used as default by flash drive manufacturers for Mac + Windows compatibility, I'd really like to see a verifiable trend (or actual flash drive manufacturer guidelines) before I'd draw any conclusion. Considering that I don't expect many Rufus users to bring their Rufus formatted drive to a Mac without reformatting them (since Rufus does not support Mac or Mac OS installation), I still don't see much point in adding exFAT support, if it's just to follow what (some/most) flash drive manufacturers choose to use for __blank drives that will be use purely for data and not for boot__.

> I understand that too few people are like me and my friends, so I guess we are a small minority of the Rufus user base.

Yes, that is really the main issue at hand here. And to be honest, if I had all the time in the world, and nothing better to do (and I didn't expect to have to spend a lot of time helping confused users understand why their USB might not boot after they picked exFAT), I wouldn't mind adding exFAT support. But, as with all the features I'd like to see Rufus include, I have to be realist.

filamento, 2016-12-19:
Yes, I understand that, because Windows 7 still has a lot of followers and a huge installed base.

But in the future, Windows 7 will be forgotten, just like Windows XP is starting to be abandoned.

That's why I insist that Rufus should support exFAT instead (probably not this decade, but the next one for sure), in the future almost nobody will want to install Windows 7, and then there'll be no problem to make the switch from NTFS!

guitarjorge24, 2016-12-19:
Let's not waste the dev's time by requesting a feature that has already been rejected for various reasons. If you need to use exFAT, try the program http://www.isotousb.com which worked for me on UEFI on an msi board.