pbatard/rufus #726: iso's booting directly from an unchanged ISO

blackcrack, 2016-03-31:
Hi Akeo and others,

the  setting of task is, let the iso untoutched.
in particular, in the case of the same versions but other achitectures
and then unpack in any case not wished .

take a look there :
http://www.howtogeek.com/196933/how-to-boot-linux-iso-images-directly-from-your-hard-drive/
and there : 
https://www.maketecheasier.com/run-iso-files-hdd-grub2/

this is also well if the iso a special compilation on this iso 
where can only works in this spezially compilation
as iso for booting .
so remains the iso like it is and can unimpeded boot how it is 
and boot in any case as "emulated CD-R-om" and 
with exact this compiliation how it is was thinked .
with this possible can we boot any iso with our usb-stick.
a Linuxderivat and a WinNT.. and therewith diffrent versions of it and  different versions of same distriebutions and manny other distriebutions..

there it is also very easy for change the iso for reactos,
if renamed the trunk in rosboot.iso and chainge it out with an oder version..
or maybe a openmandrivai586.iso and a openmandrivax86_64.iso
or edit the grub-configuration easy for change the menuentry or 
add a menuentry per hand and copy a iso onto the usb-stick.

with untouched iso's booted directly in grub2 ....

examples
```grub2
menuentry "PartedMagic" { 
insmod part_msdos 
insmod ext2 
gfxpayload=$linux_gfx_mode 
set root=(hd0,msdos3) 
loopback loop /boot/pmagic/pmagic-6.7.iso 
linux (loop)/pmagic/bzImage iso_location=/boot/pmagic edd=off load_ramdisk=1 prompt_ramdisk=0 rw vga=normal loglevel=9 max_loop=256 vmalloc=256MiB keymap=de-latin1 de_DE
initrd (loop)/pmagic/initramfs
}
```
```
menuentry “Ubuntu 14.04 ISO” {
set isofile=”/home/name/Downloads/ubuntu-14.04.1-desktop-amd64.iso”
loopback loop (hd0,1)$isofile
linux (loop)/casper/vmlinuz.efi boot=casper iso-scan/filename=${isofile} quiet splash
initrd (loop)/casper/initrd.lz
}
```
```
#!/bin/sh
exec tail -n +3 $0
# This tile provides an easy way to add a custom menu entries. Simply type the
#menuentrys you whant to add after this comment. Be careful not to change
# the'exec tail' line adove. 

menuentry "blablaisobuntu" {
set isopath="/blabla/bla/blubbuntu.iso"
echo "Booting {$isopath}..."
loopback loop (hd0,1)$isopath
linux (loop)/casper/vmlinuz.efi boot=casper iso-scan/filename=${isopath} quite splash #or not "quite splath" 
initrd (loop)/casper/initrd.lz 
}
```
and with other distriebutions like Winnt with the "NTloader" from grub2

best regards
Blacky

pbatard, 2016-03-31:
That's multiboot, [which I am not planning to add to Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support).
There's a big advantage in extracting the ISO in that you can modify the config files on it as required, which may be important for instance if you are using a Maxwell nvidia GPU with today's distros (`nomodeset`) or for plenty of other reasons. Also CD-ROM emulation can require a lot of RAM.

Besides, if you really want to do a multiboot ISO in the way you describe, Rufus already provides the option to install a blank GRUB 2 (if you click on the white arrow next to _Format Options_, to display the advanced mode, then you will be able to select the installation of a blank GRUB or Syslinux in the list), so all you need to do then is copy your ISOs and set up your GRUB config.

closed 2016-03-31 by pbatard

blackcrack, 2016-03-31:
iss clear, but if you maby can support to add one iso on a grub configuration atomatical
can be maby be the solussion and adding more entry's it's not your problem.
and therewith it is not a multiboot but an support for booting an single isofile 
as it is .. so, no mulitboot ;) and the user can and it is in able for custom it self, like i want..
:) and you have done you best's :+1: 
and the user can be also satisfied/happy so, both parts be happy :)
because the unpacking it's really ...gnh.... and not usable for other possibilitys ..

best regards
Blacky

pbatard, 2016-03-31:
As I said, if you really want this feature, you'll have to write the config file yourself.

Adding this presents more challenges than you think of, the first one being to educate Rufus users so that they are able to make the best choice, and that is __VERY DIFFICULT__ to do considering that the vast majority of users, who are only interested in creating a bootable drive for a single ISO, will have no idea why they should choose one over the other. Remember that Rufus is primarily aimed at the general public, not advanced users.

It's already problematic enough getting users to make an educated choice on whether they should write a distro in ISO or DD mode, so adding a 3rd option, when it's not really needed (except for people who think extracting an ISO is _"...gnh...."_, which doesn't exactly strike me as a rational explanation) doesn't strike me as a good development investment. Also this means forcing the use of NTFS for Linux distros (as a lot of ISOs will be >4GB), so, if providing a choice when the user presses Start, this means forcing a switch of the file system after the user selected it, or asking them to go back to do that themselves. In other words, I'd have to redesign the UI in various ways, or add quite a few new prompts, just to add a feature that, IMO, will only be useful for a very limited number of users, even more so as it looks like the whole point is really to use it for multiboot, which I repeatedly stated I won't do in Rufus.

And there are other elements to consider, such as me having to __SUPPORT__ and troubleshoot user issues with this feature __FOREVER__, which the persons suggesting the adding of a feature rarely consider. It's very easy to suggest adding something but it's a completely different story when you're the one who's going to be supporting it. You may also want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

So, once again, unless I see a __massive__ demand for GRUB ISO emulation (and I can only see that happening if the current ISO extraction mode, which AFAIK works just fine, becomes a problem with a lot of ISOs) the blank GRUB is the best you'll get. Besides, if you want multiboot, you'll __have__ to write your own config file anyway, so it doesn't change much.

Now, if you are still unhappy about that, you are of course welcome to fork Rufus and create your own version with that feature. That's the beauty of Open Source, and this may also provide you a better idea of what is really involved in adding features, and why developers will sometime choose __NOT__ to add every feature under the sun, especially if they don't bring much of anything new to what the app can already do...