pbatard/rufus #727: 2.8's device selection shows an internal disk first

voltagex, 2016-04-01:
[image]

```
Rufus version: 2.8.886
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found non-USB removable device 'Samsung SSD 850 EVO 500GB'
Device eliminated because it contains a mounted partition that is set as non-removable
Found USB 3.0 device 'ST4000DM 000-1F2168 USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 26 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'Hitachi HDS724040ALE640 USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 26 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found non-USB removable device 'OCZ-VERTEX3'
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
2 devices found
No volume information for drive 0x80
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 14593, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {7EAB2A08-9AF1-496C-A034-C356A1A7428C}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: ''
  ID: {BA51EAB3-1EE2-4720-A519-8D892D08C940}
  Size: 512 MB (536870912 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: ''
  ID: {8DE47A7A-616F-416D-9B94-215A3D107452}
  Size: 111.3 GB (119495720960 bytes)
  Start Sector: 1050624, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/2014-10-06 (from '/syslinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: '20160308-xenial-amd64'
  Size: 249561088 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: clonezilla-live-20160308-xenial-amd64.iso
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8002732032 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

The OCZ Vertex is the device in question, and probably shouldn't be able to be imaged without a warning.

2.7 doesn't have this problem.

[image]

pbatard, 2016-04-01:
Looks like this is due to non USB card reader support, which was added in 2.8 (which is precisely why I had been reluctant to add it, because differentiating between non USB card readers and internal drives is tricky).

Can I ask you to please enable verbose enumeration debug, by pressing <kbd>Alt</kbd>0<kbd>.</kbd> while Rufus is running (You should see the message `USB Debug enabled` on Rufus' status bar) and then post the log? If you do that, the log should contain a lot of additional information, which should help me fix this in the next version.

pbatard, 2016-04-03:
@voltagex,

Any chance you could provide the extended log as requested?

Without this it will be quite difficult for me to fix this issue. Thanks.

pbatard, 2016-04-07:
@voltagex,

Again, I would __REALLY__ like to obtain the extended log, which is something that should take about 30 seconds top create. Without it, it will be VERY DIFFICULT for me to properly fix this issue.

If you log an issue, please do consider that the developer may require more information and if you can't follow up on an issue you created, this causes problems...

Thank you.

voltagex, 2016-04-08:
Right, sorry. Busy week. I'll get to this ASAP

voltagex, 2016-04-08:
```
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&210f7663&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[016]: USB\VID_0781&PID_5530\200342013309CAF279B4
  Found ID[184]: USB\VID_045E&PID_02E6\435722
  Found ID[251]: USB\VID_0A12&PID_0001\6&E3E7252&0&5
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub30#4&11bb9039&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[059]: USB\VID_174C&PID_55AA\MSFT3012345678926D
Processing Hub 3:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&21d3f4f7&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[225]: USB\VID_1B1C&PID_1B13\0901C00CAE3D8C8153A90F48F5001947
  Found ID[223]: USB\VID_046D&PID_C52B\6&1A8123C&0&8
Processing Hub 4:
  Hub[4] = '\\?\usb#root_hub20#4&358d3b39&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[069]: USB\VID_8087&PID_0024\5&210F7663&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#root_hub20#4&230f5685&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[151]: USB\VID_8087&PID_0024\5&21D3F4F7&0&1
Processing Hub 6:
Processing IDs belonging to 'USBSTOR':
  USB\VID_0781&PID_5530\200342013309CAF279B4
Processing IDs belonging to 'UASPSTOR':
  USB\VID_174C&PID_55AA\MSFT3012345678926D
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_850_EVO_500GEMT0'
Found non-USB removable device 'Samsung SSD 850 EVO 500GB'
Device eliminated because it contains a mounted partition that is set as non-removable
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskHitachi_HDS724040ALE640_0___'
NOTE: Matched instance from sibling for 'USB\VID_174C&PID_55AA\MSFT3012345678926D'
  Matched with ID[059]: USB\VID_174C&PID_55AA\MSFT3012345678926D
  Matched with Hub[2]: '\\?\usb#root_hub30#4&11bb9039&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'Hitachi HDS724040ALE640 SCSI Disk Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 26 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskST4000DM000-1F2168______0___'
  Matched with ID[059]: USB\VID_174C&PID_55AA\MSFT3012345678926D
  Matched with Hub[2]: '\\?\usb#root_hub30#4&11bb9039&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'ST4000DM 000-1F2168 SCSI Disk Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 26 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_____________OCZ-VERTEX32.25'
Found non-USB removable device 'OCZ-VERTEX3'
1 device found
No volume information for drive 0x80
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 14593, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {7EAB2A08-9AF1-496C-A034-C356A1A7428C}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: ''
  ID: {BA51EAB3-1EE2-4720-A519-8D892D08C940}
  Size: 512 MB (536870912 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: ''
  ID: {8DE47A7A-616F-416D-9B94-215A3D107452}
  Size: 111.3 GB (119495720960 bytes)
  Start Sector: 1050624, Attributes: 0x0000000000000000
```

The OCZ Vertex is a non-removable SSD.

pbatard, 2016-04-08:
Thanks a lot - this really helps!

I suspect your OCZ Vertex is a drive that doesn't have any partition mounted in Windows, right?

voltagex, 2016-04-09:
Correct, there's no mounted partitions.

commit 43d9ac04 (2016-04-26): [core] don't list non-USB removable devices by default

* Unless specifically requested with Ctrl-Alt-H
* Closes #727
* Also rename usb.c to dev.c, since we may be listing more than USB devices
closed 2016-04-26 by pbatard