pbatard/rufus #730: Error during installation

crowcrow13, 2016-04-07:
http://i.imgur.com/fHZ5oDg.png

This EFI UEFI nonsense is driving me insane, ever since I switched to SSD and switched my motherboard to UEFI I am having tons of problems installing Windows 7 onto old HDD.

Story:
62 bit system, trying to install windows 7 ult, thats the error i get, bios is now set to both uefi and legacy, thinking its the HDD that is in the wrong

pbatard, 2016-04-07:
```
Check if partition needs the NTFS driver services... NOPE
```

This message comes fro, Rufus' UEFI:NTFS and indicates that the UEFI fimware of your target computer (can you indicate the manufacturer/model of the motherboard?) apparently already has its own NTFS driver, and therefore you should have been prompted with 2 boot options for the USB, one for each of the partitions created by Rufus.

You should then be able to use one of the 2 options below that I hope can help you out:
1. If you are using an __official__ version of the Windows 7 ISO, recreate your USB in Rufus, but __after selecting the ISO__, make sure you change the file system from NTFS to FAT32 (and make sure you use "GPT partition scheme for UEFI"). If you don't have a file that is larger than 4GB on that ISO, Rufus will then create a bootable media that doesn't use UEFI:NTFS and you should avoid the error.
2. If you aren't using an official Microsoft ISO, and have a >4GB file on it, simply uncheck the "Create bootable drive using..." option in Rufus and format your drive as NTFS. Then extract the files from the ISO onto it (using 7-zip for instance). If your target machine really does support NTFS, you should be able to boot that USB.

Now, I strongly suspect that the issue you are running into with UEFI:NTFS is the infinite loop that was reported at pbatard/uefi-ntfs#5 and that seems to occur with MSI hardware. This has been fixed in the latest version of UEFI:NTFS, which will be part of the next version of Rufus, so hopefully, starting with Rufus 2.9, the error you observed should be a thing of the past. However, I do not know yet when Rufus 2.9 will be released.

pbatard, 2016-04-25:
I should point out that a very early TEST version of Rufus 2.9 is available [here](https://rufus.akeo.ie/testing/rufus-2.9_TEST1.exe), with the UEFI:NTFS enhancements mentioned above. If you want to test it and report, I would greatly appreciate it.

pbatard, 2016-05-02:
Since I haven't heard back on the testing I requested, I will consider this issue as fixed and close it.

closed 2016-05-02 by pbatard