pbatard/rufus #731: Why is it sometimes called UEFI and sometimes UEFI-CSM?

kq01526, 2016-04-07:
Hi @pbatard ,

please forgive me if this is already in the FAQ, but couldn't find it:

Why is it sometimes called (with Ubuntu ISOs for example):

> MBR partition scheme for BIOS or UEFI

and sometimes called (with Windows ISOs for example):

> MBR partition scheme for BIOS or UEFI**_-CSM_**

?

pbatard, 2016-04-07:
First, please [see Wikipedia for the definition of CSM for UEFI](https://en.wikipedia.org/wiki/Unified_Extensible_Firmware_Interface#CSM_booting) (this is the first page you should get if you google `UEFI CSM`).

Once you know that, you should understand that UEFI-CSM means _"__non__-pure UEFI boot mode" (i.e. BIOS compatility mode), and that UEFI means _"pure UEFI"_ mode, and this is what Rufus tells you.

Therefore, if you see UEFI-CSM, then you should know that the drive will NOT boot in pure UEFI mode, but will need the CSM (a.k.a. BIOS compability) mode activated to be bootable on an UEFI platform.

If on the other hand you see UEFI, then it means your drive is bootable in pure UEFI, i.e. even if CSM is disabled on the target computer.

I know this can be confusing; which is precisely why I decided to use the official terminology, so that, hopefully, a quick google search is all it takes to understand what the option does.

I will close this issue, but if you have further questions, feel free to comment. And I'll consider adding some note in the FAQ, if I get evidence that many people get confused by this.

closed 2016-04-07 by pbatard

kq01526, 2016-04-07:
@pbatard :

I know what CSM means.

But I was of the impression, that when I select the first option, it will **_only_** work with BIOS (and/or UEFI CSM).

Especially after having read that Rufus is designed to only work in one mode at once and not both at the same time, see:

https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode

But it seems like that is only true for Windows ISOs (i.e. not Linux ISOs)?

pbatard, 2016-04-07:
> I was of the impression, that when I select the first option, it will only work with BIOS (and/or UEFI CSM).
> (...)
> But it seems like that is only true for Windows ISOs (i.e. not Linux ISOs)?

Well, if you use a Linux ISO (that has UEFI support), Rufus will display _"for BIOS or UEFI"_ and NOT _"for BIOS or UEFI-CSM"_, as it does with Windows ISO, which is its way of telling you that your drive can be booted in both BIOS/UEFI-CSM AND pure UEFI mode.

So, no, depending on the ISO you select, the first option can work for more than BIOS (and/or UEFI CSM), which is __precisely__ why Rufus will alter it to say either "UEFI" or "UEFI-CSM", depending.

Now, it's a bit more complex than that, as there are UEFI firmwares out there that include NTFS drivers, in which case you will be able to install a drive that was creates from a Windows ISO in both BIOS and __pure__ UEFI (even though Rufus said UEFI-CSM). But few of these firmware exist for now, and if I were to go the pedantic route, I'd have to write pages of text to indicate the various subtleties of boot compatibility, which would be a lot more confusing for most people than helpful.

Please understand that the FAQ entry you pointed is only for Windows ISOs, and it doesn't say anything about Linux.

kq01526, 2016-04-07:
@pbatard :

May I ask why this:

> This is done to avoid confusion, as it can be difficult for non expert users to know whether they actually booted in UEFI or BIOS mode, when an USB Flash Drive can be booted in both modes, and installation is meant to be a one-off operation, targeting a very specific machine and boot mode. You probably don't want to go through a full Windows installation, only to realize that it was installed in BIOS mode, when you really wanted it installed in UEFI mode. 

only applies to Windows but not to Linux?

Why are you doing it for Windows but not for Linux?

Doesn't make sense to me (no offense).

Why not do it for Linux as well?

Regards

pbatard, 2016-04-07:
> Why are you doing it for Windows but not for Linux?

Because I consider that Linux users are a lot more likely to understand that they can boot something in two completely different modes (BIOS|UEFI-CSM and "pure" UEFI) than Windows ones. Linux users are more tech-savvy, plain and simple, and tech-savvy users are less likely to complain that they installed their OS in BIOS mode when they wanted to install it in pure UEFI mode, because they will pay attention to whether their selected USB boot option is prefixed with `UEFI` or not.

kq01526, 2016-04-07:
@pbatard I kinda get what you're saying, but, sorry, no, it doesn't make sense to me to do it that way.

I think it is wrong to do it like this.

I think you should make the first option the same for everyone (Linux + Windows), i.e. either make it BIOS/CSM for everyone or make it BIOS+UEFI for everyone.

I think I would lean towards making it BIOS/CSM for everyone (i.e. doing the same for Linux what you are currently doing for Windows). If you would do that [and hide or remove the "MBR partition scheme for UEFI option"](https://github.com/pbatard/rufus/issues/732), that would simplify things quite a bit.

kq01526, 2016-04-07:
PS, by the way:

> Because I consider that Linux users are a lot more likely to understand that they can boot something in two completely different modes (BIOS|UEFI-CSM and "pure" UEFI) than Windows ones.

Let's face it: Those Windows users most probably would not use Rufus in the first place anyway, they will much more likely use the new Media Creation Tool from Microsoft :wink:.

pbatard, 2016-04-07:
> I think it is wrong to do it like this.

I disagree. Also, Rufus is pretty popular, so, yes, a lot of non tech-savvy Windows users will be using it. My inbox proves that (and usually tends to validate the design choices I make).

If you want Rufus to behave differently, feel free to create your own version.