pbatard/rufus #732: Make "MBR partition scheme for UEFI" only visible when needed?

kq01526, 2016-04-07:
Hi @pbatard ,

as far as I understand from the [FAQ](https://github.com/pbatard/rufus/wiki/FAQ#I_used_an_UEFI_bootable_ISO_based_on_grub_but_all_I_get_is_the_grub_prompt), the "MBR partition scheme for UEFI" option is only there as a workaround for things like those:

https://github.com/pbatard/rufus/wiki/FAQ#I_used_an_UEFI_bootable_ISO_based_on_grub_but_all_I_get_is_the_grub_prompt

correct?

For all other situations the "GPT partition scheme for UEFI" should be used (in case you want UEFI and not BIOS), right?

Then couldn't you make Rufus detect if "MBR partition scheme for UEFI" is needed and make it hidden when it is not? Or maybe even hide it under the advanced options?

Regards

pbatard, 2016-04-07:
> as far as I understand from the FAQ, the "MBR partition scheme for UEFI" option is only there as a workaround for things like this, correct?

No, that's incorrect. If you hover your mouse pointer on that option, you will see the tooltip (how I wish people realized that there are helpful tooltips in Rufus!):

> Use this if you want to install an Operating System in native EFI mode, but will need to access the USB content from Windows XP"

The problem is that GPT is 100% incompatible with XP, but some people might still need to access their flash drive from XP (which is not something Rufus can guess), so this option was created for them.

Oh, and by the way, that issue with GRUB, is really for #531, which Debian have now fixed in their latest ISO.

closed 2016-04-07 by pbatard

kq01526, 2016-04-07:
> @pbatard wrote:
> 
> The problem is that GPT is 100% incompatible with XP, but some people might still need to access their flash drive from XP (which is not something Rufus can guess), so this option was created for them.

One more reason to remove it from the standard options and hide it under the advanced options.

Support for Windows XP is long gone, so Rufus does not need to actively support it either (IMHO).

> Oh, and by the way, that issue with GRUB, is really for #531, which Debian have now fixed in their latest ISO.

Well, at this point I really don't see why the option should remain under the standard options. Could you please hide it in the advanced options then?

That would also make Rufus simpler: There would only be two options left - one for BIOS/CSM and one for UEFI.

pbatard, 2016-04-07:
> Support for Windows XP is long gone

Go tell that to people who are encroached in their XP usage. [I've been having long heated argument with peole trying to defend the use of XP](http://reboot.pro/topic/20358-rufus-20-has-been-released/?p=197736).

I am of course considering removing it, but I don't see how exactly this option is harmful. If your approach to creating a bootable drive is that the application should hold your hand from start to finish, and that you don't have to become a bit knowledgeable about what is actually a complete mine field, you're going to be disappointed.

As long as there's a tooltip, that clearly indicates when you should use the option, I really don't see much harm in keeping it, even more so as it can prove useful when Linux ditro maintainers screw up their GRUB builds. Debian may have fixed it, but I'm pretty sure that there exist other distros out there who have the same issue, so I'd rather still be in a position to help them.

kq01526, 2016-04-07:
> @pbatard wrote:
> 
> how I wish people realized that there are helpful tooltips in Rufus!

Well, I think I have an idea how you could easily realize that:

1. I don't think it is good that the tooltip only describes the option that is currently selected. It should describe all three options simultaneously in one tooltip, regardless of what is selected.
2. I don't think it is good that the tooltip only pops up when hovering over the drop-down list. Maybe it should also pop up when hovering over the heading (i.e. the text that says "Partition scheme and target system type")
3. Alternatively and/or additionally to 2. you could add a little colored question mark (?) button next to the heading, which, when clicked, makes the tooltip pop up

What do you think?

pbatard, 2016-04-07:
> I don't think it is good that the tooltip only describes the option that is currently selected.

I considered the alternative, but decided against.

> I don't think it is good that the tooltip only pops up when hovering over the drop-down list

As long as it describes the option, I think it makes sense.

> Alternatively and/or additionally to 2. you could add a little colored question mark (?) button

But then why stop there? I have other tooltips, and I'd rather avoid adding rather ugly question marks all over the place. I don't think it's my place to compensate for people who need an visual cue to let them know that they can get help, especially as I have more important improvements lined up.

IMO, if someone does pay attention to what the Rufus UI says, they can understand what is going on (and there's nothing in the UEFI specs that says that pure UEFI should only use GPT - as a matter of fact I could drop the GPT for UEFI option and leave only MBR for UEFI, and the only people annoyed would be those who use Rufus to create a bootable drive for a > 4TB drive... which I don't think are many).

So, once again, if you're not happy with a very deliberate decision from the developer of the app, feel free to create your own version.