pbatard/rufus #733: Rufus won't use my usb

Araschon, 2016-04-09:
I have my 2.0 8gb usb (don't know what kind it is) plugged in to usb 2.0 on the PC, but it won't show on rufus. Here is the log.

Found non-USB removable device 'WDC WD10EZEX-00BN5A0'
Device eliminated because it contains a mounted partition that is set as non-removable
0 devices found

I looked in device manager and my USB shows, but only in other devices. What do I do? I don't know what that log means either

pbatard, 2016-04-09:
> `Found non-USB removable device 'WDC WD10EZEX-00BN5A0'`

This is an internal Western Digital 1 TB HDD. It has nothing to do with your 8GB USB drive...

> I looked in device manager and my USB shows, but only in other devices.

That's the problem then. Unless your 8 GB device is seen as an USB one in device manager, then Rufus will be unable to list it. You need to see something like `USB Mass Storage Device` or similar under the USB category for your device to be listed.

I'm not sure why your USB device is listed in other devices. This could be due to a hardware issue or something else, such as your device using a custom encryption mechanism? What is the manufacturer and exact model name of your device?

Still, I would encourage you to post on Microsoft forums, as this is the first time I get such a report, and clearly, your issue appears to be external to Rufus.

You may try <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus, to get more verbose information about the device detection process in Rufus, and then post the log you get here, but I doubt it will provide more information.

Araschon, 2016-04-09:
I somehow fixed it by uninstalling my usb device driver then letting it reinstall. It is now properly listed in portable devices as "ESD-USB". Rufus recognizes it now, thanks though! (:

pbatard, 2016-04-09:
OK. I'll close this issue.

closed 2016-04-09 by pbatard

Exsilon, 2017-08-04:
Help im very new at this i downloaded rufus and no usbs showed up this was the log information 

Rufus version: 2.2.668
Windows version: Windows 10 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'UDisk2.0 Gigastone USB Device' (????:????) [ID]
Removing E: from the list: This is the disk from which Rufus is running!
0 devices found
0 devices found
Found USB device 'UDisk2.0 Gigastone USB Device' (????:????) [ID]
Removing E: from the list: This is the disk from which Rufus is running!
0 devices found