pbatard/rufus #740: Possibility to save options?

dhry, 2016-04-25:
Is there a way to save options in Rufus, or start with specific options checked/unchecked? For example, each time it starts, **Quick Format** and **Create a Bootable Disk** are checked, and I have to manually uncheck those before each operation. I would like to be able to start the GUI with those options automatically unchecked (via configuration file, registry or command line options, the method isn't important to me). Is this possible?

pbatard, 2016-04-25:
I'm going to reject this request. It would take a large number of people requesting this for me to even __consider__ it. First of all, UI changes such as personalized defaults settings are always a pain to handle, as you need to add a lot more logic than meet the eyes behind the scenes to make sure everything functions as expected. So it's a lot less problematic to start with an initial set of conditions that is always the same, as it avoids the odd behaviour that a lot of users complain about UIs that aren't responding in the expected fashion, even if it annoys a few users.
But more importantly, if I were to do what you suggest, I am positive that I would get quite a few reports from annoyed users who want to bootable settings and fast format reset to their default __always__. You may argue that this is not at all how you see it, and that user preferences should trump default settings, but I've been at it long enough to know what's in store when trying to please a __minority__ with something that is likely to annoy larger number of people.

Besides, disabling _Quick Format_ is generally a bad idea, even if initiated by the user. That's because doing full format regularly will greatly diminish the life expectancy of flash based media and doesn't offer any real advantage. I'd rather help users get maximum life of their media (especially ones who might not know better), even if it means going against what they they think they would prefer.

Finally I have to say that, if your prime usage for Rufus is not the creation of bootable media, there might be a better application for you to use.

I understand that this is not what you want to hear, but from where I stand, not granting your request is really the option that will actually benefit the largest number of Rufus users, and as such, I don't think I want to honour it.

Then again, Rufus is 100% Open Source, so you can always create your own version (or find someone who will), with the feature you request.

closed 2016-04-25 by pbatard

dhry, 2016-04-25:
It wasn't actually a feature request, it was simply a question, so the answer evidently is "no, it's not possible". FWIW I need the feature to clear out flash drives as I pass photographs I have taken of sports events to others and don't want undeletable traces of items remaining on the disks. I don't care about life expectancy because this is a 2gb flash drive and I have a box full of them. It's fine, I'll write an autohotkey script to automatically uncheck the options for me.