pbatard/rufus #741: Device could not open

RezaMathers, 2016-04-26:
According to the log on rufus 2.8 the error comes up as: 
"Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Could not open '\\?\usbstor#disk&ven_sandisk&prod_cruzer_blade&rev_1.26#4c530003120214100381&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found"
How do I sort this out??

pbatard, 2016-04-26:
__Something__ on your computer is preventing access to that USB drive, and that something is external to Rufus.

Unfortunately, it's not possible for me to tell what it is (Windows does not provide an API to indicate what may cause the denial of access), so you will have to figure it out yourself.

There's a possibility that this can be linked to using one of the application listed [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with), but it could be something else.

At any rate, the problem is external to Rufus.

RezaMathers, 2016-04-26:
Thanks. It could be that i am not the admin on this computer (it's my work laptop) but I will try on my own laptop and see how that goes. Again, thanks for the help.

pbatard, 2016-04-26:
You __must__ have administrative rights. Rufus can not work without.

closed 2016-04-26 by pbatard