pbatard/rufus #743: WTG BSOD: NTFS_FILE_SYSTEM

Michaelvas17, 2016-04-28:
Self explanatory.

pbatard, 2016-04-28:
> Self explanatory.

I think I'll need a little more:
1. Full log from Rufus of how you created your USB (please don't truncate the log). This will provide a lot of missing information I need to investigate your issue.
2. Is the target system UEFI or BIOS?
3. Have you tried on more than one PC?
4. Do you have an internal disk on that computer with Windows on it?

For the record, this is the very first report I get about such an error, and quite a few people (including myself) have been running WTG drives created with Rufus against both BIOS or UEFI machines, so I'd be leaning towards something specific to your target environment that we may have to identify. The more data you can provide on that, the better - thanks.

pbatard, 2016-04-28:
For the record, [this](https://msdn.microsoft.com/en-us/library/windows/hardware/ff557433%28v=vs.85%29.aspx?f=255&MSPPError=-2147217396) and [this](http://www.tomshardware.co.uk/answers/id-2090821/ntfs-file-system-blue-screen-death.html) could also indicate a hardware issue with the USB drive. Have you also tried with a different USB?

Michaelvas17, 2016-04-28:
Full log:
`Rufus version: 2.9.893 (Test3)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
0 devices found
0 devices found
Found USB 3.0 device 'USB Device' (13FE:5100)
Using autorun.inf label for drive D: 'UEFI_TEST'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3768, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {3D6E92D4-8CC3-4940-84A7-216F65AFCE22}
Max parts: 128, Start Offset: 17408, Usable = 30992857600 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {8332EA82-978E-40CC-B88F-21C62D902CDA}
  Size: 28.9 GB (30991826432 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CCSA_X64FRE_EN-US_DV5'
  Size: 4017000448 bytes
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1511_English_x64.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{f3260ade-8fcd-11e5-be82-24fd52450a36}\
Opened drive \\?\Volume{f3260ade-8fcd-11e5-be82-24fd52450a36} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{f3260ade-8fcd-11e5-be82-24fd52450a36}\ on D:\
Applying Windows image...
Windows To Go mode selected
[... 345 lines cut ...]
      4096 bytes in each allocation unit.
   7566103 total allocation units on disk.
   5616061 allocation units available on disk.
NTFS Fixup completed.

Found USB 3.0 device 'USB Device' (13FE:5100)
Using autorun.inf label for drive D: 'WindowsToGo'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3768, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0D970FE4
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.9 GB (30990758400 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 60530873, Boot: No, Recognized: Yes`

Target machine is UEFI
USB was able to be booted on a machine with a legacy BIOS.
Both machines have drives with Windows preinstalled

Michaelvas17, 2016-04-28:
@pbatard I'm going to reformat this with TEST4 using the bad blocks check.
Personally, I think it may be bad blocks.

Michaelvas17, 2016-04-29:
I believe that the issue was due to bad blocks, as I reformatted and created the WTG drive again. It works now.

closed 2016-04-29 by Michaelvas17

pbatard, 2016-04-29:
> I believe that the issue was due to bad blocks

Looks like it. Thanks for the update.