pbatard/rufus #744: My drive doesn't seem compatible with rufus

Hienkel, 2016-04-30:
[rufustest.txt](https://github.com/pbatard/rufus/files/243371/rufustest.txt)

I don't know if it is my lack of knowledge or if it's rufus

pbatard, 2016-04-30:
> `Unable to create file: [0x00000015] The device is not ready.`

This seems to indicate a hardware issue or possibly that there is an application on your computer that interferes with Rufus (such as one of those listed [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)). You may want to run a bad blocks check.

A `device is not ready` error is external to Rufus, so there isn't much I can do about it...

pbatard, 2016-05-06:
As I don't believe there is anything more that can be done, I will close this issue.

closed 2016-05-06 by pbatard