pbatard/rufus #745: Rufus Version 2.8 (Build 886) Creates exFAT drives w/ Cluster Size TOO SMALL

Simon5577, 2016-05-01:
When I tried formatting a 64GB USB thumb-drive using the File system - exFAT, the selection drop-down menu doesn't show a default Cluster Size. No default Cluster size is present in the selection drop-down menu. Only 512 bytes is selected automatically as it is the 1st item in drop-down list.
512 bytes is WAY TOO SMALL. Their is SIGINIFICANT performance issues during copying operations,

According to Microsoft at: https://support.microsoft.com/en-us/kb/140365

Default cluster sizes for exFAT is:

```
Volume size of: 7 MB–256 MB =====> 4 KB Cluster size
Volume size of: 256 MB–32 GB =====> 32 KB Cluster size
Volume size of: 32 GB–256 TB =====> 128 KB Cluster size
Volume size of: > 256 TB =====> Not supported
```

For my thumb-drive, the drop-down menu should have shown and selected:

`128 kilobytes (Default)`

Thanks for creating such fine software!

pbatard, 2016-05-01:
I'll look into it. Thanks for the report.

pbatard, 2016-05-02:
There was a typo in the exFAT default cluster selection for >32GB drives.
Instead of being set to 128KB it was set to 28KB, which of course didn't match any entry from the dropdown, therefore the default was never set. This will be fixed in the next version - thanks again for reporting the issue.

commit b6c0dd1b (2016-05-02): [core] fix a typo in exFAT cluster selection for >32GB drives

* Closes #745
closed 2016-05-02 by pbatard