pbatard/rufus #746: Rufus Stalled Copying ISO files

AutoKorrect, 2016-05-03:
Rufus version 2.8.886 on Windows 10.
Everything went fine until it got to copying ISO files... D:/sources/install.wim (3.6GB). Now it's just been sitting there for an hour. I've selected file system as Large FAT32 (default) instead of standard FAT32 not sure if that makes any difference. I'm using the official MS Windows 10 ISO downloaded from their website.

pbatard, 2016-05-03:
A log would be nice... Please click on the 'Log' button and paste it there.
However, by the looks of it, this seems like an issue with your USB drive. Flash memory is not everlasting, and a freezout is part of what might be expected when it is starting to become defective.

AutoKorrect, 2016-05-03:
I checked the log it was all normal until it got to the install.wim it just froze there. How long does this take normally to extract from ISO and copy on to the USB considering it's 3.6 GB? Should i have waited longer than 2 hours?

pbatard, 2016-05-03:
Please post the log. There is some information about the actual speed of your USB and possibly other things that may explain why it's taking so long. For instance, if your USB flash drive fell back to USB 1.0 speed, it could take more than 5 hours to copy the data. However, you should be able to see the progress bar change by leaving your cursor on it's current position, and then checking back a few minutes later. Have you seen any progress bar change over the last couple hours?

The log will have the actual USB speed being used, which is __one__ of the reasons I want to see it. Please understand that the reason I ask for the log is to try to help you out by getting information that I can not guess otherwise (and that may not strike you as significant, but might be crucial to explaining your issue). If you withhold the log, I can't help you.

AutoKorrect, 2016-05-03:
OK i'll restart the process and post the logs when it hangs again.

Uh oh. I interrupted the Rufus process and now my USB drive is broken. No longer can be detected by Mac or Windows. And i've tried reformatting it through Windows and Mac. 

I think it's a paperweight now...

pbatard, 2016-05-03:
> No longer can be detected by Mac or Windows.

Yes. The most logical explanation is that the flash memory on your drive failed (remember, flash memory is not everlasting and does wear out a lot faster than people think, especially with cheap flash drives), which would explain why your copy stalled.

The other thing I was about to suggest was a bad blocks check from Rufus, but if it can't be detected, there's not much point, and you already have the answer that the drive is defective...

AutoKorrect, 2016-05-03:
The device was still being detected in Device Management so i found this article that helped me to force reformat the drive. http://www.makeuseof.com/tag/external-drive-not-recognized-this-is-how-to-fix-it-in-windows/ 

That seemed to have worked so i'm back to restarting the rufus process. I'll post the log here when it hangs.

pbatard, 2016-05-07:
I'm gonna close this issue now, as there isn't anything else I can do. Feel free to report what happened with your new USB drive.

closed 2016-05-07 by pbatard

takuhii, 2017-04-25:
Hi, I know this issue is "closed" but I am having the same problem. It seems to happen when you let the display go to sleep. I can't include a log, because I have kept my display awake throughout the whole process and been unable to replicate it :/

pbatard, 2017-04-25:
> It seems to happen when you let the display go to sleep.

Are you sure it's not the whole computer that goes to sleep and not the display. There's nothing in Rufus that checks for components to go to sleep, so there's really not much to fix, and I can't guarantee that the USB will be created successfully if you set your computer to go to sleep before the creation process is complete.

> I can't include a log

Why? If it happens when the display goes to sleep mode, then once you move the mouse, you should have access to the Rufus UI and be able to copy the log. The only reason where you shouldn't be able to copy/paste from the log is if Rufus crashed, but that's not the case. Also if you can't replicate the issue then it's not something that I can investigate.

vanarebane, 2018-09-04:
I'm also having this problem. With Windows_Server_2016_Datacenter_EVAL_en-us_14393_refresh.ISO
Here is the log: https://www.dropbox.com/s/2nzudblr562qw64/rufus.log?dl=1

pbatard, 2018-09-04:
Please provide the SHA-1 for your `Windows_Server_2016_Datacenter_EVAL_en-us_14393_refresh.iso`. You can compute the SHA-1 in Rufus after selecting the image and clicking the `(✓)` button.

pbatard, 2018-09-04:
Not seeing any issue with `Windows_Server_2016_Datacenter_EVAL_en-us_14393_refresh.ISO` (SHA-1 = `772700802951b36c8cb26a61c040b9a8dc3816a3`). I suspect you probably just thought the process was frozen because your drive is slow, and flushing a 5.6 GB file to a slow USB drive may take some time. At any rate, the issue you are allegedly seeing is 100% on your side and not with Rufus. Here's the log confirming that I was able to create a bootable USB from that ISO without issues:
```
Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive H: '16 GB'
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {BE9B1738-0AAF-4E94-8448-29AF24C4947A}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {BFDE9654-A598-4E0D-B16C-9C0EEF190618}
  Size: 14.9 GB (16011781632 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {DF6226BD-E151-45DE-8F16-2BF0FFAA6CBE}
  Size: 512 KB (524288 bytes)
  Start Sector: 31275059, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'SSS_X64FREE_EN-US_DV9'
  Size: 6.5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Windows_Server_2016_Datacenter_EVAL_en-us_14393_refresh.ISO (6.5 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE6 for exclusive write access
Requesting lock...
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{05aceaab-b02a-11e8-971c-fcaa14e7bd8e}\
Volume is already mounted, but as G: instead of H: - Unmounting...
Disabling file indexing...
[... 1162 lines cut ...]
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {C43CC75E-A401-474D-B679-7D3193E08DE0}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {363D2CD7-08F7-42C4-B9C0-E827E8DE6C56}
  Size: 14.9 GB (16011781632 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {3DC6D2D4-3F60-45F8-9A61-0E91C04A8197}
  Size: 512 KB (524288 bytes)
  Start Sector: 31275059, Attributes: 0x0000000000000000

Computing checksum for 'D:\Incoming\Windows_Server_2016_Datacenter_EVAL_en-us_14393_refresh.ISO'...
  MD5:	 70721288bbcdfe3239d8f8c0fae55f1f
  SHA1:	 772700802951b36c8cb26a61c040b9a8dc3816a3
  SHA256: 1ce702a578a3cb1ac3d14873980838590f06d5b7101c5daaccbac9d73f1fb50f
```

vanarebane, 2018-09-05:
You are right, the issue was with huge file, it just took over 10 min to copy 5.6GB.
Issue here is that there is no feedback on large file copies. As we can see here, users will assume it's frozen if it's not giving any signs of progress.

pbatard, 2018-09-05:
There is a progress bar. There are only so many progress indicators I am going to add to an application. During my test, the progress bar was more than enough to indicate that the file copy wasn't stalled.

vanarebane, 2018-09-05:
Do you wish to see a video when I copy the "Windows_Server_2016_Datacenter_EVAL_en-us_14393_refresh.ISO"? Maybe you can understand better what I meant as my describing skill isn't that great.

pbatard, 2018-09-05:
I __have__ tested the issue, with __decent__ hardware, and I have not observed any freezout. Therefore, as I indicated, the problem is with your environment. With decent hardware, you will not observe the long stall you report, and, as I explained in the other issue, you have to understand that there is no extra _"magical"_ progress indicator that we can use to report progress besides the bytes that Windows reports as copied which is what we use.

You __are__ asking for something that cannot be done. The problem is not, as you seem to assume, that we forgot to report the progress in Rufus, or that we are using an inefficient progress indicator. The problem is that we are already using the best progress indicator that Windows gives us and that, if you have slow hardware, then that progress will stall due to your hardware, and your hardware alone.

vanarebane, 2018-09-05:
Wow, no need to get personal here, man! You don't know what hardware I'm using and you just assume the issue is due to some crappy hardware? I believe that USB 3 speeds are quite "decent" or what hardware you got there? 

Like I said, if many people are experiencing this and reporting it here, there is something to it. I will record a video of the progress bar and you can maybe understand the problem a bit better. So far I see you just assume the user is dumb and you got nothing wrong on your end.

pbatard, 2018-09-05:
I'm not being personal. I'm just reporting that with decent hardware, I am not observing the stall you report, which logically means that the problem is with your environment and most likely your hardware.

I tested with a SanDisk USB 3.0 Extreme 16 GB which, despite its "Extreme" name, doesn't have a very "extreme" write speed compared to today's standard: 50 MB/s is no longer that high for USB 3.0.

Now, if the flash drive you are using has a write speed that is less than 50 MB/s, then, yes, it __is__ slow hardware for 2018, and if you are serious about transferring large images, you will get your hands on better hardware. For instance, considering that the log reports that you are using a USB &harr; microSD adapter, have you tried using an actual flash drive rather than an SD card? What's the write speed rating for your SD card? And are you sure you adapter can sustain that speed?

> if many people are experiencing this and reporting it here, there is something to it.

Emphasis on "if", which is __not__ what I am seeing (3 issues in 2 years, with only one person, you, posting a log that allows me to see what kind of hardware they use). Therefore, so far, I would say that there's not much to it if, when Rufus is downloaded more than 3 million times each month, I only get 3 reports of progress stall in 2 years...

> So far I see you just assume the user is dumb and you got nothing wrong on your end.

Not at all. I'm only stating that if people refuse to acknowledge that if tests carried out with __decent__ hardware don't show stalling, then they shouldn't be __assuming__ that there's a bug to fix in an application, and instead take a closer look at the hardware they use...

You are trying to overgeneralize an issue that exceedingly few people seem to have, and that can logically be dismissed as a hardware/environmental problem. Therefore, you will need to come with a better argument than _"But I did observe it on this specific hardware, that I am completely ignoring the performance aspect of!"_. Now if you tested with a handful of USB 3.0 drives, that are rated to have a >50 MB/s speed (which is what I would call decent in 2018) that would be a different story. But I don't think you are willing to do that.

vanarebane, 2018-09-05:
Right, so no point in investing time if the problem impacts only 0.00001% of users. Point taken.

One last thing. I had this idea - what if you switch off the buffering on the USB device prior to copy? And you can flip it back on, if it was on before.
There seems to be some shell commands that enable this:
https://gallery.technet.microsoft.com/scriptcenter/Change-write-caching-576747ad
But I'm a JS Front-End dev, so I don't know much about windows platform

pbatard, 2018-09-05:
What you are proposing is disabling cache, and I already mentioned the impact this would have in the other issue when I said:

> without disabling cache altogether, which would kill write performance for most people

Disabling cache altogether is a bad idea. Sure, if you don't cache at all, then your progress reports are accurate. But then you take a __major__ performance hit when it comes to writing data, which is not acceptable. Besides, Windows already compromises in terms of caching to allow for fast removal of USB drives. This is why you will see that, for most USB drives, the global write caching setting is already disabled as per:
[image]