pbatard/rufus #747: Rufus won't start if a usb key with boot-repair-disk is inserted

butla, 2016-05-04:
Had this strange behavior:
- Got Rufus 2.8p and turned it on.
- Said 'No' when it asked me if it can download updates.
- Nothing happened (no window).
- Tried to run again - it couldn't because it was still running.
- Found Rufus in task manager and killed it (took him about 5-10 seconds to shut down).
- Did a few repeats of this run/kill cycle.
- Pulled out USB key that was in the whole time.
- Run Rufus again - it worked!
- Put the USB stick back in and Rufus detected it.

The usb key contained the ISO from https://sourceforge.net/projects/boot-repair-cd/ (copied to the USB using "dd" under Linux).
I'm on Lenovo z51-70 with Windows 8.1 Pro x64 (English version with Polish locale).

Hope this helps in debugging :)

pbatard, 2016-05-04:
Can I please ask you to install and run [DebugView](https://technet.microsoft.com/en-us/sysinternals/debugview.aspx). Rufus sends its log messages here so it may indicate why it fails to start.

butla, 2016-05-04:
I'm formatting said USB drive with Rufus at the moment, but I'll try to reproduce it later.

pbatard, 2016-05-04:
Note, I tested writing `boot-repair-disk-64bit.iso` to a flash drive using `dd` on Linux, but found no issue when starting Rufus 2.8 when the key was plugged. I should mention that `boot-repair-disk-64bit.iso` is NOT an ISOHybrid, so it doesn't make much sense to use `dd` to copy it to a flash drive, but even then, there appears to be an environmental component to your problem (i.e. the problem seems to be specific to your machine + Software installed + USB hardware, etc.)...

butla, 2016-05-04:
Damn, I can't make it happen again.

It's notable that since the time that I had the issue, I've had to reboot a few times and Windows downloaded some updates. My Windows is still a relatively fresh install.

But I guess that you can close this as non-reproducible.

pbatard, 2016-05-04:
OK. I'll keep my eyes open for people reporting similar issues. But unless they can be consistently reproduced, it'll be hard to troubleshoot.

closed 2016-05-04 by pbatard