pbatard/rufus #748: rufus can destroy flashdrive

user15648, 2016-05-07:
I didn't set options correctly at first run and maybe .iso-file was broken.
but now i have completely dead usb-stick. windows cant mount it. rufus says:

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3825, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 29.3 GB (31466323968 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

if i try to remake or simply format rufus says:

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'E:' as volume mountpoint
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Отказано в доступе.
  Wrote: 0, Expected: 512
  StartSector: 0x00000001, nSectors: 0x1, SectorSize: 0x200
unable to zero MBR/GPT

user15648, 2016-05-07:
what can i do now?

pbatard, 2016-05-07:
1. According to what you post Your drive is NOT dead. Its partitions have just been cleared (and therefore there are no partition to mount in Windows). Please be mindful about using __misleading__ titles such as "rufus can destroy flashdrive" when your drive is still detected. Your drive is NOT destroyed at all!
2. `Access denied` means there is __ANOTHER__ application that is preventing access to your drive. You will have to identify it. Unfortunately, Windows makes it impossible to tell which one it is, so it will be up to you to find it. It may be an antivirus or one of the applications that are listed at the end of [the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)

pbatard, 2016-05-13:
See also the [relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied). I'll be closing this issue now.

closed 2016-05-13 by pbatard