pbatard/rufus #752: Syslinux install crashing on bootable disk creation

Thepenguin9, 2016-05-10:
I am attempting to install the Ubuntu 16.04 LTS amd64 ISO to a 64GB flashdrive, and rufus crashes whenever it attempts to install syslinux. Due to rufus crashing I am unable to provide a log, however I can attempt to compile rufus and debug it to provide one.

This is a log before the install process:

```
Rufus version: 2.8.886
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Removing F: from the list: This is the disk from which Rufus is running!
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7651, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x13C9C42E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 58.6 GB (62931599360 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.04 LTS amd64'
  Size: 1485881344 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04-desktop-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation
```

pbatard, 2016-05-10:
Can you install and run [DebugView](https://technet.microsoft.com/en-us/sysinternals/debugview.aspx). Rufus sends its log there as well, so it could explain why it crashed. Also, can you rename the `rufus_files` directory to something else and retry so that the Syslinux files are downloaded again. There used to be a crash (that should have been fixed) when the downloaded Syslinux files were corrupted (e.g. due to a firewall), so I'm wondering if this might be a similar thing. If it still crashes, you may want to have a look at the files you have under `rufus_files` with an hex editor, to find out if your firewall simply didn't replace their content. With some firewalls you may get the data replaced with _"You are not authorized to access this file"_ or something, which would make it quite obvious.

Thepenguin9, 2016-05-10:
Just checked the files in a text editor, they both contain the blocked page for the network I'm on, yet I can access the ftp server?

Running DebugView gives this: 
```
[10296] Found volume GUID \\?\Volume{cff8d97e-c9d0-11e5-bcc9-0090f5d76b76}\ 
[10296] Installing Syslinux 6.03... 
[10296] Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys' 
[10296] Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' 
[10296] Successfully wrote 'ldlinux.sys' 
[4320] Suspending
[10296] Opened drive \\?\Volume{cff8d97e-c9d0-11e5-bcc9-0090f5d76b76} for write access
```
with the log from rufus ending there

Thepenguin9, 2016-05-10:
The cause is that the syslinux file is corrupt thanks to my firewall, but I cannot find the appropriate version of syslinux on the FTP server

pbatard, 2016-05-10:
> Just checked the files in a text editor, they both contain the blocked page for the network I'm on

Haha! Can you please e-mail me those files to pete@akeo.ie? I thought I had fixed the crash due to firewall modifications of the syslinux files, but it looks like that's not the case for your specific files. If I can get them, I should be able to replicate the crash and fix it.

> yet I can access the ftp server?

Most firewalls will allow browsers and other whitelisted applications to download files, but will block anything else. This is why you may think everything is fine when you use your browser, but Rufus will still have an issue.

There's not much I can do about the firewall issue (even if I fixed the crash, I still won't be able to make Rufus download the files) but all you should have to if you want things to work right now is download both `ldlinux.sys` and `ldlinux.bss` from http://rufus.akeo.ie/files/syslinux-6.03/release/, using your browser (this is the version you need), and replace the ones you have in `rufus_files`.

Thepenguin9, 2016-05-10:
Both of the files contain the same page sans what each file was linking to.

zipping and e-mailing now

trev0rbr, 2016-05-13:
@Thepenguin9 

Format using NTFS to correct the fault.

pbatard, 2016-05-13:
@Thepenguin9, I haven't received your files - can you please e-mail them again?

pbatard, 2016-05-16:
@Thepenguin9, I could really use the files to fix the issue... Without them, this will be very difficult, and it will lead to __other__ people ending up in the exact same situation as you did.

pbatard, 2016-05-16:
Forget it. After spending time trying to come up with some altered files, I'm pretty sure I have a good idea of what happened and how to fix it now.

closed 2016-05-16 by pbatard

Thepenguin9, 2016-05-19:
Hey, sorry for the noreply and the file no-show  my email is bugged. Glad to hear you fixed it though

commit 1a24e5bb (2016-05-16): [syslinux] fix a typo that reintroduced a possible crash

* a10a207790a27b372d01f580d93f0fa937128c33 reintroduced the crash that was fixed in ad68ccfac99d62ccd19b47f123614b30f3c09965
  due to the missing semicolon.
* Also fix a minor output message issue
* Closes #752