pbatard/rufus #753: rufus will change the UAC setting on win 10 build 14295 after running it.

jaynfs, 2016-05-11:
I set the UAC to never notify, rufus will change the UAC from never notify to always notify after running rufus.

pbatard, 2016-05-11:
Unless there's a __glaring major security bug__ in Windows 10 14295, in which case this is 100% a __pure__ Microsoft issue, Rufus can NOT alter global UAC settings.

So if you are using Windows preview and the behaviour you experience is that, if you set your global user UAC to "Never notify" (using the slider on user account options) and, if, after you launch Rufus, the slider has changed position to something else than "Never notify", then you need to contact Microsoft so that they fix what is a pure Microsoft issue.

There's nothing I can do in Rufus about this, as Rufus doesn't do anything with UAC settings apart from issuing the standard UAC request for elevation, and, unless there is a bug in the Microsoft API, that request does not change the user settings.

Closing as: not a Rufus issue.

closed 2016-05-11 by pbatard

jaynfs, 2016-05-13:
Thnaks for you reply!