pbatard/rufus #756: problem with finding USB device, while other tools have no problem with the same

itsbrody, 2016-05-13:
While first time trying to use rufus, the tool does not cache by connected USB device. Don't know where the problem is, because it is found but cannot bound by rufus, while operating system and other applications bind it as usually.

Attached logfile as zip (as needed from github)

pbatard, 2016-05-13:
Please see [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected).

Rufus is __designed__ not to list USB HDDs or internal HDDs by default, to avoid the chance of formatting a drive by mistake.

itsbrody, 2016-05-13:
its not an HDD, but a normal stick (2GB)

itsbrody, 2016-05-13:
wanted to provide the log to you, but github does not accept .log or .zip from me; any idea?

pbatard, 2016-05-13:
Then I need to see your log (which you didn't provide - please copy/paste it in github, even if it's long, don't try to attach the file).

pbatard, 2016-05-13:
Just open the file and copy the content in github - it's just text.

itsbrody, 2016-05-13:
Will attach below:

Rufus version: 2.8.886
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0407
Will use default UI locale 0x0407
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Could not open '\\?\usbstor#disk&ven_generic&prod_flash_disk&rev_8.07#92b29fbd&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Zugriff verweigert
A device was eliminated because it didn't report itself as a disk
0 devices found

Seeing "Zugriff verweigert, means 'access denied'", but the access is open, generally

pbatard, 2016-05-13:
Are you running Rufus as administrator? You __MUST__ have administrative access on your user account for Rufus to work.

itsbrody, 2016-05-13:
no, I'm not an admin. I'm running under a normal account for security reasons.

pbatard, 2016-05-13:
Then Rufus will simply not work. See [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_does_Rufus_need_elevated_privileges_to_run).

itsbrody, 2016-05-13:
sad, but thanks for your fast feedback. 
Because you provided a "portable" version and the normal tool did start and not complain about the need of admin rights, I guessed it would work.

As other tools do, too.

What are the differences or are there any differences between the portable and the normal version of rufus?

pbatard, 2016-05-13:
There's no difference between portable and normal, apart from the fact that one writes to the registry and the other to a `rufus.ini` file.

And I'll see what I can do to make it more explicit that admin rights are needed.

itsbrody, 2016-05-13:
I've downloaded several tools for writing USB sticks. Some are installers and then at least require admin rights to install. Some other complain on start about admin rights and some others work with normal rights also with the ability to format such a stick. Thus, I don't believe that admin rights are really necessary for the job.

pbatard, 2016-05-13:
> Thus, I don't believe that admin rights are really necessary for the job you provide with your tool.

You're welcome to write your own formatting tool then.

closed 2016-05-13 by pbatard