pbatard/rufus #757: ER: Check that current user account has admin rights

pbatard, 2016-05-13:
It looks that a lot of users still don't realize that they need admin privileges for Rufus to run properly (eg. #756). 

We may want to add a check to confirm the user does have admin rights.

itsbrody, 2016-05-14:
Want to let you know, that my last comment on #756 might be misunderstood from your side. Big sorry, if so - reason: I found rufus adverted on ubuntu's download page and had some pressure this day to get ubuntu on a stick, but don't have admin rights at work.
Today I tried rufus with "run as" admin rights and it worked very easy : found the right usb device, full format possible, install of ubuntu successful and the best. The USB sticks started ubuntu on several different computers with uefi, efi and standard bios - great work at all!

Further yesterday I understood that the "admin rights" for running are regarding security, e.g. that formatting the wrong disk can't be achieved. In this case I ask you to think about to secure by admin only features that are able to destroy something, e.g. the formatting of the "wrong" drive. IM(h)O finding possible right device(s) and writing (e.g. a downloaded ISO image) to the device I'd suggest to free from the need of admin. Would you think about it?

pbatard, 2016-05-14:
> I understood that the "admin rights" for running are regarding security, e.g. that formatting the wrong disk can't be achieved.

You're wrong. Once again, I'm afraid you're wishing for what you would like to hear, whilst ignoring what people who have more expertise are telling you. You simply cannot write an MBR (which is needed to make anything bootable from BIOS) or repartition a drive without admin rights on Windows. If you could, that would be a glaring security flaw from Microsoft, as any malicious app could just go around and destroy a disk.

> Would you think about it?

I did. And again, since you don't seem to be convinced, I __strongly__ invite you to spend the time to try to prove me wrong.

Besides, even if it was possible (which it isn't), I think I'd prefer not let users who don't have admin right reformat and repartition drives. If those users are being restricted privileged operations by their sysadmin, then said sysadmins probably don't want them erasing drives left and right, and I'd rather side with the admins on that one.

itsbrody, 2016-05-14:
Thank you for your feedback; I'm convinced that you are much more expert in these topics, but I also see what I got one time before knowing rufus: An uefi booting stick giving me the option to select to test the previously "copied" ubuntu 16.04 (from kubuntu-16.04-desktop-amd64.iso). standard bios start did not work - you provide both possibilies in a working manner.

Regarding having admin rights or not; this is due to we are a big company and has nothing to do with "knowledge". I had admin rights before we were integrated there in for several years and not anything bad or worsed happened with my equipment.

pbatard, 2016-05-14:
Yes, there are __very specific__ corner cases where one might be able to get away with not having admin rights. But I am not going to invest time, which I don't have, for features that will only be useful for a couple of users. Also, always repartitioning (and recreating the MBR as needed) makes Rufus a lot more reliable than the other apps (Remember the 'R' in Rufus stands for reliable), and I'm not going to compromise reliability.

Just consider this: Rufus is designed so that someone can insert a completely blank drive, and end up with a bootable drive once they're done. There is simply NO WAY you can achieve this without admin rights.

Other apps may try to take shortcuts and provide __limited__ boot functionality. But that will never be Rufus' approach. So, if you are unhappy about the current behaviour of Rufus with regards to requiring admin privileges, I will kindly ask you to you use another solution, because what you are asking for will never be provided by Rufus, ever.

itsbrody, 2016-05-15:
I fully understand your concepts and thoughts in the meantime. And the easyness of Rufus I now confirm since yesterday afternoon. 

Finally it then directs to #757 - this issue, to check that user account has admin rights and(!) clearly provide this information to the user on starting Rufus.

Tmp341, 2016-05-15:
You want that Rufus should show if the current user has admin rights or not. Am i right @itsbrody ?

itsbrody, 2016-05-15:
Yes,  I think that would be a good manner, when Rufus is missing the definitely needed admin rights to pop up a dialog with that information and not to start as it would work properly.

pbatard, 2016-05-15:
Well, the thing is, obviously, I __tested__ the whole thing on Windows 7 with a regular user account I created, and, because does embed a manifest with `requireAdministrator`, when I tested it, Rufus did refuse to launch because that user wasn't an admin. I even tried to played with UAC, but I could not get into a situation where Rufus would launch from a regular user account, which is exactly what I expect to happen from using `requireAdministrator` in the manifest, and which is also what Microsoft documents should happen.

However, it's not the first time someone reports that, despite this, they seem to have been able to launch Rufus from a regular user account without prompt, even though the application is designed not to let that happen.

So, I have no idea how you guys have configured your regular user accounts, but clearly what Microsoft says should be enough to prevent non admin users for ever running an app that __explicitly requests__ admin rights, is not enough and that's the point of this enhancement.

I am convinced that, for the vast majority of people, who have created their user account in the standard manner, the current `requireAdministrator` manifest mechanism will work, but this is done for the __minority__ of people who someone managed to configure their regular user account in a manner where it doesn't.

Now, if anyone can provide step by step details on __how exactly__ they were able to create a user account on Windows 7 (or other version), that is a regular user and yet will be able to launch Rufus without being prompted for admin rights, I'll be really interested!

itsbrody, 2016-05-15:
I have not created this account myself.
Its a domain account, which is held on an exchange server.

pbatard, 2016-05-15:
> Its a domain account, which is held on an exchange server.

Yeah, that's what I guessed. I'd still very much like to know how the heck some sysadmins seem to manage to create accounts where basic Windows mechanism such as `requireAdministrator` are completely ignored, as it completely goes against the basic behaviour that Windows should enforce. Maybe one day someone will be able to tell me...

closed 2016-05-26 by pbatard

pbatard, 2017-01-17:
Thanks to @Sopor- I finally have closure on how it is possible for a Windows environment to discard the `requireAdministrator` privilege, and therefore trigger the prompt that was added in [bf679271b842ab060a26ed34a4e150d19aee9abe]:
1. Open a command prompt in a non-elevated mode (in the same folder as the Rufus executable)
2. run: `set __COMPAT_LAYER=RUNASINVOKER`
3. Launch Rufus from the command prompt

There are probably other ways to duplicate what `set __COMPAT_LAYER=RUNASINVOKER` does, but it's nice to no longer be in the dark. 😄

itsbrody, 2017-01-18:
Hello Pete, 

Thanks for mentioning this to me. 
I tried with Rufus 2.11 without success. 

Regards 
Wolfgang 

Am 17. Januar 2017 17:13:08 MEZ schrieb Pete Batard <notifications@github.com>:
> Thanks to @Sopor- I finally have closure on how it is possible for a
> Windows environment to discard the `requireAdministrator` privilege,
> and therefore trigger the prompt that was added in [...]

itsbrody, 2017-01-18:
> My comment was for people looking at the issue at
> https://github.com/pbatard/rufus/issues/757#issuecomment-273517231
> rather than a reply to your initial request (which was to create a
> version of Rufus that works without administrative privileges). I have
> no plans to ever look into doing anything like that, so of course, none
> of the versions of Rufus you try will work in the manner you wish.

OK, understand your comment now the right way. Thanks for clarifying, 

Regards 
Wolfgang

commit bf679271 (2016-05-26): [misc] prevent application launch if running non elevated

* Since, despite what Microsoft states, having 'requireAdministrator' in a manifest STILL isn't
  enough to prevent an app from launching as non elevated for some weird account configurations...
* Closes #757