pbatard/rufus #760: Write does not complete and is not bootable

MrMoxy, 2016-05-19:
Rufus v2.9. Writing bootable windows 7 USB. USB memory key is 16G. Set to gfi partition scheme for uefi, fat32 and 8192. Writing proceeds normally and apparently fine but at the end it just returns to 'ready', never indicating 'done'. Subsequent attempts to boot from it produce a message that windows can not boot. Looking at it with file manage,r it looks fine (directories, etc.). But something is going wrong in the creation...

pbatard, 2016-05-19:
https://github.com/pbatard/rufus/wiki/FAQ#I_contacted_you_but_all_I_got_was_a_link_to_this__why

Tmp341, 2016-05-20:
Windows cannot boot itself without NTFS.

MrMoxy, 2016-05-20:
Thanks I'll try it when I rerun it to create a log as requested.

MrMoxy, 2016-05-20:
Ok, I have attached the rufus log and two screen shots (from my phone) showing the result of the subsequent attempted boot using the device. Rufus was run under Windows 10 Pro.

As an aside, I used this same USB key to install Windows 10 on this machine about two weeks. I created the Windows 10 Pro uefi boot under windows 7 Pro and with rufus. Clearly it worked fine since I used it to do the subsequent install.
[image]
[image]

pbatard, 2016-05-20:
I don't see the log. Please go to https://github.com/pbatard/rufus/issues/760 and copy/paste its content there, thank you.

Also, I hope you do realize that Windows 7 is missing A LOT of the drivers that enable the installation of Windows on a __modern__ system, which are provided natively with Windows 10 (there's actually some good reasons why Microsoft advises people to upgrade &mdash; Being able to perform an easy install on current hardware is one of them).

For instance, Windows 7 has NO SUPPORT whatsoever for USB 3.0. So if you're using an untouched Windows 7 ISO and trying to install from USB 3.0, that will never work (and you're likely to run into the issue you see above). However, this has nothing to do with Rufus...

pbatard, 2016-05-20:
PS: According to [google](https://www.google.ie/search?q=Windows+7+UEFI+failed+to+start+BCD+0xc000000d&ie=utf-8&oe=utf-8&client=firefox-b&gws_rd=cr&ei=mzg_V8ShCeubgAaZ37HICQ), a lot of people seem to be getting the error you experience due to incompatibility between Windows 7 UEFI and some hardware.

Unless you install Windows 10, you may be stuck with having to install Windows 7 in BIOS-compatibility mode... Please have a look at the posts from all the people who experienced your issue. Maybe you'll find some help there.

However, as far as I can tell, your issue is unlikely to have anything to do with Rufus.

MrMoxy, 2016-05-20:
Oops here is the log.
[rufuslog.txt](https://github.com/pbatard/rufus/files/275000/rufuslog.txt)

I am aware of the w7 usb issue, though intel have released drivers that apparently work. It is just that W10 is so lame in ergonomically...

pbatard, 2016-05-20:
Thanks for the log. This definitely does not look like a Rufus creation issue, but one due to the fact that Windows 7 is getting a bit too old to be compatible with modern hardware. As a matter of fact, even the UEFI support for Windows 7 x64 is a hack, as the Windows 7 ISOs are not natively compatible with UEFI boot. So I'm not entirely surprised that UEFI boot is causing issues there.

If you really want to install Windows 7, I would __strongly__ encourage you to do so in BIOS-compatibility mode, and not in UEFI mode. Therefore you should use "MBR for BIOS or UEFI-CSM" in Rufus when creating the drive and play with your UEFI firmware options so that it can boot in compatibility mode. Or else, you can complain to Microsoft that their Windows 7 ISOs don't work too great for installation in UEFI mode, but I'm sure you already know what they'll advise you to do... :wink:

pbatard, 2016-05-22:
As I don't think there's anything more I can do, since this is not actually a Rufus problem, but a Windows 7 hardware compatibility one, I will close this issue now.

closed 2016-05-22 by pbatard

MrMoxy, 2016-05-22:
Yep, you are right. Thanks for being so responsive and patient, Pete. 
And thanks for a great piece of software.