pbatard/rufus #763: List of upcoming translation updates

pbatard, 2016-05-27:
- [ ] `MSG_288` "Missing elevated privileges"
- [ ] `MSG_289` "This application can only run with elevated privileges"

424778940z, 2016-05-31:
raw Chinese
"缺少提升权限"
"这个程序只能在提升权限的情况下运行"
improved Chinese
"没有足够的权限" --> "no enough privilege"
"本程序需要提升权限才能运行" --> "this application need elevated privileges to run"

pbatard, 2016-05-31:
Hold on on translating these, please. 

I can not accept these changes yet, because this issue is mean to be __INTERNAL__ (so that I have a list of what's coming next) and translators should NOT do anything until the [Translation ChangeLog](https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt) has been updated.

Also, [this](https://github.com/pbatard/rufus/wiki/Localization#Editing_an_existing_translation) is the normal procedure for updating a translation, which starts by looking at the ChangeLog...

424778940z, 2016-05-31:
ok, I will read those files

pbatard, 2016-05-31:
Thanks.
 
If you want to be alerted of when there is new translation work needed for Rufus, could you please send me an e-mail to: pete@akeo.ie?

I will add you to my list of translators, and send you an e-mail when the ChangeLog is updated and am ready to add translations for the changes listed above.

closed 2017-01-16 by pbatard