pbatard/rufus #765: I have problem in making pendrive bootable in rufus 2.9

devilaltard, 2016-05-29:
Rufus giving me following log while causing problem in step "installing syslinux " : anyone help me please 
Rufus version: 2.9.934
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'SanDisk SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1927, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x088D6C30
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.8 GB (15853420544 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.04 LTS amd64'
  Size: 1485881344 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04-desktop-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{ca9f6a47-039d-11e6-bdc6-806e6f6e6963}\
Installing Syslinux 6.03...
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys'
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss'
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{ca9f6a47-039d-11e6-bdc6-806e6f6e6963} for write access
Could not patch Syslinux files.
WARNING: This could be caused by your firewall having modifed downloaded content, such as 'ldlinux.sys'...
[... 232 lines cut ...]
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{ca9f6a47-039d-11e6-bdc6-806e6f6e6963}\
Installing Syslinux 6.03...
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys'
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss'
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{ca9f6a47-039d-11e6-bdc6-806e6f6e6963} for write access
Could not patch Syslinux files.
WARNING: This could be caused by your firewall having modifed downloaded content, such as 'ldlinux.sys'...
\\?\Volume{ca9f6a47-039d-11e6-bdc6-806e6f6e6963}\ was already mounted as J:\
Re-mounted volume as 'J:' after error

Found USB 3.0 device 'SanDisk SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1927, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x008972C4
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15853420544 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-05-29:
```
Could not patch Syslinux files.
WARNING: This could be caused by your firewall having modifed downloaded content, such as 'ldlinux.sys'...
```

Please compare the `ldlinux.sys` you will find under the `rufus_files\` directory that Rufus created with the one from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys. I'm pretty sure that you will find that your current file was modified in some way (which usually happens if your firewall is set to only let a browser download content, but not applications like Rufus), which is why Syslinux can't be installed.

pbatard, 2016-05-30:
Any update on this? Did you compare the files?

pbatard, 2016-06-02:
I'm going to consider this issue as resolved.

closed 2016-06-02 by pbatard

Bibiprog, 2016-10-26:
I had the same issue because I create multiple bootable media, for different architectures. Removing the folder "rufus_files" resolved it. Rufus created a new folder with an operating syslinux.
Thanks