pbatard/rufus #766: A translation error of Chinese Simplified version

choleraehyq, 2016-05-31:
In the select box of "Partition scheme and target system type", the choice "MBR partition scheme for BIOS or UEFI" was translated into "用于 BIOS 或 MBR 计算机的 UEFI 分区方案", which is wrong. The correct translation should be "用于 BIOS 或 UEFI 计算机的 MBR 分区方案".

I'm a native speaker of simplified Chinese.

pbatard, 2016-05-31:
Thanks for the report.

I'm afraid I will need your help to fix this, as the the issue we have is that that message does not come from a static translation (Such as "用于 BIOS 或 UEFI 计算机的 MBR 分区方案"), which could be easily replaced, but from a translation that contains 2 __dynamic__ strings (["用于 BIOS 或 %s 计算机的 %s 分区方案"](https://github.com/pbatard/rufus/blob/2495f2d444855186a62e7950e19274fe6ffb02b0/res/localization/rufus.loc#L2080), where `%s` are the placeholders for the strings), and these strings have to be inserted exactly in the order: "MBR" then "UEFI".

In short, all of the translations should follow the English version, and apply the dynamic strings in the same order:
```
t MSG_031 "%s partition scheme for BIOS or %s"
```

Oh, and in case you wonder, the reason we have dynamic strings is that the first `%s` is set in an automated manner by the code, and could ultimately be something else than "MBR", and the second `%s` is set to either "UEFI" or "UEFI-CSM" depending on the circumstances. So we can't hardcode "MBR" or "UEFI" in the translation (and I don't really want to add a specific exception for Chinese, to insert the strings in reverse order, as this makes things more difficult to test and maintain).

So, my question to you is:

___Is there any way you think you could reword this translation so that the partition scheme ("MBR") appears first, and so that the target computer type ("UEFI" or "UEFI-CSM") appears last?___

Oh, and while I have a Simplified Chinese speaker around :smile:, and since it appears my current translator for Simplified Chinese is unavailable, is there any chance I could ask you to also translate the 2 following new messages, that were introduced in v2.9, but that haven't be translated yet?:
* "Zeroing drive: %0.1f%% completed"
* "Detection of non-USB removable drives"
If you want to know more about the circumstances of these 2 messages, you can have a look [here](https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt#L18-L24).

Thanks!

424778940z, 2016-05-31:
"%s partition scheme for BIOS or %s" to Chinese rawly should be "%s 分区方案 用于 BIOS 或 %s (的计算机)"

as a native Chinese speaker, I think this will be better (using printf for example)
printf("用于 BIOS 或 %s 计算机的 %s 分区方案", "UEFI", "GPT");

"Zeroing drive: %0.1f%% completed" "正在清空磁盘: %0.1f%% 已完成"
"Detection of non-USB removable drives" "检测非USB的可移除设备"

pbatard, 2016-05-31:
Thank you very much for the translations!

I'd rather not reverse the order in which the "MBR" and "UEFI" string appear, as this is part of a series of 3 messages where this is set programmatically, and introducing special cases for specific languages would quickly become a maintenance issue (since I don't speak Chinese, so, if I ever break that special case in the future, I won't even realize that it's broken).

I'll be using "%s 分区方案 用于 BIOS 或 %s 的计算机" then.

commit c2fc113f (2016-05-31): [loc] update Simplified Chinese translation to latest

* Also fix an issue with the translation for MSG_031
* Thanks go to choleraehyq for spotting the issue and 424778940z for providing the new translations
* Closes #766
closed 2016-05-31 by pbatard