pbatard/rufus #767: Unable to use Rufus with O2Micro PCI-E SD card reader

424778940z, 2016-05-31:
My laptop has a PCI-E sd card reader from O2Micro, It has been ignored in Rufus

log:
"Found non-USB removable device 'O2Micro SD SCSI Disk Device' => Eliminated"

pbatard, 2016-06-02:
Thanks for the report. Detecting non USB card readers is tricky, as there exist tons of different hardware, so I'll need all the help I can get.

Can I please ask you to enable enumeration debugging in Rufus, by pressing <kbd>Alt</kbd>-<kbd>.</kbd> and then send me the full log? This information might allow me to fix the issue.

424778940z, 2016-06-05:
Sorry for late reply, here is the log

`Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub20#4&d94d0d7&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[193]: USB\VID_8087&PID_0024\5&1515240A&0&1
Processing Hub 2:
  Hub[2] = '\\?\usb#vid_8087&pid_0024#5&1515240a&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[224]: USB\VID_046D&PID_C07C\7E8FE1DB0007
  Found ID[005]: USB\VID_413C&PID_8194\6&2024D25A&0&6
  Found ID[117]: USB\VID_0A5C&PID_5800\0123456789ABCD
Processing Hub 3:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&122e7e9d&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[195]: USB\VID_413C&PID_8187\1C659DAD1CAA
  Found ID[089]: USB\VID_1BCF&PID_280B\6&360A30AF&0&5
Processing Hub 4:
  Hub[4] = '\\?\usb#root_hub20#4&33127676&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[256]: USB\VID_8087&PID_0024\5&122E7E9D&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#root_hub30#5&eb8c102&5&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[254]: USB\VID_0BC2&PID_AB21\MSFT30NA78M718
  Found ID[183]: USB\VID_174C&PID_55AA\424778940FFF
Processing IDs belonging to 'USBSTOR':
  USB\VID_174C&PID_55AA\424778940FFF
Processing IDs belonging to 'UASPSTOR':
  USB\VID_0BC2&PID_AB21\MSFT30NA78M718
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskMsft____Virtual_Disk____1.0_'
Found VHD device 'Microsoft �������'
Removing C: from the list: This is the system disk!
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskPLEXTOR______PX-128M5Pro1.08'
Found non-USB non-removable device 'PLEXTOR PX-128M5Pro' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskO2Micro_SD______________0001'
Found non-USB removable device 'O2Micro SD SCSI Disk Device' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSeagate_BUP_Slim_BK_____0143'
  Matched with ID[254]: USB\VID_0BC2&PID_AB21\MSFT30NA78M718
  Matched with Hub[5]: '\\?\usb#root_hub30#5&eb8c102&5&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskBZH_Inc.ASM1051__DiskBox0___'
  Matched with ID[183]: USB\VID_174C&PID_55AA\424778940FFF
  Matched with Hub[5]: '\\?\usb#root_hub30#5&eb8c102&5&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
`

pbatard, 2016-06-05:
Thank you very much for the report!

I think I may have a fix for this. Can you please download `rufus-2.10_TEST1.exe` [here](https://rufus.akeo.ie/testing/rufus-2.10_TEST1.exe) and let me know if that works better?

424778940z, 2016-06-05:
It's successfully detected my sd card but when I tring put an archlinux install iso on my sd card I got an error

Here is the log
`Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\
Installing Syslinux 6.03...
Using existing './rufus_files/syslinux-6.03/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
Cannot read sector 0
Syslinux FAT access error
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x046A63F1
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-05:
I think this is an unrelated error, as all of the read and write operations are fine until Rufus attempts to install SysLinux. Do you have a link to the archinstall ISO you used? I'd like to test it to see if I can replicate the issue.

Also, can you try:
1. Installing FreeDOS (doesn't need an ISO)
2. With a different ISO (such as a Windows one)?

Thanks!

pbatard, 2016-06-05:
Also, I'd like to know what version of Windows you are using, and if it's 64 bit... I would usually get that from the log, but you cut that part... :sad:

424778940z, 2016-06-05:
FreeDos success
Windows 7 iso success
Debian 9 iso fail...

seems failed at install syslinux

Full log
`Rufus version: 2.10.952 (Test1)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x047A25D1
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ on F:\
Copying DOS files...
Successfully wrote 'F:\COMMAND.COM' (65814 bytes)
[... 1096 lines cut ...]
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0482E225
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-05:
> If I use a usb card reader with the same iso and same card it will success

OK. This might be tricky to troubleshoot since I don't have access to O2Micro hardware...
I'll see what I can do.

pbatard, 2016-06-05:
I tested on Windows 10 64-bit (Build 10586) using `archlinux-2016.05.01-dual.iso` with a SATA HDD (which is the closest thing I have to a PCI-E card reader &mdash; Rufus should handle these the same way as Windows will report them as SCSI drives), but found no issues.

Still, can I please ask you to try [`rufus-2.10_TEST2.exe`](https://rufus.akeo.ie/testing/rufus-2.10_TEST2.exe)?

It should at least provide some more info on the error, and I also tried to change a few things in the section of code that fails, to see if that helps...

424778940z, 2016-06-05:
Here is the log

Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
Could not read sector 0: [0x00000057] 参数错误。
Syslinux FAT access error
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

参数错误 = wrong arguments

Full log
`Rufus version: 2.10.952 (Test2)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0482E225
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
[... 109 lines cut ...]
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error**

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x051F772B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-05:
Yes, error `0x57` is `ERROR_INVALID_PARAMETER`.

I guess I'll need another run with [TEST_3](https://rufus.akeo.ie/testing/rufus-2.10_TEST3.exe) to see a bit clearer with regards to the parameters. Can I ask you to please do that and report?

424778940z, 2016-06-05:
New Log
`Rufus version: 2.10.953 (Test3)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x051F772B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ on F:\
Copying DOS files...
Successfully wrote 'F:\COMMAND.COM' (65814 bytes)
Successfully wrote 'F:\KERNEL.SYS' (45494 bytes)
Successfully wrote 'F:\LOCALE\DISPLAY.EXE' (3648 bytes)
Successfully wrote 'F:\LOCALE\KEYB.EXE' (11446 bytes)
Successfully wrote 'F:\LOCALE\MODE.COM' (16254 bytes)
Successfully wrote 'F:\LOCALE\KEYBOARD.SYS' (40360 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD2.SYS' (29750 bytes)
[... 121 lines cut ...]
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00227D68
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-06:
Weird... You're not getting error `0x57` but `Read 0 bytes instead of 512 requested` this time, even though all I really added was more debugging.

Also, one thing that I've always found strange from your log is this:
```
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
```

Do you have a Virtual Drive mounted? If so could you try unmounting it?
Or, is it possible that you are using a VHD as your `C:` drive? If so how?

424778940z, 2016-06-06:
em...The VHD is my system drive which is C: drive... so I cannot remount it, but I can boot in to WinPE for testing without VHD
It is possible using VHD as C: drive, Windows support that since Windows 7

And here is a little suggestion

The Volume may need to be unmanned before install syslinux. Sometime windows may hold few blocks that contains filesystem data on the device and do not allow any other application RW. I have this problem when using old version winhex, it's report got a unreadable block

pbatard, 2016-06-06:
Shoot, I now realize that TEST_3 inverted an error code, so that I could test the debug output, but in your case, it will skip that test, and that explains why you got a different message.

Please try [TEST_4](https://rufus.akeo.ie/testing/rufus-2.10_TEST4.exe). It should produce the same error as before, but with extra debug.

> It is possible using VHD as C: drive (I mean boot on a VHD), Windows support that since Windows 7

Okay... I don't believe it should interfere with Rufus operations, but that's just one odd parameter that I'd like to eliminate if possible, just to be sure.

> The Volume may need to be unmanned before install syslinux

Not sure what you mean by 'unmanned'. If you mean cleared, then Rufus most definitely clears all the boot sectors and partitions before it repartitions the drive and then installs Syslinux.

If you mean unnamed, I don't see how this could have an impact.

> Sometime windows may hold few blocks that contains filesystem data on the device and do not allow any other application RW.

I'm well aware of this. This is why Rufus requests an exclusive lock to any drive it attempts to write to (which few other formatting/partitioning applications seem to do). This ensures that:
1. If there is any other application trying to access the drive, Rufus will not be able to start its operations
2. As soon as Rufus has started its operations, it is NOT possible for any other application to RW from the drive until Rufus is done.
This use of an exclusive lock by Rufus, a opposed to other application, is actually [causing a lot of grief for Rufus users](https://github.com/pbatard/rufus/issues?utf8=%E2%9C%93&q=access+denied), as many are getting `Access Denied` errors if another application is using the disk...

One thing you may want to try, just in case, is force the use of Large FAT32 for formatting, as this will erase a few more sectors than with regular FAT32. Just press <kbd>Alt</kbd>-<kbd>L</kbd>. You should see 'Large FAT32' instead of 'FAT32' under 'File System'.

I did test with both the Arch ISO with both 'Large FAT32' and 'FAT32' though...

424778940z, 2016-06-06:
> The Volume may need to be unmanned before install syslinux
Sorry, I mean unmounted...

pbatard, 2016-06-06:
> Sorry, I mean unmounted...

Then it most certainly is.
See [this](https://github.com/pbatard/rufus/blob/66534640a42d04dc796096692b72db73c9c1aedb/src/format.c#L1862) happens before [this](https://github.com/pbatard/rufus/blob/66534640a42d04dc796096692b72db73c9c1aedb/src/format.c#L1894)...

424778940z, 2016-06-06:
Here is the log for test 4

`Rufus version: 2.10.953 (Test4)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002452E3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002452E3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
I/O boundary checks disabled
Size : 14.6 GB 30701568 sectors
Cluster size 8192 bytes, 512 Bytes Per Sector
Volume ID is 1c17:81e
32 Reserved Sectors, 14977 Sectors per FAT, 2 FATs
1916973 Total clusters
1916972 Free Clusters
Clearing out 30002 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 15009
[... 156 lines cut ...]
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
Could not read sector 0: [0x00000057] 参数错误。
pp = 0000084C, buf = 011E34EC, secsize = 512
Syslinux FAT access error
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00120382
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-06:
Thanks.
Unfortunately, the parameters that produce the 'Invalid parameter' message look absolutely fine to me:
```
pp = 0000084C, buf = 011E34EC, secsize = 512
```
I will need to think about this some more...

pbatard, 2016-06-06:
Okay, let's try a couple more things, when you have a chance (no rush):

1. Can you try with this new [rufus-2.10_TEST5.exe](https://rufus.akeo.ie/testing/rufus-2.10_TEST5.exe), but instead of choosing an ISO, just install a bare Syslinux 6.03 by:
  * Clicking on the white triangle next to _Format Options_ (See [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected) - this enables the advanced mode.
  * In the _Create bootable disk using_ dropdown, select _Syslinux 6.03_, which should now be available (It only becomes available in advanced mode).
  * Click _Start_ and post the log.

2. Can you also try with rufus 2.9 (the latest release), which you should be able to force to detect your card reader by pressing <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd>. 
__Caution:__ This may also list some of your internal drives, so please be careful when selecting the disk, and make sure this is really your internal card reader, and not another drive!

When testing with 2.9, you can either use the Arch ISO or the bare Syslinux 6.03 and post the log. The reason I'm asking for a 2.9 test is because I want to find out if the Syslinux issue has to do with ant of the changes I introduced with version 2.10.

Thanks!

424778940z, 2016-06-06:
Log:
`Rufus version: 2.10.954 (Test5)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00120382
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Downloading 'ldlinux.c32' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.c32
File length: 122308 bytes
Successfully downloaded 'ldlinux.c32'

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
I/O boundary checks disabled
Size : 14.6 GB 30701568 sectors
Cluster size 8192 bytes, 512 Bytes Per Sector
Volume ID is 16eb:1a4c
32 Reserved Sectors, 14977 Sectors per FAT, 2 FATs
1916973 Total clusters
1916972 Free Clusters
Clearing out 30002 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 15009
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\
Installing Syslinux 6.03...
Successfully wrote 'ldlinux.sys'
d_index = 84
Opened drive \\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa} for write access
d_handle = 00000410
Could not read sector 0: [0x00000057] 参数错误。
pp = 00000410, buf = 010FFB2C (buf[0] = 12), secsize = 512, &bytes_read = 07FAF8E4
Syslinux FAT access error
\\?\Volume{32b64f2d-ea22-11e5-9cd1-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'O2Micro SD SCSI Disk Device'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'BZH Inc. ASM1051  DiskBox USB Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03CFA8EB
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-06:
Thanks. Could I also please have the test with Rufus 2.9 as well (step 2.)?

For now it doesn't look like I'll be able to come up with a decent solution for something that seems to be linked to a hardware issue (there's really no reason for the failure - all the parameters are correct, and Rufus is not doing anything different than it does for other PCI-E based devices, which work just fine). So the last hope is that this might be linked to something that was introduced with 2.10, which the 2.9 test with <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> will tell...

424778940z, 2016-06-06:
2.9 Log

`Rufus version: 2.9.934
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
NOTE: The Chinese Simplified (简体中文) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub20#4&d94d0d7&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[037]: USB\VID_8087&PID_0024\5&27CED4AF&0&1
Processing Hub 2:
  Hub[2] = '\\?\usb#vid_8087&pid_0024#5&24e82f42&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[084]: USB\VID_1BCF&PID_280B\6&2875F4C9&0&5
  Found ID[195]: USB\VID_413C&PID_8187\1C659DAD1CAA
Processing Hub 3:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&27ced4af&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[002]: USB\VID_413C&PID_8194\6&A19BBE2&0&6
  Found ID[117]: USB\VID_0A5C&PID_5800\0123456789ABCD
  Found ID[224]: USB\VID_046D&PID_C07C\7E8FE1DB0007
Processing Hub 4:
  Hub[4] = '\\?\usb#root_hub20#4&33127676&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[019]: USB\VID_8087&PID_0024\5&24E82F42&0&1
Processing Hub 5:
  Hub[5] = '\\?\usb#root_hub30#5&eb8c102&6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[254]: USB\VID_0BC2&PID_AB21\MSFT30NA78M718
  Found ID[148]: USB\VID_0BC2&PID_AB21\MSFT30NA792Y14
Processing IDs belonging to 'UASPSTOR':
  USB\VID_0BC2&PID_AB21\MSFT30NA78M718
  USB\VID_0BC2&PID_AB21\MSFT30NA792Y14
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskMsft____Virtual_Disk____1.0_'
Found VHD device 'Microsoft �������'
Removing C: from the list: This is the system disk!
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskPLEXTOR______PX-128M5Pro1.08'
Found non-USB non-removable device 'PLEXTOR PX-128M5Pro' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSeagate_BUP_Slim_BK_____0143'
  Matched with ID[254]: USB\VID_0BC2&PID_AB21\MSFT30NA78M718
  Matched with Hub[5]: '\\?\usb#root_hub30#5&eb8c102&6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSeagate_BUP_Slim_BK_____0143'
  Matched with ID[148]: USB\VID_0BC2&PID_AB21\MSFT30NA792Y14
  Matched with Hub[5]: '\\?\usb#root_hub30#5&eb8c102&6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub20#4&d94d0d7&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[037]: USB\VID_8087&PID_0024\5&27CED4AF&0&1
Processing Hub 2:
  Hub[2] = '\\?\usb#vid_8087&pid_0024#5&24e82f42&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[084]: USB\VID_1BCF&PID_280B\6&2875F4C9&0&5
  Found ID[195]: USB\VID_413C&PID_8187\1C659DAD1CAA
Processing Hub 3:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&27ced4af&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[002]: USB\VID_413C&PID_8194\6&A19BBE2&0&6
  Found ID[117]: USB\VID_0A5C&PID_5800\0123456789ABCD
  Found ID[224]: USB\VID_046D&PID_C07C\7E8FE1DB0007
Processing Hub 4:
  Hub[4] = '\\?\usb#root_hub20#4&33127676&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
[... 156 lines cut ...]
Found non-USB non-removable device 'PLEXTOR PX-128M5Pro' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskO2Micro_SD______________0001'
Found non-USB removable device 'O2Micro SD SCSI Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSeagate_BUP_Slim_BK_____0143'
  Matched with ID[254]: USB\VID_0BC2&PID_AB21\MSFT30NA78M718
  Matched with Hub[5]: '\\?\usb#root_hub30#5&eb8c102&6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSeagate_BUP_Slim_BK_____0143'
  Matched with ID[148]: USB\VID_0BC2&PID_AB21\MSFT30NA792Y14
  Matched with Hub[5]: '\\?\usb#root_hub30#5&eb8c102&6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
3 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1911, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x04B12057
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15719202816 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-07:
Thanks. I think this issue is similar to what @matthewschallenkamp also reported in #673, with another card reader.

But without access to a PCI-E based card reader (USB based card readers seem to behave as expected), this will be very difficult to troubleshoot. And I looked around for PCI-E based card readers that I could purchase, but all the ones I could find are actually using the USB protocol (which can be passed through PCI-Express), which won't help...

I'll still try to think of other things we may test...

424778940z, 2016-06-07:
O2Micro only make OEM products, there is no retail.

But probably you can find a Dell laptop that using this card reader in following list

Latitude E5420
Latitude E5520
Precision M4600
Precision M6600
Latitude E5440
Latitude E5540
Latitude E6440
Latitude E6540
Latitude E7240
Latitude E7440
Precision M4800
Precision M6800
Venue 7130 Pro/7139 Pro

pbatard, 2016-06-07:
Thanks for the list.

Unfortunately, I'm just a single developer with limited resources (since Rufus is 100% gratis), so purchasing a laptop is not something I can do. And I don't know anyone, with a (non-USB) card-reader enabled laptop that I could borrow. But I'll keep looking for alternate ways to troubleshoot this...

424778940z, 2016-06-07:
Sorry, I didn't mean you should buy one...Dell laptops are common, maybe you can find someone using it...

pbatard, 2016-06-07:
There are still a few things I'd like to try, if you're still willing to test...
Can you please download [TEST_6](https://rufus.akeo.ie/testing/rufus-2.10_TEST6.exe) and let me know how it goes?

424778940z, 2016-06-07:
Log

`Rufus version: 2.10.955 (Test6)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'Generic SA08G SD Card'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x04C078DE
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse './rufus_files/syslinux-6.03/ldlinux.c32' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{94035821-2b85-11e6-9d1c-005056c00008} for write access
Size : 7.3 GB 15276032 sectors
Cluster size 4096 bytes, 512 Bytes Per Sector
Volume ID is 1af7:2a1f
32 Reserved Sectors, 14889 Sectors per FAT, 2 FATs
1905777 Total clusters
1905776 Free Clusters
Clearing out 29818 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14921
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{94035821-2b85-11e6-9d1c-005056c00008}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{94035821-2b85-11e6-9d1c-005056c00008} for write access
Successfully wrote 'ldlinux.sys'
Could not read sector 0: [0x00000057] 参数错误。
Syslinux FAT access error
\\?\Volume{94035821-2b85-11e6-9d1c-005056c00008}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'Generic SA08G SD Card'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03EAD97E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-07:
Thank you so much!

Wow, this sure is weird... I was expected TEST_6 to fail earlier, but it didn't....

In light of this, let's try one last time for today, with [TEST_7](https://rufus.akeo.ie/testing/rufus-2.10_TEST7.exe) if that's okay with you.

I really appreciate your willingness to perform all these tests.

424778940z, 2016-06-07:
Test 7 Log

`Rufus version: 2.10.955 (Test7)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'Generic SA08G SD Card'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03EAD97E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse './rufus_files/syslinux-6.03/ldlinux.c32' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{94035821-2b85-11e6-9d1c-005056c00008} for write access
I/O boundary checks disabled
Size : 7.3 GB 15276032 sectors
Cluster size 4096 bytes, 512 Bytes Per Sector
Volume ID is 1b19:351a
32 Reserved Sectors, 14889 Sectors per FAT, 2 FATs
1905777 Total clusters
1905776 Free Clusters
Clearing out 29818 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14921
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{94035821-2b85-11e6-9d1c-005056c00008}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{94035821-2b85-11e6-9d1c-005056c00008} for write access
Successfully wrote 'ldlinux.sys'
Could not read sector 0: [0x00000057] 参数错误。
Syslinux FAT access error
\\?\Volume{94035821-2b85-11e6-9d1c-005056c00008}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found card reader device 'Generic SA08G SD Card'
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x040A2516
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-08:
Thanks for the quick update.

I think this will be all for today, as this really is a tricky issue, so I'll have to think about it some more...

424778940z, 2016-06-08:
That's fine, I will keep following this issue

pbatard, 2016-06-08:
Time for another test! :smiley: Here is [TEST_8](https://rufus.akeo.ie/testing/rufus-2.10_TEST8.exe).

I still don't have any idea what to fix. But I'm hoping this test version can help identify more precisely why the read fails. The part I'm interested in in the log should go something like this:
```
Installing Syslinux 6.03...
Opened drive \\?\Volume{3c9ea97e-83a5-11e5-8487-fcaa14e7bd8e} for write access
TESTING
1. Direct ReadFile(00000398, 00732168, 512, 08CCFB5C, NULL)
   SetFilePointerEx(00000398, 0, NULL, FILE_BEGIN)
2. Direct (Post SetFilePointerEx) ReadFile(00000398, 00732168, 512, 08CCFB5C, NULL)
3. Direct libfat_readfile(00000398, 00732168, 512, 0)
4. libfat_get_sector (fs->readptr = 00000398)
ALL PASSED
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
```
I'm hoping this will fail before `ALL PASSED` when you test, which should tell me more about the nature of the error.

424778940z, 2016-06-09:
`Rufus version: 2.10.956 (Test8)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x040A2516
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse './rufus_files/syslinux-6.03/ldlinux.c32' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa} for write access
I/O boundary checks disabled
Size : 7.3 GB 15276032 sectors
Cluster size 4096 bytes, 512 Bytes Per Sector
Volume ID is 1cf4:1a26
32 Reserved Sectors, 14889 Sectors per FAT, 2 FATs
1905777 Total clusters
1905776 Free Clusters
Clearing out 29818 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14921
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa} for write access
TESTING
1. Direct ReadFile(00000304, 02C23870, 512, 0414FA04, NULL)
   SetFilePointerEx(00000304, 0, NULL, FILE_BEGIN)
2. Direct (Post SetFilePointerEx) ReadFile(00000304, 02C23870, 512, 0414FA04, NULL)
3. Direct libfat_readfile(00000304, 02C23870, 512, 0)
4. libfat_get_sector (fs->readptr = 00000304)
Could not read sector 0: [0x00000057] 参数错误。
libfat_get_sector() failed
\\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x097C0983
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-09:
Thanks!

With your help, I think we're starting to make progress on the issue, but I still need a bit more information. Can you please try [TEST_9](https://rufus.akeo.ie/testing/rufus-2.10_TEST9.exe)?

424778940z, 2016-06-09:
`Rufus version: 2.10.956 (Test9)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x097C0983
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse './rufus_files/syslinux-6.03/ldlinux.c32' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa} for write access
I/O boundary checks disabled
Size : 7.3 GB 15276032 sectors
Cluster size 4096 bytes, 512 Bytes Per Sector
Volume ID is 1100:2a2d
32 Reserved Sectors, 14889 Sectors per FAT, 2 FATs
1905777 Total clusters
1905776 Free Clusters
Clearing out 29818 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14921
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa} for write access
TESTING
1. Direct ReadFile(00000028, 02DC3870, 512, 077AFA04, NULL)
   SetFilePointerEx(00000028, 0, NULL, FILE_BEGIN)
2. Direct (Post SetFilePointerEx) ReadFile(00000028, 02DC3870, 512, 077AFA04, NULL)
3a. Direct libfat_readfile(00000028, 02DC3870, 512, 0)
   libfat_readfile: ReadFile(00000028, 02DC3870, 512, 077AF974, NULL)
3b. Repeat libfat_readfile(00000028, 02DC3870, 512, 0)
   libfat_readfile: ReadFile(00000028, 02DC3870, 512, 077AF974, NULL)
4. libfat_get_sector (fs->readptr = 00000028)
   libfat_get_sector - not in cache
   libfat_get_sector - fs->read(00000028, 02DC3094, 512, 0)
   libfat_readfile: ReadFile(00000028, 02DC3094, 512, 077AF924, NULL)
Could not read sector 0: [0x00000057] 参数错误。
libfat_get_sector() failed
\\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0C1A66BC
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-09:
Thank you!!!

I'm fairly certain we have a buffer alignment issue on our hands, which would make a lot sense with the `ERROR_INVALID_PARAMETER` code.

I think the underlying SD card driver is only able to read sectors to data buffers that are aligned to 8 bytes, so when it receives a request from the `ReadFile()` call, for a data buffer like `02DC3094`, is returns `ERROR_INVALID_PARAMETER` because that buffer does not end in `0` or `8`. And the other calls work because the data buffer address is `02DC3870`, which is aligned to 8 bytes.

Most disk devices don't seem to care about data alignment, but it looks like this one does.

I have now fixed part of the code, in [TEST_10](https://rufus.akeo.ie/testing/rufus-2.10_TEST10.exe), so that the Syslinux sector buffers should always be aligned to 16 bytes (just in case there are other devices for which an 8-byte data alignment is not enough). Can you please test and report?

424778940z, 2016-06-09:
Great! It WORKS

`Rufus version: 2.10.956 (Test10)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0C1A66BC
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse './rufus_files/syslinux-6.03/ldlinux.c32' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa} for write access
I/O boundary checks disabled
Size : 7.3 GB 15276032 sectors
Cluster size 4096 bytes, 512 Bytes Per Sector
Volume ID is 13e3:3366
32 Reserved Sectors, 14889 Sectors per FAT, 2 FATs
1905777 Total clusters
1905776 Free Clusters
Clearing out 29818 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14921
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
[... 48 lines cut ...]
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
Using autorun.inf label for drive F: '7_8GB'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0CA4CB91
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-09:
Alright!! :smile: 

I'll need to clean up my code a little bit, and I may have a version to test with the final patch to test, but I really can't thank you enough for all of this. You've been great help!

424778940z, 2016-06-09:
No problem, I'm happy to help people who developing great software.

pbatard, 2016-06-09:
OK, I believe I have finalized the patch now. Can you please run one last test with [TEST_11](https://rufus.akeo.ie/testing/rufus-2.10_TEST11.exe)?

If you report that it works, I'll just commit the patch and close the issue. Thanks again!

424778940z, 2016-06-09:
Syslinux then Archlinux iso

`Rufus version: 2.10.957 (Test11)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0804
Will use default UI locale 0x0804
Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
Using autorun.inf label for drive F: '7_8GB'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0CA4CB91
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse './rufus_files/syslinux-6.03/ldlinux.c32' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{d56f21be-2c6d-11e6-9d1e-1c659dad1caa} for write access
I/O boundary checks disabled
Size : 7.3 GB 15276032 sectors
Cluster size 4096 bytes, 512 Bytes Per Sector
Volume ID is 141a:393d
32 Reserved Sectors, 14889 Sectors per FAT, 2 FATs
1905777 Total clusters
1905776 Free Clusters
Clearing out 29818 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14921
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
[... 215 lines cut ...]
Created: F:autorun.ico

Found VHD device 'Microsoft 虚拟磁盘'
Removing C: from the list: This is the system disk!
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found UAS (USB 3.0) device 'Seagate BUP Slim BK SCSI Disk Device' (0BC2:AB21)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found card reader device 'O2Micro SD SCSI Disk Device'
Using autorun.inf label for drive F: 'ARCH_201606'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0CD7A600
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-06-09:
Great! :+1:

commit 3fdf6229 (2016-06-09): [syslinux] fix Syslinux installation for some media

* Some disk devices, such as O2Micro PCI-E SD card readers, return ERROR_INVALID_PARAMETER
  when attempting to use ReadFile() on a volume against a non 8-byte aligned buffer.
* This patch ensures that the Syslinux buffers for sector patching are 16-byte aligned.
* HUGE THANKS to 424778940z, for the numerous testing that allowed to troubleshoot this issue.
* Closes #767
closed 2016-06-09 by pbatard