pbatard/rufus #769: Unable to boot Ubuntu after writing USB with Rufus.

Cookcha, 2016-06-04:
I used Rufus when creating the bootable USB and Ubuntu loaded perfectly, no issues. I used the try feature and later installed Ubuntu. I tried to download Google Chrome, and couldn't so I rebooted to see if that would help fix the issue. When I attempted to boot back into Ubuntu, it wouldn't load in any mode. I checked my Rufus log and found some odd things. (At least to me they seem odd. I'm new at all this so I'm probably just guessing.) Would that have anything to do with the boot issue? 

After nosing around in your repository, I noticed you had a fix for the ISO9660 problems... I'm not sure what to do with that information if it is a fix.

pbatard, 2016-06-05:
There's nothing out of the ordinary in your log (The latest Ubuntu seem to use a mix of GRUB and Syslinux, but we don't care about GRUB for this boot, so not detecting the GRUB version is not a problem).

> I noticed you had a fix for the ISO9660 problems

Yeah, if you somehow managed to __build__ your own version of Rufus 2.10 (the next version, which  won't be released for another few weeks... or months), using the very specific Clang/C2 __non default__ compiler under Visual Studio, during the ~3 hours during which I had just added support for Clang, which is what introduced the ISO9660 breakage (but __only__ when compiling with Clang, and NOT with MinGW, which is the compiler I use for releases), and hadn't had a chance to fix (since that would only affect people using a __developer-only__ feature that I had just introduced, and I [clearly flagged Clang as experimental](https://github.com/pbatard/rufus/commit/3b0c4e80839486a3b0ed98cc771390028d6193c1))

In other words, that ISO9660 can __NOT__ affect you in any way since you are using the release, and there is NO breakage that I am aware of regarding ISO9660 and Rufus 2.9.

What's most likely is that:
1. You are running into a Live system limitation (such as having part of your file system __read-only__)
2. Your flash drive got either corrupted for whatever reason while you were running Ubuntu (you'll have to check the Ubuntu forums for people with a similar issue)
3. Your drive decided to fail as you were using it, since flash memory has a limited number of write cycles before failing.

You may wanna try to run a bad blocks check in Rufus to confirm that it's not a hardware issue. If it isn't, then you will have to take it with the Ubuntu people. Or you may also try to write your image in DD mode instead of ISO mode (which Rufus __will__ ask you about when you click Start) and see if that helps.

pbatard, 2016-06-06:
Any update on this? For the record I tested `ubuntu-16.04-desktop-amd64.iso` in live mode and had no issue saving files from the internet.

Cookcha, 2016-06-07:
Thanks for the feedback. As I said I'm new to all this so excuse me. I actually ended up kind of killing my laptop, I'm hoping I will be able to restore it with Ubuntu. I tried uninstalling Ubuntu by deleting it's partitions and starting over (factory reset of the whole PC) but I'm not sure what happened... All that starts up is a GRUB command line with limited commands. Before I began the reset I wiped the USB so I am going to attempt to usensure another computer to create the Bootable USB and then see if I can't get it working from there.

pbatard, 2016-06-22:
No update &rarr; Closing.

closed 2016-06-22 by pbatard

Cookcha, 2016-06-23:
Sorry it took so long to update. Several things came up and I was unable to access my computer or phone. I was able to restore my computer with Ubuntu but I was having the same problem as before with the freezing. The issue was fixed by booting with legacy mode active.