pbatard/rufus #770: [INTERNAL] List of 2.10 small enhancements and fixes

pbatard, 2016-06-05:
- [x] Add Clang/C2 support in Visual Studio
- [ ] Images like `ubuntu-16.04-desktop-amd64.iso` may confuse users due to the `Could not read Grub version from 'boot/grub/i386-pc/normal.mod'`, since the Grub they use is for UEFI only. We may want to silence these benign reports
- [ ] Can't reset partitions first time around after writing `super_grub2_disk_x86_64_efi_2.02s4.iso` in DD-mode (since they too use GRUB/EFI but didn't include a `/efi/boot/bootx64.efi` on the ISO FS, but in the boot img... which we can't use anyway, since their GRUB is crippled with regards to FAT32 support)

SvsGit, 2016-07-06:
GUI enhancement request: Reset progress bar after writnig is done.

pbatard, 2016-07-06:
> GUI enhancement request: Reset progress bar after writnig is done.

1. Issues tagged as [INTERNAL] are only for developers. If you are not a Rufus developer, you shouldn't comment on them, as they are not for general public consumption. They are only in the issue tracker for convenience.
2. The progress bar not being reset is by design, and it makes no sense to me to alter this behaviour as the vast majority of Rufus usage is for a single image, where the user might do something else (or get away from the computer if it's a long operation), in which case it seems logically to want to keep the big green indicator that the operation is fully complete.

So, no.

closed 2016-07-19 by pbatard