pbatard/rufus #772: Completion of error handling

elfring, 2016-06-07:
Would you like to add more error handling for return values from functions like the following?
- [malloc](http://pubs.opengroup.org/onlinepubs/9699919799/functions/malloc.html "Memory allocation") ⇒ [get_loc_cmd](https://github.com/pbatard/rufus/blob/c3f47ada06422ee6232fd210b2a34b5e9a5ef986/src/parser.c#L47)
- [printf](http://pubs.opengroup.org/onlinepubs/9699919799/functions/printf.html "Print formatted output") ⇒ [PrintUsage](https://github.com/pbatard/rufus/blob/f0b13361beeb67c642ee876313982257412b59bf/src/rufus.c#L2709)

pbatard, 2016-06-07:
* `get_loc_cmd()`? Probably. Thanks.
* `PrintUsage()`? No. Failure to print the output is not a big deal, and if `printf()` fails in the first place, I wouldn't be able to alert the user about it anyway.

closed 2016-06-07 by pbatard

elfring, 2016-06-07:
Thanks for [your small source code improvement](https://github.com/pbatard/rufus/commit/a10ea1498bee62dcbdc7c3c1b19cb9c4e6d8f507 "More checks for memory allocation failures").

>…, I wouldn't be able to alert the user about it anyway.

* How do you think about to indicate a file output failure by [the program exit status](https://en.wikipedia.org/wiki/Exit_status#C_language "Usage of the exit status?")?
* Are there [any more update candidates left over](https://github.com/pbatard/rufus/blob/39d6b904472abcf262e61f2ab44ee33c87e87500/src/rufus.c#L2730 "Update candidate: SetHogger()")?

pbatard, 2016-06-07:
> How do you think about to indicate a file output failure by the program exit status?

Yeah, I thought about that too, but I just don't think it's worth it.

> Are there any more update candidates left over?

How about you submit a patch? You seem to be inclined to identify them all, and that would save me time...

elfring, 2016-06-07:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Unchecked return values") a bit more. Would you like to detect every error situation as early as possible?

How do you think about to improve [static source code analysis](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2C_C.2B.2B "Tools for static code analysis") also for this software?

pbatard, 2016-06-07:
I would suggest you also understand how code gets developed, and how a developer's time is limited, and he or she therefore tends to focus on things that matter.

An alloc that fails will produce a crash. A crash will get a user's attention, and be reported and investigated. Therefore, focusing on malloc checks is not a productive use of a smart developer's time, as you can __spread__ part of your limited development time onto users.

That's not to say I tend to ignore malloc checking. But there will always be a few that get missed for varied reasons, and experience shows that it's really not that big a deal.

> How do you think about to improve static source code analysis also for this software?

I use Visual Studio static analysis and [Coverity](https://scan.coverity.com/projects/pbatard-rufus?tab=overview) and sometime the WDK analyzer as well (OACR). So it's not like I'm ignoring those either. As you will see, I actually pay a lot of attention of what Coverity reports and fix it. I like static analyzers, but I'm also not going to go out of my way to run more than 3 of them...

elfring, 2016-06-07:
Do the mentioned source code analysis tools indicate function calls like the following for further considerations?
* [calloc](http://pubs.opengroup.org/onlinepubs/9699919799/functions/calloc.html "Memory allocation")
* [CreateMutex](https://msdn.microsoft.com/en-gb/library/windows/desktop/ms682411%28v=vs.85%29.aspx "Create a mutex object")

Are you interested to apply [aspect-oriented software development](http://aspectc.org/ "AspectC++")?
How do you think about to encapsulate error detection and corresponding exception handling as [a reusable aspect](http://en.wikipedia.org/wiki/Aspect_%28computer_programming%29 "Combination of a pointcut with advice") in your software?

pbatard, 2016-06-07:
As I said, it looks like both the defaults for the analysis tools I use, and myself, share the same viewpoint that not checking for a malloc value is not that big a deal.

> Are you interested to apply aspect-oriented software development?

After about 30 years of active software development, am I interested for yet another person to try to push their pet development methodology onto me? No.

commit a10ea149 (2016-06-07): [misc] add more check for malloc() errors

* Closes #772
* With thanks to @elfring