pbatard/rufus #778: Write ISO to USB 'without' requiring boot files

drankinatty, 2016-06-18:
All, there needs to be an option in rufus to create a USB from an ISO without requiring boot file be written, e.g.:

[ ] create USB from image [ISO filename]

It could be placed right under the create bootable USB from image. The reason being, I needed to create a USB from the MSI driver DVD that came with the Z170 motherboard in order to get past the Win7 installation hang (when installing from USB) wanting drivers before proceeding.

Creating/booting the Win7 install with rufus was a piece of cake. However, rufus would not let me create a USB from the MSI driver ISO. (rather than just copying files to a formatted USB).

Ultimately I just wrote the ISO to the DVD with `dd` from Linux, but it would be nice to be able to do the same thing from rufus.

pbatard, 2016-06-18:
This feature is already there. If you were able to write your ISO using `dd` in Linux, then you should have seen a prompt after pressing start, asking you whether you wanted to write in ISO image mode or DD mode. Of course that only works for ISOHybrid images (i.e. images that can written both in ISO or DD mode).

Also, you can select DD Image in the dropdown, instead of ISO, for pure (non ISO) bootable images.

So I'm not sure what you want exactly that's not already there. Are you sure you carefully looked at all the options that Rufus proposed to you?

pbatard, 2016-06-21:
Since there has been no update, I'm going to close this issue.

Oh, and in case your request was for automatically extracting non bootable ISOs to an USB drive, I'm not planning to do that for the following reasons:
* Because it's non bootable, it's next to impossible to know what file system will actually work, so I could let users, say, extract DOS drivers to an NTFS file system (or UDF or ReFS), which they wouldn't be able to read. And if I were to force FAT32 only, this would create other issues.
* I would need to educate people on what the new mode is really about. With 2 million downloads of Rufus each month, it's very likely that quite a few users would select the non bootable extraction mode, and then complain that their ISO doesn't boot. Or you'd need to add warning prompts ('Are you __really__ sure you want to create a non bootable drive'), which would annoy everyone.
* You'd have to be __very__ lazy to want an app to format and extract an ISO for you, while you can just 7-zip to extract the ISO
* This is pure [feature creep](https://en.wikipedia.org/wiki/Feature_creep), and I just don't see enough demand for this to justify its inclusion in Rufus.

closed 2016-06-21 by pbatard

drankinatty, 2016-06-24:
On 06/18/2016 06:52 PM, Pete Batard wrote:
> This feature is already there. If you were able to write your ISO using `dd`
> in Linux, then you should have seen a prompt after pressing start, asking you
> whether you wanted to write in ISO image [...]

The confusion I had was every time I would check to burn an ISO, that option was
only offered in combination with installing the system files. In this case I
simply wanted to burn an ISO for the MSI Motherboard Installer disk which
contained the xHCI drivers needed for the Win7 install from USB. There was only
a single option to check to get to the ISO dialog and that option also made the
disk bootable. I didn't want it bootable, just written as a plain old ISO.

I'll check again and write back.

-- 
David C. Rankin, J.D.,P.E.
Rankin Law Firm, PLLC
510 Ochiltree Street
Nacogdoches, Texas 75961
Telephone: (936) 715-9333
Facsimile: (936) 715-9339
www.rankinlawfirm.com