pbatard/rufus #779: Bought Two New Flash Drives - Now they Can't be Written To

TzZek, 2016-06-19:
I ran rufus and selected first drive trying to make a Bootable USB of antergos , after that it failed to write and now i can't add files to it and nothing works. I tried regedit and the terminal disk unlock methods and neither worked.

`Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Found conflicting isolinux.bin versions:
  '/arch/boot/syslinux/isolinux/isolinux.bin' (4.05/0x4f23eb18) vs '/isolinux/isolinux.bin' (6.03)
  Detected Syslinux version: 4.05/0x4f23eb18 (from '/arch/boot/syslinux/isolinux/isolinux.bin')
  Warning: Conflict between Isolinux version and the presence of ldlinux.c32...
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ANTERGOS'
  Size: 1791655936 bytes
  Uses: Syslinux/Isolinux v4.05
    With an old menu.c32
    With an old vesamenu.c32
  Uses: EFI
Using image: antergos-2016.06.14-x86_64.iso
Will replace obsolete 'menu.c32' from ISO with the one found in './rufus_files/syslinux-4.07/menu.c32'
Will replace obsolete 'vesamenu.c32' from ISO with the one found in './rufus_files/syslinux-4.07/vesamenu.c32'

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown FAT16 or FAT32 Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x0000045D] The request could not be performed because of an I/O device error.
Could not reset partitions
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:00E1)`

TzZek, 2016-06-19:
I'm not saying Rufus broke them but i'm wondering what's the correct way to fix them after using Rufus.

TzZek, 2016-06-19:
Flash Drive Mode: PNY 16GB Retract USB 2.0 Flash Drive 2-Pack P-FD16GX2RTC-GE

pbatard, 2016-06-19:
1. Try to reformat them in Linux. I am aware of a specific issue on Windows (#759) where trying to clean the partitions will make the drive unwriteable, unless it is repartitioned and reformatted on a different OS. Note that this problem has nothing to do with Rufus. Using `diskpart`'s `clean` command in Windows will produce the exact same results.
2. Are you sure you weren't a victim of "fake" drives? These are drives that report more capacity than they actually have (e.g. 16GB instead of actually 1GB or less), which can render the drive inoperable as soon as you format it. This is why you should usually run a bad blocks check on any new drive purchased, as Rufus will detect if the drive is "fake".
3. Have you tried from a different computer? If you purchased your drives from a reputable source and both failed with an I/O device error on the very first operation (clean partitions, which isn't anything fancy or even risky at all), I would tend to suspect that something is wrong with the USB hardware on your computer. Maybe your USB port is flaky, or you are using a problematic extension cable, or something else.

TzZek, 2016-06-19:
I did purchase from a reputable source , reformatting in Linux led to "Read-Only write system" , i'll just return them to the store and get a different brand and hope for the best.

pbatard, 2016-06-19:
Okay. I would encourage you to at least try with a different USB port on your PC with the new drives, as it may be a hardware issue.

TzZek, 2016-06-19:
I have tried it in 4 USB 3.0 Ports and 6 USB 2.0 Ports , all have failed. Will post back later to see if it may have just been that specific drive. Going to buy a couple new ones.

TzZek, 2016-06-20:
Alright so I went out and purchased a few for testing as well as a set of the one's that failed me for testing purposes , all the other flash drives allowed me to make them bootable with versions of Linux however the : PNY 16GB Retract USB 2.0 Flash Drive 2-Pack P-FD16GX2RTC-GE broke in every attempt , I'd advice no one to purchase these for making a Bootable USB.

pbatard, 2016-06-20:
Thanks for the report.

Have you tried a full bad blocks check in Rufus on one of the new PNY's?
I'd really like to know the outcome of that (Rufus log) to see if there's anything that could explain the drive failure.

I'll see if I can get my hands on these drives, because there's something very wrong with flash drives that appear to fail from a benign reformat and copy operation. Unfortunately, these seem to only be sold by Walmart in the U.S., so that might be difficult...

Oh, and since I also spotted it from your log, the next version of Rufus will fix the small issue of Rufus installing Syslinux 4.05 instead of 6.03 when using Antergos (4f3e27b0894261a73b891c42fe33288f5e00bd85):
```
  Found conflicting isolinux.bin versions:
  '/arch/boot/syslinux/isolinux/isolinux.bin' (4.05/0x4f23eb18) vs '/isolinux/isolinux.bin' (6.03)
  Detected Syslinux version: 4.05/0x4f23eb18 (from '/arch/boot/syslinux/isolinux/isolinux.bin')
```

TzZek, 2016-06-20:
I don't have the logs unfortunately when is the next version ETA release? Also is it recommended when the pop up ask's if Rufus can upgrade some files in the .iso to do so ? Or just leave it alone

TzZek, 2016-06-20:
Thanks didn't know those Milestones existed , and I'm not really sure my system says it's in Legacy + UEFI , but when i check my Windows settings it says i'm in MBR.

pbatard, 2016-06-20:
If you see any mention of UEFI for your system, then it's UEFI. Windows can use MBR even on UEFI.

TzZek, 2016-06-20:
Alright thanks couldn't get the Live GUI to pop up with normal .iso do made a dd one and it worked than but after i could see the GUI and chose to launch the live it went back to black screen , I know this isn't Rufus i doubt but wish i could get this dual boot working lol. You have any place of contact or anything if you would be willing to help me you've been amazing with the support.

TzZek, 2016-06-20:
Since i just did a DD image how can i reformat that drive to become usable again I attempted it and it disappeared from my list of plugged in USB's but still appears in disk manager.

pbatard, 2016-06-20:
> it went back to black screen

Typical of recent Linux distros with newer (Maxwell) nVidia GPUs (750 or later). Unfortunately, Maxwell support in nouveau (the Open Source nVidia Linux driver) is very poor for newer GPUs, and almost all distros suffer from this.

You may be able to modify the kernel config to pass `nomodeset` or something. Please see the support forums for your specific distro.

pbatard, 2016-06-22:
Since I don't think there's more that can be done on this issue, I will close it.

closed 2016-06-22 by pbatard

zman7123, 2016-07-06:
PNY has a firmware posted online that fixes the right protect error for the 16GB Retract USB: http://www.pny.com/support/technical-support/usb-flash-drives/16gb-retract-usb-firmware

pbatard, 2016-07-06:
Aha! It all makes sense thne.

It is exceedingly likely that the firmware for those PNY 16GB Retract flash drives is buggy, and can switch to write protect when conducting legitimate operations. By publishing a firmware update, PNY have implicitly acknowledged that issue (since I know of NO flash drive manufacturer that would publish firmware updates for non-premium drives, unless they have found a major problem with them).