pbatard/rufus #780: Rufus + elementary OS

buonviri, 2016-06-20:
I downloaded rufus as recommended by one of the install guides for elementary OS. When loading the ISO I get the log below. I found some older issues at this site with the same type of warning/error but those were fixed. Should I be concerned?

Scanning image...
ISO analysis:
  Image is an ISO9660 image
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'elementary OS 0.3.2 Freya'
  Size: 1151336448 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
Using image: elementaryos-0.3.2-stable-amd64.20151209.iso

pbatard, 2016-06-20:
> Should I be concerned?

No. All of the warnings you get are benign, and some of them will be silenced in the next version.

closed 2016-06-20 by pbatard

Eliasdf, 2016-07-01:
I wanted to ask about the "Could not" warnings above:

  Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version

Those are benign as well??

pbatard, 2016-07-01:
Yes they are.