pbatard/rufus #781: Rufus-formatted flash drive doesn't appear during boot

github-account-because-they-want-it, 2016-06-22:
Let me start by saying this may not be a Rufus issue per-se, but I appreciate any help on this.

So I just received this HP laptop today. It has Insyde20 bios which I updated to revision F.35 from HP's website.

I first started by creating the usual MBR formatted drive and extracting my ISO on it. It doesn't show up during boot unless I go into bios to enable legacy mode. But then Windows won't install on the internal SSD because it's GPT formatted.

Then I found Rufus. I used the pure GPT/UEFI option to create my disk. Here's the log:
[rufus.txt](https://github.com/pbatard/rufus/files/327016/rufus.txt)

The disk won't appear when I reboot in UEFI mode (legacy mode disabled). I also tried enabling/disabled protected boot with no effect.

pbatard, 2016-06-22:
```
Using image: Windows 10 AIO Build 1511 May 2016 (x86x64).iso
```

That's a non official Windows 10 image. I could stop right there and tell you that, in terms of Windows, I don't support anything that doesn't originate from Microsoft, and All In One images certainly don't. Even more so as the fact that you're using a non official ISO seems to be the root of your issue.

Official Windows 10 ISOs are __exceedingly__ easy to come by, so please do yourself a favour and go to [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench) to download one.

As to your issue, I'm pretty sure this is it:
```
Extracting: H:\efi\boot\bootia32.efi (953.8 KB)
```
If there is no `efi\boot\bootx64.efi` file (and your "All-In-One" ISO clearly doesn't have one &mdash; so much for pretending to be "x86x64"), then it will never boot against an 64-bit UEFI firmware in UEFI mode, which your machine probably is.

The AIO people __screwed up__. Their ISO is NOT x64 compatible. So do yourself a favour and stop using images from unreputable sources as it's a waste of time for everybody, including yourself.

Since I believe I answered your question, I will close this issue.

closed 2016-06-22 by pbatard

github-account-because-they-want-it, 2016-06-23:
You were absolutely right. Another image resolved the issue. Thank you.

xerces8, 2016-08-22:
Hi!

I'll post here as the issue is similar. I created an UEFI stick for win7, that does not work(boot), due to missing \efi\boot\bootx64.efi file on the stick. I opened the ISO file (with 7zip and by mounting it in win10 by double-click) and it has no \efi\boot\ folder at all. The ISO is official:
Windows 7 Enterprise with Service Pack 1 (x64) - DVD (English)
en_windows_7_enterprise_with_sp1_x64_dvd_u_677651.iso
SHA1: A491F985DCCFB5863F31B728DDDBEDB2FF4DF8D1 (checked and it matches)

If I put the missing file on the stick, it works fine.

Did MS botch that ISO image?
I did not try to use it on actual DVD disk.

pbatard, 2016-08-22:
Please post the __FULL LOG__ of how you create the USB with Rufus.

But please also realize that Windows 7 is getting quite old, and therefore UEFI support is sketchy (for instance, the 32-bit Windows 7 ISOs cannot be booted through UEFI). For more comprehensive UEFI support (as well as chipset support - for instance Windows 7 does __NOT__ support booting from USB 3.0 at all unless you manually fix the image) you should consider upgrading to Windows 8 or later.