pbatard/rufus #783: Erreur: erreur de partitionnement

jdeyrieux, 2016-06-23:
Hello,

i have an error on my USB Key after try to write with GTP - pour UEFI option.

My detailled log is below, 

ps: I tried to use diskpart to check partition but now my USB key is always in READONLY mode and i can't use it.

Thanks for your help

Jacques

```

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x0000045D] Impossible de satisfaire à la demande en raison d’une erreur de périphérique d’E/S.
Could not reset partitions
Re-mounted volume as 'G:' after error

Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&210f7663&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[248]: USB\VID_0A12&PID_0001\6&E3E7252&0&2
  Found ID[124]: USB\VID_0951&PID_1666\00045F057BE8BF1159610482
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub30#4&11bb9039&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[233]: USB\VID_045E&PID_076D\5&1E313DEC&0&2
Processing Hub 3:
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&21d3f4f7&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[149]: USB\VID_045E&PID_0745\6&1A8123C&0&5
Processing Hub 5:
  Hub[4] = '\\?\usb#root_hub20#4&358d3b39&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[069]: USB\VID_8087&PID_0024\5&210F7663&0&1
Processing Hub 6:
  Hub[5] = '\\?\usb#root_hub20#4&230f5685&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[151]: USB\VID_8087&PID_0024\5&21D3F4F7&0&1
Processing IDs belonging to 'USBSTOR':
  USB\VID_0951&PID_1666\00045F057BE8BF1159610482
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_850_EVO_250GEMT0'
Found non-USB removable device 'Samsung SSD 850 EVO 250GB' => Eliminated
Removing C: from the list: This is the system disk!
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskKingstonDataTraveler_3.0PMAP'
  Matched with ID[124]: USB\VID_0951&PID_1666\00045F057BE8BF1159610482
  Matched with Hub[1]: '\\?\usb#vid_8087&pid_0024#5&210f7663&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is an USB 3.0 device operating at lower speed...
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000DM001-1CH1CC43'
Found non-USB removable device 'ST2000DM001-1CH164' => Eliminated
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3825, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x078523F0
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.3 GB (31465275392 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&210f7663&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[248]: USB\VID_0A12&PID_0001\6&E3E7252&0&2
  Found ID[124]: USB\VID_0951&PID_1666\00045F057BE8BF1159610482
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub30#4&11bb9039&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[233]: USB\VID_045E&PID_076D\5&1E313DEC&0&2
Processing Hub 3:
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&21d3f4f7&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[149]: USB\VID_045E&PID_0745\6&1A8123C&0&5
Processing Hub 5:
  Hub[4] = '\\?\usb#root_hub20#4&358d3b39&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[069]: USB\VID_8087&PID_0024\5&210F7663&0&1
Processing Hub 6:
  Hub[5] = '\\?\usb#root_hub20#4&230f5685&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[151]: USB\VID_8087&PID_0024\5&21D3F4F7&0&1
Processing IDs belonging to 'USBSTOR':
  USB\VID_0951&PID_1666\00045F057BE8BF1159610482
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSamsung_SSD_850_EVO_250GEMT0'
Found non-USB removable device 'Samsung SSD 850 EVO 250GB' => Eliminated
Removing C: from the list: This is the system disk!
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskKingstonDataTraveler_3.0PMAP'
  Matched with ID[124]: USB\VID_0951&PID_1666\00045F057BE8BF1159610482
  Matched with Hub[1]: '\\?\usb#vid_8087&pid_0024#5&210f7663&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is an USB 3.0 device operating at lower speed...
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST2000DM001-1CH1CC43'
Found non-USB removable device 'ST2000DM001-1CH164' => Eliminated
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3825, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x078523F0
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.3 GB (31465275392 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

```

pbatard, 2016-06-23:
It looks like the USB controller from your flash drive detected that the flash memory was getting defective (flash memory has a limited number of write cycles before it fails, and depending on how much you paid for your flash drive, it may fail much earlier than anticipated), and switched to read-only mode as a result.

I'm afraid that, when that happens, there's not much that can be done besides purchasing a new flash drive. See also [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media).

jdeyrieux, 2016-06-24:
I think too, i have read all FAQ and search on many forum but without technical solution. 

thanks for your answer, your product is very impressive ! 

Bravo 

Jacques

pbatard, 2016-06-28:
I will close this issue now.

closed 2016-06-28 by pbatard