pbatard/rufus #787: Help?

milkiski, 2016-06-28:
Rufus version: 2.9.934
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0CAC922B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ARCH_201606'
  Size: 780140544 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: archlinux-2016.06.01-dual.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'I:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Bad Blocks: Checking from block 0 to 3906963455
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Cancelling
Bad Blocks: Interrupting at block 80323648
Bad blocks: Check failed.
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors

Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x81
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive does not have an x86 Master Boot Record
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'G:' as volume mountpoint
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{0cbfafd8-0000-0000-0000-100000000000} for write access
Size : 1.8 TB 3906961408 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1714:2e56
32 Reserved Sectors, 476808 Sectors per FAT, 2 FATs
61031371 Total clusters
61031370 Free Clusters
Clearing out 953712 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476840
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{0cbfafd8-0000-0000-0000-100000000000}\
\\?\Volume{0cbfafd8-0000-0000-0000-100000000000}\ was already mounted as G:\
Installing Syslinux 6.03...
Using existing './rufus_files/syslinux-6.03/ldlinux.sys'
Using existing './rufus_files/syslinux-6.03/ldlinux.bss'
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{0cbfafd8-0000-0000-0000-100000000000} for write access
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512
Could not write Syslinux boot record: [0xC003001D] The system cannot write to the specified device.
\\?\Volume{0cbfafd8-0000-0000-0000-100000000000}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0CBFAFD8
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

It keeps failing after the format. I am trying to install Arch Linux

milkiski, 2016-06-28:
It keeps failing after the format. I am trying to install Arch Linux

pbatard, 2016-06-28:
It seems to be failing during the Syslinux installation.
I have some Syslinux enhancements lined up for 2.10.

Can you try this [test version of 2.10](https://rufus.akeo.ie/testing/rufus-2.10_TEST12.exe) and report what you get?

pbatard, 2016-06-29:
Any update on this? Did you try the test version I pointed you to?

milkiski, 2016-06-29:
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0CBFAFD8
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
Checking test channel...
No new test version found.
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ARCH_201606'
  Size: 780140544 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: archlinux-2016.06.01-dual.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{137e3180-0000-0000-0000-100000000000} for write access
I/O boundary checks disabled
Size : 1.8 TB 3906961408 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1af2:2a30
32 Reserved Sectors, 476808 Sectors per FAT, 2 FATs
61031371 Total clusters
61031370 Free Clusters
Clearing out 953712 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476840
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{137e3180-0000-0000-0000-100000000000}\
\\?\Volume{137e3180-0000-0000-0000-100000000000}\ was already mounted as G:\
Installing Syslinux 6.03...
Opened drive \\?\Volume{137e3180-0000-0000-0000-100000000000} for write access
Using existing './rufus_files/syslinux-6.03/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512
Could not write Syslinux boot record: [0xC003001D] The system cannot write to the specified device.
\\?\Volume{137e3180-0000-0000-0000-100000000000}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'WD My Passport 0827 USB Device' (1058:0827)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x137E3180
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Still not working

pbatard, 2016-07-07:
OK, I'm still trying to figure out something about this issue, but every report I read on how `WriteFile()` could return SUCCESS (the `0xC003001D` is simply a code that Rufus sets to indicate that there was a problem despite the SUCCESS code)  yet report that 0 bytes have been written (which is what really is happening here - you can't get the messages above unless both these conditions are true) seems to be pointing to an external problem with the Windows disk driver.

In other words, if an application that requests 512 bytes to be written using the Windows APIs (`WriteFile()`) gets a report from said APIs that the operation "Completed without issues" and yet, when querying the actual number of bytes written, gets the information that zero bytes were actually written, then the issue is not within the application, but with the OS or the driver.

Now, I would assume that you are using the native Windows USB Mass Storage driver, which are not really known to have issues, but I'd like to get this confirmed.

First, could I please ask you to open the Device Manager then select to your WD drive (it should be listed under _Disk drives_), and, once selected, select the menu _View_&rarr;_Devices by connection_. You should end up with something like this:
[image]

If you can paste a screenshot of what you see on your computer, I would appreciate it.

Also, with your WD drive still plugged in (no need to try reformatting it - just having it plugged will do) I would like you to run Rufus and then press <kbd>Alt</kbd>-<kbd>.</kbd> to enable USB debug mode. Then if you could copy/paste the log, which you will see will contain additional debug data, that would be great.

Thanks!

milkiski, 2016-07-11:
http://prntscr.com/bresf0

doesn't show when i alt-.

pbatard, 2016-07-11:
I'm sorry but your screenshot doesn't show the disk (NB: you can drag and drop images in the github issue tracker - it's more convenient than linking to an external file). Can you please make sure your disk is selected so that we can see the whole branch that goes to it.

At to <kbd>Alt<kbd>-<kbd>.<kbd>, this is __from within Rufus__ (not in Device Manager). With Rufus open, just press <kbd>Alt<kbd> and <kbd>.<kbd> on your keyboard. You should see a message on the status line that says _USB Debug enabled_ and if you look at the log, you should see something like:
```
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_04b3&pid_301a#5&23ce17bd&0&12#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[063]: USB\VID_04B3&PID_301B\6&1D18823C&0&1
  Found ID[068]: USB\VID_046D&PID_C03E\6&1D18823C&0&3
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#vid_045b&pid_0209#5&23ce17bd&0&9#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[199]: USB\VID_0424&PID_2134\6&2489E876&0&2
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&e427330&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[018]: USB\VID_8087&PID_8002\5&9D9154E&0&1
(...)
```

pbatard, 2016-07-18:
Hi. Just wondering if you can provide an update with what I requested. Unfortunately, your previous data was not exactly what I asked for, so if you can update it according to my last comments, that would be great. Thanks.

pbatard, 2016-08-02:
I can't really make progress on this issue until I receive the data I asked for.
I will close it until this data is posted.

Malcore, 2016-09-01:
I'm actually encountering a similar issue. I'll put both my log and a picture of my device manager here. I have a 2TB Micronet Fantom external HDD, and whenever I run Rufus it fails half way through. I have tried it on both Windows 10 and Windows 7 systems, with the issue appearing in the log on both.
Also, I can open a new issue if that would be easier.

> Rufus
version: 2.10.973
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'MICRONET FANTOM DRIVE USB Device' (13FD:1340)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white > triangle)
0 devices found
Found USB 2.0 device 'MICRONET FANTOM DRIVE USB Device' (13FD:1340)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0550DB26
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397885440 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Found conflicting isolinux.bin versions:
  '/arch/boot/syslinux/isolinux/isolinux.bin' (4.05/0x4f23eb18) vs '/isolinux/isolinux.bin' (6.03)
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ANTERGOS'
  Size: 1804861440 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: antergos-2016.07.19-x86_64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

>Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'I:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{21bec69a-7094-11e6-b634-bc5ff421e0b9} for write access
I/O boundary checks disabled
Size : 1.8 TB 3907027120 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 19e0:3a57
32 Reserved Sectors, 476816 Sectors per FAT, 2 FATs
61032397 Total clusters
61032396 Free Clusters
Clearing out 953728 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 476848
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{21bec69a-7094-11e6-b634-bc5ff421e0b9}\
\\?\Volume{21bec69a-7094-11e6-b634-bc5ff421e0b9}\ was already mounted as I:\
Installing Syslinux 6.03...
Opened drive \\?\Volume{21bec69a-7094-11e6-b634-bc5ff421e0b9} for write access
Using existing './rufus_files/syslinux-6.03/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512 - retrying...
  Wrote 0 bytes but requested 512
Could not write Syslinux boot record: [0xC003001D] The system cannot write to the specified device.
\\?\Volume{21bec69a-7094-11e6-b634-bc5ff421e0b9}\ was already mounted as I:\
Re-mounted volume as 'I:' after error

>Found USB 2.0 device 'MICRONET FANTOM DRIVE USB Device' (13FD:1340)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243201, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00185F10
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.8 TB (2000397885440 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.

[image]

mbrownnycnyc, 2016-09-04:
I am also having this same issue ([0xC003001D]) with a Western Digital disk and VMware ESXi on 2.11 BETA.
[image]

pbatard, 2016-09-05:
@mbrownnycnyc, I would guess that the reason for your error is that your WD drive uses an additional SES driver for access, and as this is a security feature, this likely is the cause of the issue.
As you can understand, I cannot support custom hard drive security features in Rufus.

Do you see this error if you try a drive that doesn't use SES?

pbatard, 2016-09-06:
Aha, I am finally able to replicate this issue, using one of the newer Silicon Image based Seagate USB &harr; SATA controllers.

I'll see if I can figure out what the issue is.

pbatard, 2016-09-06:
Okay. After playing with a bunch of drives and controllers, it appears that a bunch of SATA &harr; USB controllers will erroneously report short writes (i.e. write operation complete, but zero bytes written), even though the data was properly written.

After testing the same controller with a 1TB, 1.5TB and 2TB HDD, I can conclude that this issue only seems to manifest itself for drives that are larger than 1TB (which is consistent with the reports above, that we all with 2TB HDDs) so I have now added a workaround in Rufus to detect large drives and ignore the short write reports.

NB: Some of the recent Seagate/Silicon-Image SATA &harr; USB controllers are __weird__. For instance, they will report that a <2TB drive is actually 2TB in size, which creates all kind of issues. Of course, Seagate only uses these controllers with >2TB HDDs, but still...

Anyway, there is now [a new BETA](https://rufus.akeo.ie/downloads/rufus-2.11_BETA2.exe), with the new workaround applied, if you want to try.

mbrownnycnyc, 2016-09-06:
Great!  Thank you.

pbatard, 2016-09-06:
If the new version works for you, please report, so I can close this issue.

Malcore, 2016-09-06:
Yep, that did it. Looks like everything is working perfectly. Thanks for being so active on your project!