pbatard/rufus #790: Rufus stuck at "Copying ISO files..."

WingingIt, 2016-07-01:
I'm using rufus to create a bootable flash drive with a linux mint iso.  I'm writing to a new 32Gb USB drive, and I've set the file system format to FAT32.  My computer is a 64-bit HP pavilion laptop running Windows 10.  

Rufus seemed to run fine at first, and seems to be almost done, but it's been stuck at the same stage of "Copying ISO files..." for over a half hour now.  The log's pasted below:

>  Rufus version: 2.9.934
> Windows version: Windows 10 64-bit (Build 10586)
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta3
> System locale ID: 0x0409
> Will use default UI locale 0x0409
> Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:C08A)
> 1 device found
> Disk type: Removable, Sector Size: 512 bytes
> Cylinders: 3724, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x04DD5721
> Drive has an unknown Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Size: 28.5 GB (30635688448 bytes)
>   Start Sector: 63, Boot: Yes, Recognized: Yes
> Scanning image...
> ISO analysis:
>   Image is an ISO9660 image
>   Will use '/isolinux/isolinux.cfg' for Syslinux
>   Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
> Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
>   Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
>   Could not detect Grub version
> Disk image analysis:
>   Image has an unknown Master Boot Record
>   Image is a bootable disk image
> ISO label: 'Linux Mint 17.3 MATE 64-bit'
>   Size: 1677721600 bytes
>   Has a >64 chars filename
>   Uses: Syslinux/Isolinux v4.05
>   Uses: EFI
> Using image: linuxmint-17.3-mate-64bit.iso
> 
> Format operation started
> Requesting disk access...
> Opened drive \\.\PHYSICALDRIVE1 for write access
> Will use 'G:' as volume mountpoint
> I/O boundary checks disabled
> Analyzing existing boot records...
> Drive has an unknown Master Boot Record
> Volume has an unknown Partition Boot Record
> Deleting partitions...
> Clearing MBR/PBR/GPT structures...
> Erasing 128 sectors
> Partitioning (MBR)...
> Closing existing volume...
> Waiting for logical drive to reappear...
> Formatting (FAT32)...
> Using cluster size: 16384 bytes
> Quick format was selected
> Creating file system...
> Format completed.
> Writing master boot record...
> Drive has a Zeroed Master Boot Record
> Partition is already FAT32 LBA...
> Set bootable USB partition as 0x80
> Using Syslinux MBR
> Found volume GUID \\?\Volume{145ebda8-3e7f-11e6-8293-90489a5443cc}\
> Installing Syslinux 4.05...
> Successfully wrote 'ldlinux.sys'
> Opened drive \\?\Volume{145ebda8-3e7f-11e6-8293-90489a5443cc} for write access
> Successfully wrote Syslinux boot record
> Successfully remounted Volume{145ebda8-3e7f-11e6-8293-90489a5443cc}\ on G:\
> Copying ISO files...
[... 321 lines cut ...]
> Extracting: G:\isolinux\sk.tr (2.5 KB)
> Extracting: G:\isolinux\sl.tr (2.2 KB)
> Extracting: G:\isolinux\splash.png (39.2 KB)
> Extracting: G:\isolinux\sq.tr (2.2 KB)
> Extracting: G:\isolinux\sr.tr (3.9 KB)
> Extracting: G:\isolinux\sv.tr (2.2 KB)
> Extracting: G:\isolinux\te.tr (5.8 KB)
> Extracting: G:\isolinux\tl.tr (1.9 KB)
> Extracting: G:\isolinux\tr.tr (2.3 KB)
> Extracting: G:\isolinux\uk.tr (4.0 KB)
> Extracting: G:\isolinux\vesamenu.c32 (149.4 KB)
> Extracting: G:\isolinux\vi.tr (2.8 KB)
> Extracting: G:\isolinux\zh_CN.tr (2 KB)
> Extracting: G:\isolinux\zh_TW.tr (2.2 KB)
> Extracting: G:\MD5SUMS (23.7 KB)
> Extracting: G:\pool\contrib\b\b43-fwcutter\b43-fwcutter_1%3a018-2_amd64.deb (25.8 KB)
> Extracting: G:\pool\main\b\bcmwl\bcmwl-kernel-source_6.30.223.248+bdcom-0ubuntu0.2~lp1415880~1_amd64.deb (1.4 MB)
> Extracting: G:\pool\main\b\bogl\bogl-bterm_0.1.18-9ubuntu1_amd64.deb (28.5 KB)
> Extracting: G:\pool\main\e\efibootmgr\efibootmgr_0.5.4-7ubuntu1.2_amd64.deb (27.7 KB)
> Extracting: G:\pool\main\g\grub2\grub-efi_2.02~beta2-9ubuntu1.3_amd64.deb (2.5 KB)
> Extracting: G:\pool\main\g\grub2\grub-efi-amd64_2.02~beta2-9ubuntu1.3_amd64.deb (43.6 KB)
> Extracting: G:\pool\main\g\grub2\grub-efi-amd64-bin_2.02~beta2-9ubuntu1.3_amd64.deb (633.4 KB)
> Extracting: G:\pool\main\g\grub2-signed\grub-efi-amd64-signed_1.34.4+2.02~beta2-9ubuntu1.3_amd64.deb (239.9 KB)

Any help would be appreciated.  I've been trying to set a dual-boot with linux, and have already wasted money on a cheap flash drive that was broken and unwriteable straight out of the package.  This new one is an ADATA 32Gb drive, which is hopefully more reliable, if I haven't broken it myself already.    I might actually tear my hair out if I have to buy a third one.

Eliasdf, 2016-07-01:
Did you made an MD5 check on the iso file?? Maybe that's the problem.

WingingIt, 2016-07-01:
I downloaded and ran Microsoft's fciv md5 checker, and the result from my iso file matches the checksum given by the LinuxMint devs.

I've cancelled the write process.  I can still poke around in the flashdrive, and had windows reset its format to the factory defaults, and my minitool partition wizard doesn't show it as being unwriteable, so I doubt I've bricked the flash drive. 

I'm still not sure about what the problem could be.  I'll try it again and post the results.

pbatard, 2016-07-01:
I'll download `linuxmint-17.3-mate-64bit.iso` to test.
But in the meantime, please run a bad blocks check against your drive.

pbatard, 2016-07-01:
PS: Just so you know, you could perform the MD5 check in Rufus, by clicking the `#` button on the status bar.

pbatard, 2016-07-01:
No issue for me:
```
Rufus version: 2.9.934
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'ARCH_201605'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0051F255
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Linux Mint 17.3 MATE 64-bit'
  Size: 1677721600 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
Using image: linuxmint-17.3-mate-64bit.iso
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE7 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{3c9ea97e-83a5-11e5-8487-fcaa14e7bd8e}\
Installing Syslinux 4.05...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{3c9ea97e-83a5-11e5-8487-fcaa14e7bd8e} for write access
Successfully wrote Syslinux boot record
[... 360 lines cut ...]
Extracting: F:\pool\main\w\wvstreams\libwvstreams4.6-extras_4.6.1-7_amd64.deb (431.5 KB)
Extracting: F:\pool\non-free\s\sl-modem\sl-modem-daemon_2.9.11~20110321-9_i386.deb (501.3 KB)
Extracting: F:\preseed\cli.seed (225 bytes)
Extracting: F:\preseed\linuxmint.seed (633 bytes)
Extracting: F:\README.diskdefines (221 bytes)
Created: F:\syslinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Linux Mint 17.3 MATE 64-bit'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00484CBA
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

You may want to make sure that your USB is plugged directly into your PC, and aren't using an extension cable. Also try a different USB port just in case.

WingingIt, 2016-07-01:
I've cleared the memory on the USB stick, and had rufus check for bad blocks before writing.  It worked fine this time.  It finished in about ten or fifteen minutes.  I have no clue why it stopped before.

pbatard, 2016-07-01:
Might have been a bad connection. Or it could just be an OS issue, if the OS was doing something CPU intensive in the background.

closed 2016-07-01 by pbatard