pbatard/rufus #792: Undetermined Error

Derkse14, 2016-07-05:
I am getting the same error trying to mount any iso file. Here's the log

Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CEDA_X64FRE_EN-US_DV5'
  Size: 4216752128 bytes
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_10_education_version_1511_updated_apr_2016_x64_dvd_8712927.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'F:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'F:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1956, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0243F513
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.0 GB (16094593024 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1956, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0243F513
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.0 GB (16094593024 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt

pbatard, 2016-07-05:
1. Have you tried a different USB drive?
1. Can you run a bad blocks check in Rufus?
1. What happens if you try to format your drive using Windows (right click on the drive then select "Format").

Usually, the error you describe has nothing to do with Rufus, but everything to do with Windows not being able to format the drive for one reason or another. So I doubt there's much I can do to help you out...

Derkse14, 2016-07-05:
I did try a different usb, but it was an even cheaper one so I wasn't surprised when it didn't work either. I am able to format my drive back to its default. I ran a back blocks check, here's the output.

Rufus version: 2.9.934
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Could not get node connection information for 'USB\VID_0011&PID_7788\45588FA9': [0x00000057] The parameter is incorrect.
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1956, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0243F513
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.0 GB (16094593024 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'J_CEDA_X64FRE_EN-US_DV5'
  Size: 4216752128 bytes
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_10_education_version_1511_updated_apr_2016_x64_dvd_8712927.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Bad Blocks: Checking from block 0 to 31436799
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: 16343040
Bad Blocks: 16343041
Bad Blocks: 16343042
Bad Blocks: 16343043
Bad Blocks: 16343044
Bad Blocks: 16343045
Bad Blocks: 16343046
........
Bad Blocks: 16343272
Bad Blocks: 16343273
Bad Blocks: 16343274
Bad Blocks: 16343275
Bad Blocks: 16343276
Bad Blocks: 16343277
Bad Blocks: 16343278
Bad Blocks: 16343279
Bad Blocks: 16343280
Bad Blocks: 16343281
Bad Blocks: 16343282
Bad Blocks: 16343283
Bad Blocks: 16343284
Bad Blocks: 16343285
Bad Blocks: 16343286
Bad Blocks: 16343287
Bad Blocks: 16343288
Bad Blocks: 16343289
Bad Blocks: 16343290
Bad Blocks: 16343291
Bad Blocks: 16343292
Bad Blocks: 16343293
Bad Blocks: 16343294
Bad Blocks: 16343295
Too many bad blocks, aborting test
Bad Blocks: Check completed, 256 bad blocks found. (0/0/256 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'F:' after error

Could not get node connection information for 'USB\VID_0011&PID_7788\45588FA9': [0x00000057] The parameter is incorrect.
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [ID]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1956, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0046D2CA
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.0 GB (16094593024 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Could not get node connection information for 'USB\VID_0011&PID_7788\45588FA9': [0x00000057] The parameter is incorrect.
Found USB device 'Generic Flash Disk USB Device' (0011:7788) [ID]
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1956, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0046D2CA
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.0 GB (16094593024 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt

pbatard, 2016-07-06:
```
Bad Blocks: 16343040
Bad Blocks: 16343041
Bad Blocks: 16343042
Bad Blocks: 16343043
Bad Blocks: 16343044
```

Your USB is defective. This is not a Rufus issue.

closed 2016-07-06 by pbatard