pbatard/rufus #797: didn't report as a disk

jaimeleeruiz, 2016-07-15:
Rufus version: 2.9.934
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
Could not open '\\?\usbstor#disk&ven_smi&prod_usb_disk&rev_1100#aa04012700023369&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 2.0 device 'SMI USB DISK USB Device' (090C:1000)
Could not open '\\?\usbstor#disk&ven_smi&prod_usb_disk&rev_1100#aa04012700023369&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found

trying to create a bootable windows usb drive and the above error occurs. please assist

pbatard, 2016-07-15:
Too bad you chose not to complete the checklist. Else you would have found this [relevant entry](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) in the FAQ.

Access is denied is an issue that is external to Rufus. So I'm afraid there isn't much that can be done to help you out. Are you sure you have full administrative privileges on the computer you are running Rufus one. Also (__as you should have done if you had completed the checklist__) have you tried a different USB, or from a different computer?

pbatard, 2016-07-18:
Since there hasn't been a reply, I will close this issue.

closed 2016-07-18 by pbatard