pbatard/rufus #800: Display Write Speed (MB/s) while "Copying ISO files..."

a-raccoon, 2016-07-21:
nt

pbatard, 2016-07-21:
Rufus is not a benchmarking tool. If you want to compare write speeds on various drives, you can already use the timer to do that, so there isn't much point in trying to compute the write speed (which may slow operations down).

Rejected.

closed 2016-07-21 by pbatard

a-raccoon, 2016-07-21:
I merely recommended for 2 reasons.  The practical reason being to easily tell whether a copy is stalling or something's the matter -- assurance that it's progressing, steadily.  The second reason being that Rufus specifically boasts about its performance speed, but that information isn't readily displayed.

Sure, I bust out a calculator.  It's lame, but I could.

pbatard, 2016-07-21:
The performance is a comparison of total time with other tools. There's little point in displaying a fluctuating value (current speed) when what matters is, given the same ISO, how much time does Rufus take compared to these other tools.

As to getting assurance that it's progressing steadily, the progress bar should be more than enough. The write speed isn't something that is expected to fluctuate wildly, so, IMO, and especially if the write operation is slow, this is a very poor indicator of whether the program is stuck. Better display a completion percentage (which is something that Rufus already does when writing in DD image mode) or open up the log and see if the list of files being processed is updated.

Finally, please see [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard). All things considered, the feature you request can hardly be considered a priority (especially as it may have an impact on performance due to the introduction of instrumentation for speed computation), when compared to other features, and my time as a developer is a very limited resource. As such, I tend to concentrate on what I deem beneficial to the majority of users, who, I currently assert, won't care much about what is pure benchmarking data, so long as they are able to create a bootable USB.