pbatard/rufus #802: Could not reset partitions - 2 USB Drives turned RAW

Tekkano, 2016-07-25:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello.
I was trying to create a Windows 7 installation USB via Rufus. First time I used Verbatim Store N Go flash drive. I checked my USB drive for bad blocks to make sure everything goes smooth, none were detected. The drive itself is 8GB and has no data on it. The file system was FAT32, as usual.
I checked the options (same as on screenshot) and clicked "start".
As soon as Rufus started deleting partitions, an error popped up: Could not reset partitions.
I then tried unplugging my drive from a hub and plugging into the back sockets of the computer.
Could not reset partitions.
Only then I decided to check the drive in My Computer.
It turned into RAW. 
The worst thing about RAW drives is that to try repairing one, I need to install Windows 7. To install Windows 7, I need a working USB drive. 
Then I used another perfectly working and clean USB drive, now a 4GB Kingston DataTraveler. Using the same options, I started the operation again.
Could not reset partitions.
Another drive instantly became RAW.
This time I saved the log and it has the same error message as the previous one. Full contents below.
Attempting to use format/chkdsk is pointless, RAW drives are not supported. Rufus gives the same error on both drives. 
I searched for similar cases around the net, none helped. Those cases were mostly drives with bad sectors or various read errors.  In my case, I can recover any data with GetDataBack for FAT from them- the second drive had one file on it and I can recover it just fine.
Please, if you can explain why did it turn RAW in the first place (so I can make them usable again), and what do I do now?
I would gladly try using Rufus on a normally working drive conected to a back socket if I had any more left and if I knew that in case another drive turns RAW, I will be able to recover it. And at the moment I am unable to connect the drives to any other computer (I personally have only one).
Here's the screenshot:
[image]
Thanks in advance!

Log
---
```
Rufus version: 2.10.973
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0419
Will use default UI locale 0x0419
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (05E3:0608)
Found USB device 'Verbatim STORE N GO USB Device' (18A5:0304) [GP]
Found USB device 'WD My Passport 074A USB Device' (1058:074A) [GP]
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x295E1E81
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4006937088 bytes)
  Start Sector: 63, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Windows 7 Ultim x64'
  Size: 3116302336 bytes
  Uses: EFI
  Uses: Bootmgr
Using image: Windows 7 Ultimate SP1.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Could not delete drive layout: [0x0000001F] Присоединенное к системе устройство не работает.
Could not reset partitions
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (05E3:0608)
Found USB device 'Verbatim STORE N GO USB Device' (18A5:0304) [GP]
Found USB device 'WD My Passport 074A USB Device' (1058:074A) [GP]
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
2 devices found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 3.7 GB (4007624704 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

```

pbatard, 2016-07-25:
> Please, if you can explain why did it turn RAW in the first place

A RAW device just means that it doesn't have any partitions (just like a completely blank HDD that you would purchase from the store). It doesn't mean that it's dead, just that it's completely blank. Usually USB flash drives are sold already partitioned, so you don't get them in RAW mode, but in itself, there's nothing wrong about a RAW drive: it just needs to be repartitioned to be usable.

> and what do I do now?

Well, usually I would tell you to use Rufus (and you certainly don't need Windows 7 to repartition a RAW drive - any version of Windows will do), but you seem to have a major issue with your drives, in that they are reporting `[0x0000001F] A device attached to the system is not functioning`. This usually indicates a __hardware__ error. Are you using an USB extension cable by any chance?

One thing you can try is see if Rufus 2.9 works better (which you can download [here](https://rufus.akeo.ie/downloads/)). There were a few changes to partitioning in Rufus 2.10, so first I'd like to make sure this hasn't anything to do with that.

If Rufus 2.9 still doesn't work, then you should use `diskpart`, which is how you usually partition a drive.
Typically, you will need to run something like this, from an elevated command prompt (NB: this works in ALL versions of Windows):
```
C:\WINDOWS\system32>diskpart

Microsoft DiskPart version 10.0.10586

Copyright (C) 1999-2013 Microsoft Corporation.
On computer: VOID

DISKPART> list disk

  Disk ###  Status         Size     Free     Dyn  Gpt
  --------  -------------  -------  -------  ---  ---
  Disk 0    Online          238 GB  1024 KB        *
  Disk 5    Online         1853 GB      0 B        *
  Disk 6    Online         1853 GB      0 B        *
  Disk 7    Online           14 GB      0 B
```
__VERY IMPORTANT: Make sure you identify the RIGHT disk at this stage!__
For instance, in my example, Disk 7, which is the one I am after, is a 16GB USB (which I can recognize because the size gets reported as 14GB by in Windows, so I know it's the USB).
__IF YOU SELECT THE WRONG DRIVE HERE, YOU COULD LOSE VERY IMPORTANT DATA, SO MAKE SURE YOU DOUBLE CHECK THE DRIVE BEFORE CONTINUING!!__
```
DISKPART> select disk 7

Disk 7 is now the selected disk.

DISKPART> clean

DiskPart succeeded in cleaning the disk.

DISKPART> create partition primary

DiskPart succeeded in creating the specified partition.

DISKPART> assign letter=F

DiskPart successfully assigned the drive letter or mount point.

DISKPART>
```
If all of the above was successful, then you should get a prompt to format your drive.

However, if, as I suspect, your issue has to do with your hardware, you may get a similar error to the one reported by Rufus when using `diskpart`...

Just for curiosity, did you format the drives you used in a specific manner before you tried to use them in Rufus? Or did you use them with a specific image?

Tekkano, 2016-07-25:
Okay, thanks for an incredibly fast reply.

I did not format any of my flash drives, they were pre-formatted and ready to use. 

Unfortunately, no USB flash drives were displayed after diskpart disk list. Some googling suggested that in Widows XP diskpart does not list usb flash drives, although it sees usb HDD just fine. I tried listing drives with another 2gb usb drive I got, it was not listed either. It's a feature (or a lack of one), I guess.
While trying to successfully list disks, I realised that some of the services related to disks were not running. I thought that they might have been the cause, so I enabled all that were related to drives.

That didn't fix the issue, though. Although I realised that now I get a prompt to format my USB drive (before both said "insert disk in H:" (I don't know the exact phrase since my Windows' language isn't english)). I gave it a try. Usually in such cases it didn't even do anything, but now it did and successfully formatted the drive. I then tried using Rufus and it worked! Everything went smooth and ISO was successfully recorded (don't know yet if the installation works, but at least my USB drive works and has everything set).

Although I still had another drive as RAW, the 4GB one. I decided to do some tests and see what made everything work. The results are:
Rufus 2.9- same error as in the first post.
Rufus 2.10- same. 
Then I plugged the drive into the back socket to be sure (Yes, I was using a USB hub before). 
2.9- same results.
2.10- same.
So it doesn't seem that the changes in partitioning have any connection to the issue.
Then I decided to format the drive via "Format?" prompt. 
Another success. Rufus did not give any errors either and formatted everything correctly.

Although the issue on my side is solved and the goal is accomplished (at least the drives are recovered and the ISO was written), the way I solved it seems weird to me. Why would Rufus need the drive it made RAW to be formatted via Windows tool? I have the logs of all testing attempts, they are identical to the one in the first post. If any other testing is needed, I will be glad to help. I'm installing Win7 to another hard drive, so I'll be able to test in the current environment anyway.

P.S. I'm providing a detailed description of what I did mostly to help if anyone else has such a problem.
Thank you so much for answering! I hope I'll be able to install Win7 successfully.

pbatard, 2016-07-25:
>  I realised that some of the services related to disks were not running. I thought that they might have been the cause, so I enabled all that were related to drives.

Aha!

Now I think things are starting to make sense.
Before it can write an ISO to an USB drive, Rufus needs to do 2 things:
1. Clear and repartition the drive
2. Format the drive

All of these rely heavily on the Windows disk services. Part of the clearing is done directly (i.e. without invoking Windows APIs), but the rest relies on asking Windows to do the remaining of the clearing, partitioning and formatting. That's because, these Windows services have been extensively tested by Microsoft, and it wouldn't make sense to try to duplicate them (which would be __very__ time consuming and increase the size of the application) and probably do a poorer job at it, since Microsoft has had a lot of opportunities to make sure these services work very reliably.

As such, if you do disable disk services, it is likely that Windows will throw an error like the one you got by the time Rufus is asking Windows to complete the clearing of the partitions from the drive (which is the first disk API call that Rufus issues to the system). Unfortunately, the error message isn't as informative as something like "The relevant service is not running so Windows couldn't perform the requested operation". But it does make sense that, if you disable some disk services, then applications that need to perform disk operations will cease to work.

At least, I'm sure your issue will help people in the same situation (and I'll make sure to ask people if they disabled any disk services if they get such an error).

As to why Rufus wasn't able to format the drive but Windows could, this might have been an access issue, especially if you had the format prompt still open. But without the Rufus log of what happened when you tried to format the drive at that time, it's hard to say.

Anyway, I too hope you'll get Windows 7 installed, especially as I am no longer officially supporting XP usage, ever since Windows XP was retired by Microsoft more than 2 years ago.

pbatard, 2016-07-27:
Since I don't think there's much I can do, I'm gonna close this issue now.

closed 2016-07-27 by pbatard