pbatard/rufus #803: USB Boot Drive Not Recognized

NealChabot, 2016-07-27:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
USB Boot Drive is Not Recognized.   I have created a bootable USB drive with Rufus, which works, but now Windows 10 will not recognize it.   It is not recognized in Windows explorer nor in Disk Management.   Even Rufus will not recognize it.   This question also applies to other bootable USB drives created with other software.   So question:
How do I return this USB drive to its original state where it is recognized and no longer invisible?

Log
---
```
<FULL LOG>
```

pbatard, 2016-07-27:
> I have created a bootable USB drive with Rufus, which works

How so? What ISO did you use to create the USB? Did you use ISO or DD image mode?

Also, even if Rufus does not recognize the drive, I would still like to see the FULL LOG from Rufus from when your drive is plugged in (please don't check the _"I clicked the `Log` button"_ or _"The log I am copying is the FULL log"_ if you're not actually pasting anything under the Log section...), as it may have some info on why your drive is not listed. This is why I __ALWAYS__ ask for the log.

If your drive is not recognized by either Disk Management or Rufus, I'd be very surprised if it can still be used for boot. Have you tried booting from it again since you found it wasn't detected in Windows?

NealChabot, 2016-07-27:
Thank you so much for writing back.   I have stumbled upon a solution, which is to reboot Windows.   I had to do this for another reason and when I did, Win 10 recognized both of the USB boot drives that it would not recognize before.   The USB drive that Rufus made was in the USB slot when Win 10 rebooted, but  I don't know whether that mattered.   After I made the bootable USB drive with Rufus I used it to boot up a 2nd computer.   When I returned it back to the 1st good computer, that is when I noticed that it was not recognized.   So if anyone else has the same problem, I would say "reboot".

pbatard, 2016-07-27:
OK. Happy to hear you managed to sort it out. Since this seems to be a Windows problem, I'll close this issue.

closed 2016-07-27 by pbatard