pbatard/rufus #804: allow rufus to write non bootable iso

ByReaL, 2016-07-29:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
cannot write with rufus a non bootable iso
i have an iso that is not bootable and i tried to write it to a usb, but rufus refuses to write it.
will it be possible to allow rufus to write non bootable iso

Log
---
```
Rufus version: 2.10.973
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3738, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0E6DCB17
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 40 MB (41943040 bytes)
  Start Sector: 172, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'NAME'
  Size: 2465972224 bytes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'NAME'
  Size: 2465972224 bytes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'NAME'
  Size: 2465972224 bytes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'NAME'
  Size: 2465972224 bytes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Using Windows 7 MBR
Found volume GUID \\?\Volume{b8075f7a-b89c-11e5-82ae-180373c76cbf}\
Successfully remounted Volume{b8075f7a-b89c-11e5-82ae-180373c76cbf}\ on F:\

Found USB 2.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3738, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x4ED401F8
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.6 GB (30750951424 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'name'
  Size: 2465972224 bytes

```

pbatard, 2016-07-29:
It is exceedingly simple to:
1. Format an USB to whatever file fomat you want using the native Windows tools
2. Extract content from an ISO to an USB using [7-zip](http://www.7-zip.org/) or other applications

As such, I will reject this request: if you are using Windows, you must be able to do something like that on your own.

Any application developer will set a minimum level of proficiency for users of its application (i.e. something they will __NEVER__ try to compensate for in terms of features because they consider it a basic computing skill that all users shall have). Being able to extract files from an ISO and being able to format a non bootable USB on their own is one of these, so it doesn't make sense to spend time adding such a feature to Rufus.

closed 2016-07-29 by pbatard

ByReaL, 2016-07-29:
Hi,
i already did this but the moment i open the usb stick in windows it adds a hidden folder "System Volume Information" and who knows what else it's doing. this is why i was preferring my favorite tool to be able to write the files, anyway i thought it's just a small change in the code to relax some condition check.probably same could be achieved if i uncompressed and provide as target the usb drive maybe, but that may be error prone.
Thanks,Mircea Dan

pbatard, 2016-07-29:
> but the moment i open the usb stick in windows it adds a hidden folder "System Volume Information" and who knows what else it's doing.

A drive created by Rufus will also see a "System Volume Information" folder, since that's something modern version of Windows always create. You may get lucky if you unplug the drive fast enough, but Rufus certainly does not take any measures to avoid its creation and, as a matter of fact, this is something that distros, such as QubeOS, [have to work around](https://www.reddit.com/r/Qubes/comments/4mpfkb/installing_qubes_with_rufus_problem_to_usb_stick/d3y4zct), because Rufus certainly can't.

And, no it wouldn't be a small change, as I would likely have to explain to Rufus users who picked the non bootable option by mistake, why their USB doesn't boot (which, I can guarantee from experience, __will__ be a lot due to the application being downloaded 2 million times each month). So I have to reiterate that, even if it was an easy __coding__ change (which I don't really think it is in the first place, as it would require some UI redesign which is always a big deal), it's still a much better deal not to do it, due to the time spent supporting such a feature, than doing it.

I hope that makes sense.

ByReaL, 2016-07-29:
Hi,
yes it makes total sense and tanks very much for taking the time to provide a detailed response.It looks like i have to do some more work on my side because i need somehow to write the files from the iso to teh usb without the OS adding it's junk. (i may have to switch to a linux machine.
Thanks,Mircea Dan